google/CFU-Playground #7: Enhancement: Add Vexriscv cache miss counters, stall counters

tcal-x, 2021-03-10:
We could add any number of event counters to the VexRiscv, accessible as CSRs. This would be useful explaining variance in total cycles counts.

danc86, 2021-06-09:
After our discussion yesterday about SPI flash, I started wondering whether we could measure CPU stalls due to SPI flash -- that would be a better metric to optimise.

Am I right in assuming that while the caches are filling from SPI flash, the entire CPU is totally stalled? Or is it more sophisticated than that?

Does Vexriscv already have the ability to count those kinds of stalls, and we just need to expose it as a CSR? Or will it require some deeper digging in the guts of the CPU?

tcal-x, 2021-06-10:
Hi @danc86 ,

I did add a cache miss stall counter about a year ago (I can't remember if it was Icache or Dcache), although I'm worried I won't be able to find it.   I added it in VexRiscv, in the SpinalHDL, which I barely understood at the time and am still pretty limited with.  I'm pretty sure it was added as a RISC-V CSR, not a LiteX (memory-mapped) CSR.

It would be a lot easier if we could add stuff outside of VexRiscv, but I don't know if we could figure out what is happening exactly from the outside, by just looking at the Wishbone interface...probably not.

I'll dig around a little and let you know if I can find it.   Otherwise, I can try to recreate it, and maybe get Charles' help.

tcal-x, 2021-06-10:
@danc86 , I found my code and pushed it to the `dcache-count` branch on my fork of VexRiscv.   The commit is here: https://github.com/tcal-x/VexRiscv/commit/067137d6f02d21646abe8d6968e57322dd8bbe7e .

If you want to start building your own VexRiscv for CFU Playground, I can help get you started.