google/CFU-Playground #19: scripts/setup: If the user answers "n" to installing new packages, the script tries to install them anyway.

tcal-x, 2021-03-11:
```
tim@tim-ubuntu:~/tcal-x/CFU-Playground$ ./scripts/setup 
Missing packages: ccache
Install [y]/n? n
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Suggested packages:
  distcc | icecc
The following NEW packages will be installed:
  ccache
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 121 kB of archives.
After this operation, 369 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu focal/main amd64 ccache amd64 3.7.7-1 [121 kB]
Fetched 121 kB in 1s (221 kB/s)
Selecting previously unselected package ccache.
(Reading database ... 276510 files and directories currently installed.)
Preparing to unpack .../ccache_3.7.7-1_amd64.deb ...
Unpacking ccache (3.7.7-1) ...
Setting up ccache (3.7.7-1) ...
Updating symlinks in /usr/lib/ccache ...
Processing triggers for man-db (2.9.1-1) ...
Please install vivado
```

tcal-x, 2021-03-11:
I type "n" then return.

The variable `answer` inside the script is empty.

Ayushsunny, 2021-03-14:
I would like to work on this issue

tcal-x, 2021-03-15:
Hi @Ayushsunny , sure, please take a look, thanks!   I am concerned that this behavior might be host / terminal dependent, perhaps something related to CR vs CRLF.   So it will be useful to have people on different systems try it, and also try any proposed fix (what fixes it on one system might break it on another system). 

The first step would be for you to check if it works correctly on your system: if you type "n"<return>, the setup script *should* just exit.

12562, 2023-01-16:
I have corrected the issue in the pull-request #769