google/CFU-Playground #24: Add some linting / sanity checks / physical design summary  output

tcal-x, 2021-03-13:
We had a situation where a small mistake caused the creation of a 2048b counter.   This consumed 2000 extra LUTs and had a 85ns path, but it wasn't flagged during the build and the user (me) didn't notice anything amiss.   But on the board, there were nondeterministic failures due to the long path, and only after some investigation did we find the issue.   The following additions would have helped us find the issue sooner:

* Visible printout of the LUT count and critical path delay after building the bistream
* Maybe some linting in nMigen to flag Signals wider than say 128b
* Others?

mithro, 2021-03-13:
@tcal-x -- Non-deterministic failures are normally cause by timing not being met. Any idea why timing failure wasn't caught?

tcal-x, 2021-03-15:
> @tcal-x -- Non-deterministic failures are normally cause by timing not being met. Any idea why timing failure wasn't caught?

@mithro , it should have occurred to me sooner.   Only after everything worked in simulation, and Alan couldn't reproduce the issue, did I finally start thinking about failure in the hardware.  And then it took a while longer before I thought to run the same test multiple times to see if the failure was always the same.   The synth/place/route aspect had gotten too push-button I guess.   So I want to make it more noticeable when LUT count suddenly jumps by 2000, or CPD doubles.

tcal-x, 2021-03-15:
After discussion with TimA, we've decided that not meeting timing should be a hard error, but one that can be overridden temporarily.