google/CFU-Playground #90: about add new FPGA board

akioolin, 2021-05-25:
Hi, 

I'm trying to let CFU-Playground  run on  qmtech wukong board. The FPGA is Xilinx Artix-7 XC7A100T. 

I could build and run a 4-cores vexrisc-v + linux on this board. right now, I'm tring to add some hw design build in. the CFU-playground provides a very good way to add on custom design. 

I could add qmtech wukong in CFU-playground project and generate fpga bitstream very smoothly. but I found the generated bitstream can't make fpga booting after I download the bitstream. even the BIOS boot up message doesn't show. Is any where I missed? 

the following is what I done. 
1. add qmtech wukong board related setting in proj/proj.mk.
2. add two new file in soc folder, qmtech_wukong.mk, qmtech_wukong.py.
3. add flash code in CFU-Playground/third_party/python/litex_boards/litex_boards/targets/qmtech_wukong.py

thanks in advance.

BR, Akio

tcal-x, 2021-05-25:
Hello @akioolin , thank for your interest in the project!

I am currently not able to rerun these designs due to issue #91.

Can you find and post the critical path in your design that fails timing?

akioolin, 2021-05-26:
Hi, @tcal-x :

Thank you. Many Thanks for this project.

The timing check fail message is the following screenshots

[image]

[image]

It seems the mmcm clock output related. but for other project. there is no such problem. 

BR, Akio

tcal-x, 2021-05-26:
> for example_cfu_v, it shows "synthesis failed".

Ah, this is due to a recent change in the interface.   Thank you for finding it!   I'd be grateful if you can send a PR to fix it --- simply add an input 'rst' to the `module Cfu`.   It will be unused internally in this CFU (just like the 'clk' input), but it is still required on the interface.

akioolin, 2021-05-26:
Hi, @tcal-x :

Thank you very much. Follow your guide, I had done add rst signal into cfu.v and finished the test of example_cfu_v. It ran very OK.

the table will be update to the following.

[image]

BTW, I also upload qmtech wukong board support related file.

hope this could help more people to use CFU-Playground project.

BR, Akio

closed 2021-05-27 by akioolin