google/CFU-Playground #98: about proj_template demo code

akioolin, 2021-05-26:
Hi, @tcal-x :

I found this sample will stop output in my current setting, qmtech wukong board + buad rate 3000000. so I make some change to pass the stop problem.

here is the code.

static void do_exercise_cfu_op0(void)
{
    puts("\r\nExercise CFU Op0 aka ADD\r\n");
    int count = 0;
    int pass_count = 0;
    int fail_count = 0;

    for (int a = -0x71234567; a < 0x68000000; a += 0x10012345)
    {
        for (int b = -0x7edcba98; b < 0x68000000; b += 0x10770077)
        {
            int cfu = cfu_op0(0, a, b);
            if (cfu != a + b) 
            {
                printf("[%4d] a: %08x b:%08x a+b=%08x cfu=%08x FAIL\r\n", count, a, b, a + b, cfu);
                fail_count++;
            } else {
                pass_count++;
            }
            count++;
        }
    }
    printf("\r\nPerformed %d comparisons, %d pass, %d fail\r\n", count, pass_count, fail_count);
}

BTW, I also met the undefined code behaviour error, if I reduce the  if (cfu != a + b){} statement. the gcc will complain the "a + b" is not defined. so I change the optimization level form O3 to O1 to avoid this error.

BR, Akio

tcal-x, 2021-05-27:
Hi @akioolin , thank you for the report.   Yes, I think I've seen that too, where the USB gets overwhelmed.   It looks like it hangs, but then if you hit return a couple times, it will start interacting again.   

But yes, I agree with the change; we shouldn't print out every successful match.   I think in one of the other tests, I set it up to print out status only every N iterations...here:  https://github.com/google/CFU-Playground/blob/main/common/src/functional_cfu_tests.c#L61 

Your solution is fine too though -- can you submit a PR?

akioolin, 2021-05-27:
Hi, @tcal-x :

Yes. What you siad is what I met. the PR had been submitted. but It seems meet the O3 vs O1 problem. 

BR, Akio

akioolin, 2021-05-27:
Hi, @tcal-x :

add "-fno-aggressive-loop-optimizations" in common/Makefile's SHAREFLAGS could avoid this error.

BR, Akio

tcal-x, 2021-05-27:
Hi @akioolin , I think I'm beginning to understand the 2nd issue -- I don't see the "not defined" warning, but I do see the strange behavior when the original printf is removed.   I was being somewhat sloppy with signed/unsigned, since the cfu() operation expects unsigned.   I think the better solution is to change variables "a", "b", and "cfu' to unsigned, and adjust the "for" loop ranges accordingly.

I would prefer to not add the flag to disable optimizations, since we do in fact want the code (including the tensorflow code) to get optimized as much as possible.

Thank you for helping to figure this out and raise these issues.

akioolin, 2021-05-27:
Hi, @tcal-x :

Thank you. Sorry for my poor description. I will do some test under remove add-on flags, and change the variable setting from signed to unsigned. If it pass, I could do a another PR. 

BR, Akio

closed 2021-05-28 by tcal-x