google/CFU-Playground #122: try to using CFU under Linux OS

akioolin, 2021-06-17:
Hi, All:

I have some questions about how to enable cfu.v in litex.

1. for currently CPU support, it seems VexRisc-V only, not for VexRisc-V SMP.
2. from current project samples, the CFU function enable by using RISC-V custom instruction. the environment is bare metal machine. for linux enable sw/fw, is there 
any kernel driver module should be written for CFU operation.
3. for CFU instruction issued, how about the context switch handling? 
4. for CFU design, the memory access seems still under develop. how to consider about cpu cache, DMA, DRAM, and CFU. 

Many thanks.

BR, Akio

tcal-x, 2021-07-26:
Hi @akioolin , these are all good questions, and ones I'd been thinking about (along with other people).   I have not attempted using a CFU in any situation other than the simple bare-metal environment that we have with CFU Playground.  But that is not due to a limitation of VexRiscv.  I simply have not tried to put together a system with both CFU and multiprocessing  and virtual memory.

There is a group that I work with that has been giving it some thought.  For example, we could give the CFUs multiple state contexts, so that when we switch to a different software process, we can keep the previous process's state in the CFU and protect it  from the new process.

I will try to write more later.

akioolin, 2021-07-28:
Hi, Tim:

Many thanks. I am total agree what you said. after some study about possible architecture based on current CFU design.

I think the first limitation is project resource, like funding, man power.
the goal of CFU  is very attractive to students, makers, hackers. for real application, the limitation of device is the FPGA like chip could be more popular and cheaper. 
current FPGA solution, they could be classified to three major categories. 
1. pure fpga
2. fpga + ARM
3. fpga + RISC-V

what I'm thinking is based on CFU to provide a uniform design flow based on FPGA + RISC-V configuration. 

as I know the microchip's polarfire series fpga configuration is FPGA + RISC-V.
but I don't know the RISC-V hard core support CFU or user add on instruction or not.

from this point of view, using graph compiler compile a custom graph, and then could generate the respect ML acceleration design and the related sw stack could come true.

hope CFU-Playground could be more powerful!

BR, Akio