google/CFU-Playground #124: Conflicting uses of TARGET

tcal-x, 2021-06-17:
This is the original message; see also my following replies. 

Hi, @alanvgreen  :

After some testing with this update,  I met a problem. when compile tflm library, it complained no found to "qmtech_wukong_makefile.inc" wihch the board I used is wukong board.

Would you like to share how to solve this problem? or just simple copy the mcu_riscv_makefile.inc to digilet_arty_makefile.inc? 

Many thanks. 

BR, Akio

_Originally posted by @akioolin in https://github.com/google/CFU-Playground/issues/118#issuecomment-863387163_

tcal-x, 2021-06-17:
My followup comment:

I think I see the issue. It is due to the interaction of this PR and PR #105. #105 added the use of TARGET to specify which LiteX board we're using. This PR causes more of the TFLite-Micro system to be used for generating the build/src/tensorflow/lite/micro/ directory, and that also uses TARGET but for a different purpose. Hopefully the solution is as easy as just unsetting TARGET before calling the TFLM generation.

tcal-x, 2021-06-17:
To reproduce:
```
make TARGET=digilent_arty software
```
output:
```
/home/tim/google/CFU-Playground/third_party/tflite-micro/tensorflow/lite/micro/tools/make/Makefile:544: tensorflow/lite/micro/tools/make/targets/digilent_arty_makefile.inc: No such file or directory
make[1]: *** No rule to make target 'tensorflow/lite/micro/tools/make/targets/digilent_arty_makefile.inc'.  Stop.
make[1]: Leaving directory '/home/tim/google/CFU-Playground/third_party/tflite-micro'
make: *** [../proj.mk:209: tflite-micro-src] Error 2
```

akioolin, 2021-06-18:
Hi, @tcal-x :

Yes! You are right. CFU is a very young baby, but It will be a Giant in some day. 
after some test last night, I found the simple work around way is copy mcu_riscv_makefile.inc to qmtech_wukong_makefile.inc, and the build 
process will be continued. the outcome seems work fine. 

BR, Akio

akioolin, 2021-06-19:
Hi, @tcal-x  :

after add the unset , the build process is very smooth. 

BR, Akio

tcal-x, 2021-06-19:
Thanks for reporting the issue, @akioolin !

closed 2021-06-19 by tcal-x