google/CFU-Playground #133: On Icebreaker, TFLM unit test fails, and microbenchmarks hang

tcal-x, 2021-06-20:
I saw this in `proj/proj_template_v/`, with no models compiled in.  `CONV TEST` fails, but `DEPTHWISE_CONV TEST` passes.

This is likely a generic issue with the limited memory, not specific to Icebreaker.  

I suspect that the stack grows down into the allocated `.data` section, causing corruption/hangs.

* We should run this in Renode to confirm that this is indeed the issue.

* The BUF_SIZE used in the memory benchmarks should be shrunk down, at least for the boards with limited main_ram/sram.

tcal-x, 2021-06-20:
I tried inserting printfs in the "cached loads" benchmark, after reducing BUF_SIZE to 2kB, and captured the following output before it hang:  `buf: 0x00dee0-0x00fedc   acc: 0x00dedc`

Using `nm` to look at symbol addresses in `.data`, the highest locations I could find were:
```
00003468 b rx_buf
000034e8 b tx_buf
00003568 B __malloc_current_mallinfo
00003590 B errno
00003598 B _ebss
00003598 B _end
```

So it looks like there's plenty of space...maybe it's not actually hanging, or my theory is wrong.

OH -- it didn't hang, it was just *very slow* --- 742 cycles per load when executing out of spiflash!

For this board, the VexRiscv build has NO Icache or Dcache, and the code is located in spiflash, not main_ram or sram.

JosephBushagour, 2021-06-20:
Good catch on the TFLM unit tests, I was confused because I thought I tested them before submitting the PR. It turns out the unit tests fail only when no models are included and only for the Icebreaker (at least on the PR branch). I'll need to look into why the unit tests are behaving that way, I'm currently not sure why they'd pass when a model is included and only fail for the icebreaker when one isn't. 

As for the benchmarks, a buffer size of 128kB is a bit big for the Icebreaker, I agree we should think about including a configuration that decreases the size a bit.

tcal-x, 2021-06-24:
Fixed with #136.

closed 2021-06-24 by tcal-x