google/CFU-Playground #142: Switch to newlib

alanvgreen, 2021-06-30:
@danc86 points out that the way CFU-Playground currently builds mixes some elements of GCC's library with LiteX's libbase. In particular, some headers come from LiteX, and some from GCC, while linking has both libbase and GCC's libraries.

In order to avoid subtle bugs, we should find some way to compile the LiteX app without libbase. I believe newlib is the library shipped with riscv64-gcc.

Challenges:
- I/O routines in libbase should be extracted to another codebase. In particular, UART routines.
- Still need access to files in include/generated, both the linker scripts and header files.
- To fully port newlib, need implment "syscall" functions as described in [the Sycalls section of the newlib manual](https://sourceware.org/newlib/libc.html#Syscalls) and William Gatliff's [Porting and Using Newlib in Embedded Systems](https://www.cs.ccu.edu.tw/~pahsiung/courses/esd/resources/newlib.pdf)

mithro, 2021-07-08:
Might want to look into picolibc - https://github.com/picolibc/picolibc

It would actually be really nice to get LiteX to move away from the custom libbase to something more standard like newlib. With LTO and modern compilers removes much of the reason for libbase to exist.

mithro, 2021-07-08:
FYI - @mateusz-holenko who has previously worked on getting LTO enabled on LiteX.