google/CFU-Playground #144: Tensorflow model evaluation hangs when CFU support is turned off

danc86, 2021-06-30:
I was building for HPS with CFU support turned off in the design, using a local hack like this:

```patch
--- a/soc/hps.mk
+++ b/soc/hps.mk
@@ -34,7 +34,7 @@ endif

 PROJ_DIR:=  $(CFU_ROOT)/proj/$(PROJ)
 CFU_V:=     $(PROJ_DIR)/cfu.v
-CFU_ARGS:=  --cpu-cfu $(CFU_V)
+#CFU_ARGS:=  --cpu-cfu $(CFU_V)

 SOC_NAME:=  hps.$(PROJ)
 OUT_DIR:=   build/$(SOC_NAME)
```

and then running:

    make -C proj/proj_template PLATFORM=hps bitstream software

If I use the menu to try and execute CFU op0, the application freezes. This is expected, since the CPU tried to execute an illegal instruction.

If I use the menu to try and evaluate pdti8, the application also freezes here:

```
Tests for pdti8 model
=====================
 1: Run with zeros input
 2: Run with no-person input
 3: Run with person input
 g: Run golden tests (check for expected outputs)
 x: eXit to previous menu
pdti8> 3

Running Run with person input
Classify Person
Copied 9216 bytes at 0x0x40005680
Running pdti8
```

Since I am building `proj/proj_template` there are no customisations being applied to Tensorflow so I don't think it could be executing a CFU op. I was expecting it to evaluate the model as normal.

akioolin, 2021-06-30:
Hi, @danc86 :

After tracing code of tflite_classify() in build/src/tflite.cc.

In my own git clone,  I  found performance counter related code. could you add some message to check about the hang is before perf_reset_all_counters() or not?

BR, Akio

akioolin, 2021-06-30:
Hi, @danc86 :

the following is the test outcome in my qmtech wukong board.

Tests for pdti8 model
=====================
 1: Run with zeros input
 2: Run with no-person input
 3: Run with person input
 g: Run golden tests (check for expected outputs)
 x: eXit to previous menu
pdti8> 3

Running Run with person input
Classify Person
Copied 9216 bytes at 0x0x400ddc30
Running pdti8
 Counter |  Total | Starts | Average |     Raw
---------+--------+--------+---------+--------------
    0    |     0  |     0  |   n/a   |            0
    1    |     0  |     0  |   n/a   |            0
    2    |     0  |     0  |   n/a   |            0
    3    |     0  |     0  |   n/a   |            0
    4    |     0  |     0  |   n/a   |            0
    5    |     0  |     0  |   n/a   |            0
    6    |     0  |     0  |   n/a   |            0
    7    |     0  |     0  |   n/a   |            0
   184M (   183635444) cycles total
  result is 226

BR, Akio

akioolin, 2021-06-30:
Hi, @danc86 :

please check GenCoreDefault.scala,  in third_party/python/pythondata_cpu_vexriscv/pythondata_cpu_vexriscv/verilog/src/main/scala/vexriscv.

in 

  class PerfCsrPlugin( val csrCount : Int ) extends Plugin[VexRiscv]{
    override def build(pipeline: VexRiscv): Unit = {
    import pipeline._
    import pipeline.config._

    pipeline plug new Area{
      for (idx <- 0 until csrCount) {

          val cycleCounter = Reg(UInt(32 bits))
          val enable =       Reg(UInt(32 bits))   // only the LSB is used

          when ( enable(0) ) {
            cycleCounter := cycleCounter + 1
          }

          val csrService = pipeline.service(classOf[CsrInterface])
          csrService.rw(0xB04 + idx * 2,     cycleCounter)
          csrService.w( 0xB04 + idx * 2 + 1, enable)
      }
    }
  }
}

there some performance counter register related stuff. 

If you want to disable CFU and run the demo pure CPU only, then you have to check the following settings.

1. please change cpu setting from "full+cfu+debug" to "full+cfu" in soc/board_specific_workflows/workflow_args.py ,
parser.set_defaults(csr_csv='csr.csv',
                        uart_name='serial',
                        uart_baudrate=921600,
                        cpu_variant='full+cfu+debug',
                        with_etherbone=False)

2. enable NPROFILE in proj/proj_template/Makefile. 

hope this two points could help you to skip the cpu hangs problem.

BR, Akio

tcal-x, 2021-07-02:
Yes, add NPROFILE to DEFINES in the Makefile, or you'll encounter a hang accessing the performance counters.   I should attempt to use defines from LiteX-generated header files to do this automatically (disable the calls to the perf counters).

danc86, 2021-07-04:
Thanks for the explanations. It wasn't obvious to me that the performance counters would depend on the CFU support being enabled.

tcal-x, 2021-07-05:
A little more detail -- it's not even guaranteed that every VexRiscv built with CFU support also has the perf counters -- they are two different options to the generation script.   And in fact, for tiny FPGAs like ICE40up5k, we will need to use a VexRiscv with no perf counters even for CFU-supporting versions, since there's just no room for them.

So, we can't just assume that every CFU-enabled VexRiscv also has the perf counters.   Some alternatives:
* Add perf counter metadata to the yaml file generated with each VexRiscv, and read that into LiteX and output it in header files.
* Hardcode a table in litex/soc/cores/cpu/vexriscv/core.py containing the info for each variant whether it has the perf counters (ick -- that's a different repo that would need to be kept in sync)

Or perhaps we could make the performance counters LiteX CSRs rather than VexRiscv CSRs -- then they'd be independent of the VexRiscv variant we choose.   I'm not sure how many extra cycles (if any) each access would incur if we did it that way.