google/CFU-Playground #168: Update 'renode' target to use board-specific workflows

tcal-x, 2021-07-16:
Board-specific workflows allows us to target boards that have flows that differ from the "usual" LiteX flow.   For example, with TARGET=kosagi_fomu.     The workflow mechanism is used for the `make TARGET=<board> prog` and `make TARGET=<board> software` targets.

The newly-supported `make TARGET=<board> renode` target could perhaps(?) also use the workflow mechamism.   Currently with Fomu, I see this:
```
% make TARGET=kosagi_fomu renode
make -C /home/tim/tcal-x/CFU-Playground/soc -f /home/tim/tcal-x/CFU-Playground/soc/common_soc.mk litex-software
make[1]: Entering directory '/home/tim/tcal-x/CFU-Playground/soc'
MAKEFLAGS=-j8 /home/tim/tcal-x/CFU-Playground/scripts/pyrun ./common_soc.py --output-dir build/kosagi_fomu.proj_template_v --csr-csv build/kosagi_fomu.proj_template_v/csr.csv --cpu-cfu /home/tim/tcal-x/CFU-Playground/proj/proj_template_v/cfu.v --uart-baudrate 3686400 --target kosagi_fomu
Traceback (most recent call last):
  File "./common_soc.py", line 54, in <module>
    main()
  File "./common_soc.py", line 50, in main
    workflow.run()
  File "/home/tim/tcal-x/CFU-Playground/soc/board_specific_workflows/general.py", line 111, in run
    soc = self.make_soc()
  File "/home/tim/tcal-x/CFU-Playground/soc/board_specific_workflows/general.py", line 74, in make_soc
    return self.soc_constructor(**base_soc_kwargs)
TypeError: __init__() missing 1 required positional argument: 'bios_flash_offset'
make[1]: *** [/home/tim/tcal-x/CFU-Playground/soc/common_soc.mk:96: build/kosagi_fomu.proj_template_v/software/bios/bios.bin] Error 1
make[1]: Leaving directory '/home/tim/tcal-x/CFU-Playground/soc'
make: *** [../proj.mk:256: litex-software] Error 2
```

I think we saw this same error regarding `bios_flash_offset`  _before_ the board-specific workflow was added.

FYI @JosephBushagour @robertszczepanski

JosephBushagour, 2021-07-16:
Yes the `bios_flash_offset` error is related to the lack of a Fomu specific workflow. Because the `BaseSoC` classes of each target board don't have a common interface we need to manually override this for some boards. The Fomu specific workflow should be coming quite soon now that we have the illegal instruction handler PR merged.

tcal-x, 2021-07-16:
Thanks @JosephBushagour , I forgot that the Fomu workflow isn't fully merged yet.

robertszczepanski, 2021-07-26:
Hi @tcal-x I guess it's not a problem anymore? 
As I can see `make renode` is based on the same recipe as `make software`. Both have `$(SOFTWARE_ELF)` or `$(SOFTWARE_BIN)` as a prerequisite and it works for me.

tcal-x, 2021-07-26:
Hi @robertszczepanski , I'll close this for now, and re-open if there are issues after Fomu and Icebreaker support is merged.

closed 2021-07-26 by tcal-x