google/CFU-Playground #174: avg_pdti8 does not work

alanvgreen, 2021-07-20:
The avg_pdti8 project has been broken for an unknown period.

Should it be deleted?

tcal-x, 2021-07-20:
I noticed also that (1) it also fails with software emulation of the CFU, and (2) model golden tests pass.

So it seems the gateware matches the intended behavior, but the intended behavior may only support the layer parameterizations seen in the pdti8 model, and not be correct for other parameterizations as tested by the TFLite unit tests.

So I suspect it is an easy fix (make sure the specialized kernels are only invoked for the intended parameterizations, otherwise fall back to the general kernel).

alanvgreen, 2021-07-24:
It looks like there is no software emulation code for this accelerator. I'll add a note to the makefile so people know not to try to enable it.