google/CFU-Playground #182: Upgrade to use GCC 10.x toolchain

tcal-x, 2021-07-21:
@danc86 saw a significant performance improvement when using GCC 10.x toolchain.   We can update the docs to point to the newer version of the toolchain.

JosephBushagour, 2021-07-21:
Link to relevant discussion with numbers here: https://github.com/google/CFU-Playground/pull/172#discussion_r672673739

akioolin, 2021-07-22:
Hi :

I got a problem after update to gcc 10.2.0 from sifive release. 

In this release, it seems have no prebuild libstdc++.a for rv32im, which the march setting in CFU-Playground. 

Could any one kindly to tell me which version could be used or change march setting to which architecture?

Many thanks.

BR, Akio

danc86, 2021-07-22:
avg noticed this issue with the Sifive prebuilt 10.2 toolchain. He observed that the 10.1 release still had rv32im, so we could use that in the short term.

akioolin, 2021-07-22:
Hi, @danc86 :

Many thanks. downloading 10.1.0.

BR, Akio

mithro, 2021-07-22:
We should probably move to our own custom built toolchains in conda -> https://github.com/hdl/conda-compilers

tcal-x, 2021-09-06:
I've been experimenting with [make-env](https://github.com/SymbiFlow/make-env) (Conda) in CFU Playground, and currently the only issue is finding a Conda package for the GCC RISC-V toolchain _that includes g++_.   I've opened https://github.com/hdl/conda-compilers/issues/25 to request a package for GCC v10.1 including g++.

mithro, 2021-09-06:
> I've opened [hdl/conda-compilers#25](https://github.com/hdl/conda-compilers/issues/25) to request a package for GCC v10.1 including g++.

This is something we should be able to do ASAP. @PiotrZierhoffer - Who should look into this?

PiotrZierhoffer, 2021-09-06:
I created https://github.com/hdl/conda-compilers/pull/26 to kick this off