google/CFU-Playground #184: about image format of hps_models

akioolin, 2021-07-22:
Hi:

right now, I'm trying to add some framebuffer utility functions to help me to show some information on framebuffer via qmtech wukong board's hdmi output.

I met a problem which is about image format of hps_model. 
when I use https://rawpixels.net to check the mnv2's image data, it is RGB24. pixel order is RGB, packed. the image is correct shown very normal.

after apply the same setting on the image of hps_models, there is not shown correctly.
like the following image.

[image]

If i change the rawpixel setting to the following shows. I got the nearest image.

[image]

Could some body do me a favor to tell me the right format of hps_model's image ??

BR, Akio

alanvgreen, 2021-07-22:
Hi Akio,

hps_model's input format is 320x240 8-bit grayscale. Could you please try cat_picture.dat with those settings?

Alan

akioolin, 2021-07-22:
Hi, Alan:

Many thanks. It is right and this could be a grayscale test image for my current code.
Here is the output.

[image]

BR, Akio

alanvgreen, 2021-07-23:
Looking good!

akioolin, 2021-07-23:
Hi, Alan:

the same result is shown on hdmi monitor. 
Many Thanks.

BR, Akio

closed 2021-08-03 by akioolin