google/CFU-Playground #204: crash-course/gateware.html hello_led.py with Arty A7

marcmerlin, 2021-07-29:
@tcal-x 
https://cfu-playground.readthedocs.io/en/latest/crash-course/gateware.html
"If Using The Arty A7" gives some substitutions, but does not say what to do for main()
I randomly guessed:
```
from nmigen_boards.arty_a7 import *

if __name__ == "__main__":
  dut = TestLEDModule()
  p = ArtyA7Platform()
  p.build( dut )
```

but that does not work
```
(nmigen-tutorial) sauron:~/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7$ python leda7.py
Traceback (most recent call last):
  File "/home/merlin/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7/leda7.py", line 24, in <module>
    p = ArtyA7Platform()
NameError: name 'ArtyA7Platform' is not defined
```

marcmerlin, 2021-07-29:
Found a clue here:
```
>>> dir(nmigen_boards.arty_a7)
['ArtyA7_100Platform', 'ArtyA7_35Platform', 'Attrs', 'ButtonResources', 'Clock', 'Connector', 'DDR3Resource', 'DiffPairs', 'DiffPairsN', 'DirectUSBResource', 'Display7SegResource', 'I2CResource', 'IrDAResource', 'LEDResources', 'NORFlashResources', 'PS2Resource', 'Pins', 'PinsN', 'Platform', 'RGBLEDResource', 'Resource', 'ResourceError', 'SDCardResources', 'SDRAMResource', 'SPIFlashResources', 'SPIResource', 'SRAMResource', 'Subsignal', 'SwitchResources', 'TemplatedPlatform', 'UARTResource', 'ULPIResource', 'VGAResource', 'Xilinx7SeriesPlatform', '_ArtyA7Platform', '__all__', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'display', 'dsl', 'interface', 'memory', 'os', 'plat', 'res', 'run', 'subprocess', 'user']
```

Looks like I need:
```
if __name__ == "__main__":
  dut = TestLEDModule()
  p = ArtyA7_35Platform()
  p.build( dut )
```

marcmerlin, 2021-07-29:
But then the build still fails
```
(nmigen-tutorial) sauron:~/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7$ python leda7.py

****** Vivado v2017.4 (64-bit)
  **** SW Build 2086221 on Fri Dec 15 20:54:30 MST 2017
  **** IP Build 2085800 on Fri Dec 15 22:25:07 MST 2017
    ** Copyright 1986-2017 Xilinx, Inc. All Rights Reserved.

source top.tcl
# create_project -force -name top -part xc7a35ticsg324-1L
# add_files top.v
# read_xdc top.xdc
# synth_design -top top
Command: synth_design -top top
Starting synth_design
Using part: xc7a35ticsg324-1L
Attempting to get a license for feature 'Synthesis' and/or device 'xc7a35ti'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7a35ti'
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 888658 
---------------------------------------------------------------------------------
Starting RTL Elaboration : Time (s): cpu = 00:00:05 ; elapsed = 00:00:06 . Memory (MB): peak = 1235.133 ; gain = 82.996 ; free physical = 469 ; free virtual = 152806
---------------------------------------------------------------------------------
INFO: [Synth 8-638] synthesizing module 'top' [/home/merlin/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7/build/top.v:118]
INFO: [Synth 8-5534] Detected attribute (* keep = "TRUE" *) [/home/merlin/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7/build/top.v:124]
INFO: [Synth 8-638] synthesizing module 'cd_sync' [/home/merlin/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7/build/top.v:4]
INFO: [Synth 8-638] synthesizing module 'STARTUPE2' [/opt/Xilinx/Vivado/2017.4/scripts/rt/data/unisim_comp.v:50875]
	Parameter PROG_USR bound to: FALSE - type: string 
	Parameter SIM_CCLK_FREQ bound to: 0.000000 - type: float 
INFO: [Synth 8-256] done synthesizing module 'STARTUPE2' (1#1) [/opt/Xilinx/Vivado/2017.4/scripts/rt/data/unisim_comp.v:50875]
WARNING: [Synth 8-350] instance 'U$$0' of module 'STARTUPE2' requires 13 connections, but only 1 given [/home/merlin/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7/build/top.v:11]
ERROR: [Synth 8-3438] module 'BUFGCTRL' declared at '/opt/Xilinx/Vivado/2017.4/scripts/rt/data/unisim_comp.v:661' does not have any parameter 'SIM_DEVICE' used as named parameter override [/home/merlin/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7/build/top.v:15]
ERROR: [Synth 8-285] failed synthesizing module 'cd_sync' [/home/merlin/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7/build/top.v:4]
ERROR: [Synth 8-285] failed synthesizing module 'top' [/home/merlin/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7/build/top.v:118]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:06 ; elapsed = 00:00:07 . Memory (MB): peak = 1277.664 ; gain = 125.527 ; free physical = 514 ; free virtual = 152822
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
7 Infos, 1 Warnings, 0 Critical Warnings and 4 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Wed Jul 28 17:15:20 2021...
Traceback (most recent call last):
  File "/home/merlin/fpga/nmigen-tutorial/nmigen_getting_started/hello_leda7/leda7.py", line 25, in <module>
    p.build( dut )
  File "/home/merlin/fpga/nmigen-tutorial/lib/python3.9/site-packages/nmigen/build/plat.py", line 99, in build
    products = plan.execute_local(build_dir)
  File "/home/merlin/fpga/nmigen-tutorial/lib/python3.9/site-packages/nmigen/build/run.py", line 98, in execute_local
    subprocess.check_call(["sh", "{}.sh".format(self.script)])
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 1.
```

My full code, is:
```
from nmigen import *

class TestLEDModule( Elaboratable ):
  def __init__( self ):
    self.count = Signal( 32, reset = 0 )

  def elaborate( self, platform ):
    m = Module()

    m.d.sync += self.count.eq( self.count + 1 )
    if platform is not None:
      rgb = platform.request( 'rgb_led', 0 )
      m.d.comb += [
        rgb.g.o.eq( self.count[ 20 ] ),
        rgb.b.o.eq( ~rgb.g.o )
      ]

    return m

from nmigen_boards.arty_a7 import *

if __name__ == "__main__":
  dut = TestLEDModule()
  p = ArtyA7_35Platform()
  p.build( dut )
```

marcmerlin, 2021-07-29:
I then tried edge_top.py, and same problem (after changing platform = ArtyA7_35Platform() )
```
(nmigen-tutorial) sauron:~/fpga/nmigen-tutorial/playground/lights$ python edge_top.py

****** Vivado v2017.4 (64-bit)
  **** SW Build 2086221 on Fri Dec 15 20:54:30 MST 2017
  **** IP Build 2085800 on Fri Dec 15 22:25:07 MST 2017
    ** Copyright 1986-2017 Xilinx, Inc. All Rights Reserved.

source top.tcl
# create_project -force -name top -part xc7a35ticsg324-1L
# add_files top.v
# read_xdc top.xdc
# synth_design -top top
Command: synth_design -top top
Starting synth_design
Using part: xc7a35ticsg324-1L
Attempting to get a license for feature 'Synthesis' and/or device 'xc7a35ti'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7a35ti'
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 897335 
---------------------------------------------------------------------------------
Starting RTL Elaboration : Time (s): cpu = 00:00:05 ; elapsed = 00:00:06 . Memory (MB): peak = 1235.133 ; gain = 82.996 ; free physical = 1442 ; free virtual = 153267
---------------------------------------------------------------------------------
INFO: [Synth 8-638] synthesizing module 'top' [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:128]
INFO: [Synth 8-5534] Detected attribute (* keep = "TRUE" *) [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:134]
INFO: [Synth 8-638] synthesizing module 'U$$0' [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:4]
INFO: [Synth 8-155] case statement is not full and has no default [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:17]
INFO: [Synth 8-256] done synthesizing module 'U$$0' (1#1) [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:4]
INFO: [Synth 8-638] synthesizing module 'cd_sync' [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:24]
INFO: [Synth 8-638] synthesizing module 'STARTUPE2' [/opt/Xilinx/Vivado/2017.4/scripts/rt/data/unisim_comp.v:50875]
	Parameter PROG_USR bound to: FALSE - type: string 
	Parameter SIM_CCLK_FREQ bound to: 0.000000 - type: float 
INFO: [Synth 8-256] done synthesizing module 'STARTUPE2' (2#1) [/opt/Xilinx/Vivado/2017.4/scripts/rt/data/unisim_comp.v:50875]
WARNING: [Synth 8-350] instance 'U$$0' of module 'STARTUPE2' requires 13 connections, but only 1 given [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:31]
ERROR: [Synth 8-3438] module 'BUFGCTRL' declared at '/opt/Xilinx/Vivado/2017.4/scripts/rt/data/unisim_comp.v:661' does not have any parameter 'SIM_DEVICE' used as named parameter override [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:35]
ERROR: [Synth 8-285] failed synthesizing module 'cd_sync' [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:24]
ERROR: [Synth 8-285] failed synthesizing module 'top' [/home/merlin/fpga/nmigen-tutorial/playground/lights/build/top.v:128]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:06 ; elapsed = 00:00:06 . Memory (MB): peak = 1277.664 ; gain = 125.527 ; free physical = 1453 ; free virtual = 153283
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
10 Infos, 1 Warnings, 0 Critical Warnings and 4 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Wed Jul 28 17:44:39 2021...
Traceback (most recent call last):
  File "/home/merlin/fpga/nmigen-tutorial/playground/lights/edge_top.py", line 27, in <module>
    platform.build(Top(), do_program=True)
  File "/home/merlin/fpga/nmigen-tutorial/lib/python3.9/site-packages/nmigen/build/plat.py", line 99, in build
    products = plan.execute_local(build_dir)
  File "/home/merlin/fpga/nmigen-tutorial/lib/python3.9/site-packages/nmigen/build/run.py", line 98, in execute_local
    subprocess.check_call(["sh", "{}.sh".format(self.script)])
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 1.

```

tcal-x, 2021-07-29:
Hi @marcmerlin ,

I've replicated your error, using Vivado 2017.2.   I hadn't done this tutorial before, so I've learned a bit.

One possible explanation is that we have an old Vivado and a new nMigen.  I can try on another laptop where I have Vivado 2019.1.

Here's a semi-workaround: 
* cd into build/
* edit top.v to delete the line `.SIM_DEVICE("7SERIES")`
* rerun the build: `bash ./build_top.sh`

This should get you a bitstream, but it won't put it on the board for you.

This seems to be the story behind why `SIM_DEVICE` is added: https://github.com/nmigen/nmigen/issues/438

tcal-x, 2021-07-29:
I don't see the error when using Vivado 2019.1 (no edit of top.v required).    It works on the board (I slowed down the speed of the alternating colors by using `self.count[ 26 ]` instead of `self.count[ 20 ]`).

I also found that the build finishes successfully with 2017.2 without the manual top.v edit if I delete this line in my local installation of nMigen and re-install: https://github.com/nmigen/nmigen/blob/master/nmigen/vendor/xilinx_7series.py#L324

@alanvgreen , for the purpose of this tutorial, should we just require that the user install a Vivado no older than 2019.1?

marcmerlin, 2021-07-29:
Thanks. I thought the intermediate language was standard and would work across versions.
I have an old version because I wasn't able/allowed to download myself, because their export verification wasn't able to verify that I reside in the US :-/
I guess I'll try again to see if I can get the new version