google/CFU-Playground #221: Missing Documentation for adding custom models.

akku-t, 2021-08-09:
I've been playing around with CFU playground and want to test my custom model. It would be great if the steps for adding one's own model gets documented.

JosephBushagour, 2021-08-12:
Hi @akku-t, thanks for making this issue; I agree that having docs for adding a model would be good. In the meantime though, here are some pointers that might help you get your TFLM model running on the CFU-Playground.

Models, test inputs, and model tests are located in `/common/src/models`. When adding a new model, create a new directory (`/common/src/models/<your-model-here>`) with your `.tflite` model file and model tests that let you interact with your model from the CFU-Playground menu system. You can also place hex dumps of test inputs in `.dat` files that you can reference in your model tests, a good example of this is the [keyword spotting model directory](https://github.com/google/CFU-Playground/tree/main/common/src/models/mlcommons_tiny_v01/kws).

After creating your model folder, make it accessible from the menus by adding it to [/common/src/models.c](https://github.com/google/CFU-Playground/blob/main/common/src/models/models.c). Finally, add the minimum `kTensorArenaSize` required by your model here: https://github.com/google/CFU-Playground/blob/b7d73dd51dcd61b198c3b5cf1f203c9c6c83c325/common/src/tflite.cc#L78-L107

With this you should be able to include your model from any project folder. Let me know if anything requires more clarification!

hoanggiamta, 2022-07-13:
I could find model_name.h and model_name.cc (for example pdti8.h and pdti8.cc) along with the .tflite file. How can I generate the header file and the .cc file? Or they are generated automatically by CFU Playground?