google/CFU-Playground #230: Can't bump tflite-micro: new "PIL" (Pillow) Python dependency

tcal-x, 2021-08-15:
I tried bumping the tflite-micro submodule to current main, and encounter "No module named 'PIL'" when building software.

I don't think we have a precedent of using pip to install a missing Python package.   I tried cloning https://github.com/python-pillow/Pillow under `third_party/python/`, but it was not found.   It may be that the Pillow repository is structured a bit differently than other Python repos -- there is a PIL directory at `Pillow/src/PIL`.

Or perhaps we're currently not setting up PYTHONPATH for the tflite-micro build.

danc86, 2021-08-15:
PIL has some C bits that need compiling, which is why it's just not a plain Python package that you can clone and import.

Maybe the best option is to just `apt install python3-pil`?

tcal-x, 2021-08-16:
Thanks @danc86 , I thought that would be the answer!   But it didn't work as expected on my Ubuntu 20.04LTS laptop nor on my work laptop.   I tried `pip3 install Pillow==7.1.2` on my personal laptop, and it got me past this issue to the next issue (a complaint about not including the correct header for `printf`).

tcal-x, 2021-08-17:
After working around the above two issues, I encounter this issue:
```
make[1]: Entering directory '/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build'
  CXX reduce.cc reduce.o
In file included from /home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/common.h:29,
                 from /home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/reference/reduce.h:19,
                 from /home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/micro/kernels/reduce.cc:16:
/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/types.h: In function 'TfLiteStatus tflite::ops::micro::reduce::EvalMean(TfLiteContext*, TfLiteNode*)':
/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/types.h:185:33: error: 'resolved_axis' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         if (idx == axis[axis_idx]) {
                    ~~~~~~~~~~~~~^
/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/types.h:185:33: error: 'resolved_axis' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         if (idx == axis[axis_idx]) {
                    ~~~~~~~~~~~~~^
/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/kernels/internal/types.h:185:33: error: 'resolved_axis' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         if (idx == axis[axis_idx]) {
                    ~~~~~~~~~~~~~^
cc1plus: all warnings being treated as errors
make[1]: *** [Makefile:188: /home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/src/tensorflow/lite/micro/kernels/reduce.o] Error 1
make[1]: Leaving directory '/home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build'
make: *** [../proj.mk:190: /home/tim/tcal-x/CFU-Playground/proj/proj_template_v/build/software.bin] Error 2
```

tcal-x, 2021-10-19:
> How are you installing the Python dependencies? You should just need to add it to your `requirements.txt` or `environment.yml` file?

Yes, when we switch to Conda* installation, that is what we will do.   Currently, we explicitly set PYTHONPATH to include everything under `third_party/python/`.   For example, to handle the new `meson` dependency from picolibc, we added `meson` as a submodule (https://github.com/google/CFU-Playground/commit/ad5cbfc8d5ec48cc2b3b11a12e09c79d929fd849) instead of adding it to a `requirements.txt` file.

\* Two of the barriers have been addressed (differences in performance Conda vs fresh-built, and lack of a RISCV toolchain package with g++), so we should be able to move forward with using Conda / `make-env` now.

alanvgreen, 2021-10-21:
@mithro All the mechanisms for setting up environments have interesting shortcomings. The `/environment` script we have is not how I would have chosen to do it, but it's been working well. Perhaps we could discuss replacing it in person or on a different issue?

alanvgreen, 2021-10-21:
@tcal-x In terms of timing I think we'll be happy with the current TfLM version for the next six months or so, so happy to wait while the environment issues are sorted out.

mithro, 2021-10-21:
The three primary options for Python environments are;
 * (a) System install (IE using packages / pip and/or `pip --user`). No isolation from what the user has previously installed.
 * (b) "Python-only" `virtualenv` system (couple of choices being `virtualenv`, `venv` and `pipenv`). Creates a "self contained" environment containing Python. Anything not in directly in Python is provided by the system. Good for pure Python systems.
 * (c) Conda. A self contained environment containing Python and other packages which would be provided by the system otherwise. Good for systems which want to support multiple platforms (Linux, Windows, Mac).

In both (b) and (c) any locally provided dependencies (like through submodules) need to get "installed" into the environment.

In (a) any locally provided dependencies (like through submodules) **need** to get installed onto the users system. However, people hate doing this so they slowly end up re-creating either (b) or (c) poorly (I know because I've fallen into this trap myself).

It is actually pretty easy to start with (b) and then add / extend to (c).  I have [some notes around this topic here](https://docs.google.com/document/d/1AaL460Uf_8_9v8EIy4pAZYTFOFXbv5O4RRU7Jv4t_Z4/edit#heading=h.eods90sxr89u).