google/CFU-Playground #236: `mnv2_first` CFU timeout in Renode

robertszczepanski, 2021-08-23:
CFU has been recently added to Renode and I am working on using it in CI in CFU-Playground. `example_cfu_v` seems to work fine but I've encountered a problem with `mnv2_first` project.

The verilated peripheral is built using [Renode Verilator Integration repository](https://github.com/antmicro/renode-verilator-integration/tree/cfu-samples) and [Verilator Integration Library](https://github.com/renode/renode/tree/master/src/Plugins/VerilatorPlugin/VerilatorIntegrationLibrary) inside Renode VerilatorPlugin. This generates library that is then bind with Renode by adding [Verilated.CFUVerilatedPeripheral](https://github.com/renode/renode/blob/master/platforms/cpus/verilated/litex_vexriscv_verilated_cfu.repl#L33) to a platform. After that whenever there is an opcode pattern that matches CFU pattern CFU is being executed [in Renode](https://github.com/renode/renode/blob/master/src/Plugins/VerilatorPlugin/Verilated/Peripherals/CFUVerilatedPeripheral.cs#L254) which executes a [function from verilated peripheral library](https://github.com/renode/renode/blob/master/src/Plugins/VerilatorPlugin/VerilatorIntegrationLibrary/src/buses/cfu.cpp#L38).

I've already noticed that `functionID` is incorrectly retrieved from instruction pattern and there is a fix about to be merged soon. It will be done by merging `funct7` with `funct3` like `(funct7 << 3) + funct3`. The problem is everything works fine for `example_cfu_v` but doesn't work for `mnv2_first`. For mnv2 project every CFU execution ends with `Operation timeout` from our CFU which means that `rsp_valid` is never set to `1`. I also tried checking `rsp_payload_response_ok` instead of `rsp_valid` and then there's no timeout but all tests fail anyway so I'm not sure if that's a good approach.

Maybe CFU expects something more to be set from CPU in case of `mnv2_first` and it's not properly handled in `execute()`?
If you have any ideas I would be grateful for help :)

FYI @tcal-x

tcal-x, 2021-08-23:
If functionID is not hooked correctly, there is a good chance that is causing the hang.   Each CFU might examine as many or as few (even none!) of the functionID bits as it wants.   `example_cfu_v` only examines the lower two bits of `funct3`, so it will work correctly even if `funct7` is not correctly connected. 

 `mnv2_first` uses the `funct7` bits, and also it is stateful, so if either the `funct7` bits are incorrect, or if one instruction is accidentally sent to the CFU twice, it can mess up the internal counts and cause the CFU to hang.

robertszczepanski, 2021-08-23:
Sure I understand that, I guess I didn't specify that but situation described above happens with `funct7` being merged with `funct3` to create `functionID`. Timeout happens already on first CFU execution and second execution doesn't happen before first one is finished.

tcal-x, 2021-08-23:
Can you capture waveforms from the Verilated execution?    Which menu item are you using?    What is the first functionID?

robertszczepanski, 2021-08-24:
Unfortunately I can't capture waveforms right now because it's not supported for verilated CFU yet. I was testing "TFLM Models Menu (1) -> Person Detection int8 model (1) -> Run with zeros input (1)". Software was generated with `make software` in `mnv2_first` project directory.

After a bit closer investigation I see that I was wrong. There are many CFU functions that are completed succesfully but all commands that refer to `funct7=34` fail with timeout. If we look at software_cfu.cc it is `oq_get()`: https://github.com/google/CFU-Playground/blob/main/proj/mnv2_first/src/software_cfu.cc#L333
No matter what is the whole instruction it always fails with timeout when `funct7` equals `34`.

tcal-x, 2021-08-25:
How many cycles is allowed before a timeout?   That instruction is probably the first to attempt to retrieve outputs from the CFU, but the CFU may still be working at producing the results.  It will not respond to this instruction until it finishes its current block of computation.   This may be a full matrix x vector multiplication.

This file has definitions of the use of registers specified by funct7:

https://github.com/google/CFU-Playground/blob/main/proj/mnv2_first/src/mnv2_cfu.h#L49

robertszczepanski, 2021-08-25:
Default timeout is 2000 ticks but I tried to change it to 20 000 000 or even put it in an infinite while loop that ends when `resp_valid` asserts and it didn't leave a loop for few minutes.

alanvgreen, 2021-08-28:
mnv2_first has a [golden op test](https://github.com/google/CFU-Playground/blob/main/proj/mnv2_first/src/golden_op_tests.cc#L1286) - accessible from the project menu - which might be useful for debugging this issue. 

It performs one ML operation with known inputs and an expected output and should give you an idea whether all responses are wrong or only some of them. If nothing else, running a single op will be quicker than a whole model.

robertszczepanski, 2021-09-02:
Hi @tcal-x @alanvgreen, just letting you know that I managed to run `mnv2_first` CFU and it passes all pdti8 tests and a test in project menu. It has been merged into [Renode](https://github.com/renode/renode) and [Renode-Verilator-Integration](https://github.com/antmicro/renode-verilator-integration) upstream.

In case you would want to enable tracing just add `--trace` to `VERI_LIB_ARGS` [here](https://github.com/antmicro/renode-verilator-integration/blob/master/samples/cfu_mnv2/CMakeLists.txt#L18) and your dump will be saved in `simx.vcd`.
For full tutorial about building library please refer to Renode docs about [co-simulating verilated peripheral](https://renode.readthedocs.io/en/latest/tutorials/verilator-cosimulation.html#building-a-verilated-peripheral). Just keep in mind that you only need libVtop so `make libVtop` is the command you want to use instead of `make`.
You can also use generated libs from [artifacts](https://github.com/antmicro/renode-verilator-integration/actions/runs/1190518920) in GHA of Renode-Verilator-Integration repo.

closed 2021-09-06 by robertszczepanski