google/CFU-Playground #254: Vivado timing check is not done for Arty

danc86, 2021-09-06:
If I build for Arty, like this:

```
make PLATFORM=common_soc TARGET=digilent_arty bitstream
```

the timing check for Vivado is not done. Timing failures are ignored. I really want them to be reported instead.

The Makefiles do the timing check when `USE_VIVADO` is defined. The problem is that Litex will use Vivado by default for Arty (and other boards using Xilinx parts I guess), even though `USE_VIVADO` is not explicitly defined. The timing check is needed in those cases too.

This is only a problem for Vivado. For nextpnr, timing failures are indicated with a non-zero exit status. For Radiant, Litex checks the logs itself. It's only Vivado where we have to check the logs ourselves in the Makefile.

mithro, 2021-09-06:
This sounds like LiteX should be checking the logs rather than doing it ourselves?

tcal-x, 2021-09-06:
It's strange @danc86 , this was the first timing check that I added  (just a grep of the Vivado build log), and I haven't disabled it as far as I know.   Let me poke around.

Yes, we noticed earlier that LiteX is inconsistent in this way -- some build backends have the check, others don't.

mithro, 2021-09-06:
Please make LiteX consistent rather then working around it.

tcal-x, 2021-09-06:
https://github.com/google/CFU-Playground/blob/main/soc/common_soc.mk#L79-L92

I suspect that now the timing check is performed only if you explicitly set "USE_VIVADO", but it isn't performed if you use Vivado as the default.

Yes, the Makefile ifdef logic is quite bug-prone.

danc86, 2021-09-06:
Right, as a workaround while I'm developing hps_accel I am now running this, to get the timing check:

```
make PLATFORM=common_soc TARGET=digilent_arty USE_VIVADO=1 software
```

tcal-x, 2021-09-06:
> Please make LiteX consistent rather then working around it.

:+1: