google/CFU-Playground #260: make renode is broken

alanvgreen, 2021-09-13:
Running `make renode` in proj/proj_template results in this error:

```
make[1]: Leaving directory '/usr/local/google/home/avg/src/CFU-Playground/proj/proj_template/build'
/usr/local/google/home/avg/src/CFU-Playground/scripts/generate_renode_scripts.py /usr/local/google/home/avg/src/CFU-Playground/soc/build/digilent_arty.proj_template/csr.json digilent_arty /usr/local/google/home/avg/src/CFU-Playground/proj/proj_template/build/renode/ --repl /usr/local/google/home/avg/src/CFU-Playground/proj/proj_template/build/renode/digilent_arty_generated.repl
Traceback (most recent call last):
  File "/usr/local/google/home/avg/src/CFU-Playground/scripts/generate_renode_scripts.py", line 154, in <module>
    main()
  File "/usr/local/google/home/avg/src/CFU-Playground/scripts/generate_renode_scripts.py", line 139, in main
    f.write(generate_litex_renode_repl(args.conf_file, etherbone_peripherals, args.autoalign_memor_regions))
  File "/usr/local/google/home/avg/src/CFU-Playground/scripts/generate_renode_scripts.py", line 47, in generate_litex_renode_repl
    result = litex_renode.generate_repl(csr, etherbone_peripherals, autoalign)
TypeError: generate_repl() takes 2 positional arguments but 3 were given
make: *** [../proj.mk:176: renode-scripts] Error 1
```

I have seen this error in hps_accel, too.

tcal-x, 2021-09-13:
Hi Alan,

The line 47, `result = litex_renode.generate_repl(csr, etherbone_peripherals, autoalign)`, was a recent change by me to match changes in litex_renode.   This was to get it to work in Github Actions, not interactively.   It may be that we now need to make another update so that it works interactively.   

Although I think the interactive `make renode` was broken even before my change (#256).

alanvgreen, 2021-09-13:
@tcal-x Ah, thanks for chasing this down. Not sure if this is still relevant, but I ran scripts/setup again and can confirm that I still see the above error.

robertszczepanski, 2021-09-13:
Hi @alanvgreen, I checked this on my machine and it works fine for me. I think you may be missing update of LiteX Renode submodule and you may be using version before [this](https://github.com/litex-hub/litex-renode/commit/3b8d8080c5fc33790b950c7a71ac42df1066e28d#diff-0e0f9f217c37333a4ae8ba53d3f9fe00d9b3e3267f8a1522949a78b4894443d8R520) change. Please, make sure your submodules are matching submodules in this repo and let me know if there's still an issue occuring.

tcal-x, 2021-09-13:
* I ran ./scripts/setup again and got a new version of Renode portable.   This *might* be a factor.
* In `proj/proj_template_v/`, `make PLATFORM=common_soc renode` worked.  (Why do I need to specify PLATFORM now?)
* In `proj/proj_template/`, it didn't work.   BUT, if I created an empty subdirectory `renode/`, then it worked.

The action item is to fix the issue we see when ./renode/ isn't present:  either create it if it doesn't exist, OR, instead use `./build/renode/`.    The latter choice seems better.

PR #263 (2021-09-13): Fix #260, remove unnecessary copy of software.elf to ./renode/.
Fixes #260.    There was a copy in the make recipe from the old way of doing things,
and it would cause an error when there was not already a ./renode directory.
commit 8d22f455 (2021-09-13): Fix #260, remove unnecessary copy of software.elf to ./renode/.
closed 2021-09-13 by alanvgreen