google/CFU-Playground #271: Renode: there should be a single-command to test in Renode

alanvgreen, 2021-09-15:
When I test on arty, I can run

```
$ make unit
```

Which will load the C code, choose some menu items then exit back to the command line.

It would be nice if there were an equivalent with renode, as it would allow faster iteration. 

I seem to recall that there used to be something like this for renode? Or am I misremembering?

tcal-x, 2021-09-15:
On the board, there are both `make unit` and `make run`, which both use the expects script.   The expects script is given a sequence of menu choices, so the exact thing being tested can be set in the project's Makefile.

With `PLATFORM=sim` (Verilator simulation), there is `make unit`, same as above.

In CI, we run tests in Renode using robot scripts instead of an 'expects' script.   It's possible that we could drive a Renode simulation with an expects script, although we'd need to remap the TTY to the terminal where Renode is started (I'm pretty sure this is possible, I just don't know how).   Or, we could instead use robot.

@PiotrZierhoffer , would you recommend using robot for a non-CI test like this?

mithro, 2021-09-15:
@kgugala / @PiotrZierhoffer - This is most certainly something that should be fixed. Renode should be *easier* to use than the real hardware.

PiotrZierhoffer, 2021-09-16:
It is actually possible to reroute our uart to console.

As Renode is not a `PLATFORM`, I'd add a `make unit-renode` target, if that's fine.

I believe I have a working version, so I'll PR.

@tcal-x re using robot - I would very strongly recommend using robot for this!

It's a bit of an overhead at first, but it gives you proper reports, shows exactly what happened when, allows to fine-control the emulation from within the test etc. I understand the usefulness of `expect` - it's super easy to do simple things, but when we get to more complex scenarios then they stop being readable or controllable.

So I'll push the PR in a sec, but looking into Robot may prove to be beneficial in everyday use :)

closed 2021-09-17 by tcal-x

alanvgreen, 2021-09-17:
Now works for me, thanks!

PR #273 (2021-09-17): Renode expect support
This adds 3 targets:

- renode-headless, which just runs Renode in a single console
- unit-renode, same as `make unit` but running in Renode
- run-renode, similar to `make run`, added for completeness

All of these accept the `TARGET` variable to select the actual target.

Closes #271