google/CFU-Playground #286: hps_accel on PLATFORM=hps fMax is low

alanvgreen, 2021-09-20:
fMax for builds of hps_acccel on hps is quite low.

commit 3302c0 ([link](https://github.com/google/CFU-Playground/pull/251/commits/3302c04971850566c5c8a588ed6a973d356ab2d8)) sets the clock divisor to 12, making the clock speed about 37.5MHz, and the toolchain often report fMax in high 40MHz range.

We'd like to understand what keeps fMax low, and develop a plan to increase it.

Ideas:

* what is fMax of Full VexRiscV config (i.e most common build) vs SlimPerf+Cfu
* is this a misconfiguration?
* is it due to the fix for CFU instruction in branch delay slot?

alanvgreen, 2021-09-20:
@kgugala Please take a look

kgugala, 2021-09-21:
@piotr-binkowski please take a look at this one

piotr-binkowski, 2021-09-21:
I did some initial tests using this code https://github.com/antmicro/CFU-Playground/tree/hps (this uses #280 as a base and uses updated LiteSPI core)
From my tests it looks like Vex Slim and Vex Full when built for the minimal SoC show the same fMax but the full SlimPerfCFU configuration shows only around ~43MHz
```
./scripts/pyrun soc/hps_soc.py --build

Info: Max frequency for clock 'por_clk$glb_clk': 89.05 MHz (PASS at 41.25 MHz)

./scripts/pyrun soc/hps_soc.py --build --slim_cpu

Info: Max frequency for clock 'por_clk$glb_clk': 89.05 MHz (PASS at 41.25 MHz)

proj/hps_accel/ > PLATFORM=hps make bitstream

Info: Max frequency for clock 'por_clk$glb_clk': 43.20 MHz (PASS at 41.25 MHz)
```

I will need to do more testing.

alanvgreen, 2021-09-22:
@tcal-x reports that with trivial CFU can still get 79MHz.

Perhaps problem is with cfu.py infra?

Next step: check with JSON critical path report.

piotr-binkowski, 2021-09-22:
I did some more builds and it looks like `hps_accel` has its critical path inside VexRiscV

```
Info: Critical path report for clock 'por_clk$glb_clk' (posedge -> posedge):
Info: curr total
Info:  0.4  0.4  Source VexRiscv.writeBack_arbitration_isValid_FD1P3IX_Q.Q
Info:  0.6  1.0    Net VexRiscv.lastStageIsValid budget 2.240000 ns (40,11) -> (39,11)
Info:                Sink VexRiscv.memory_to_writeBack_MEMORY_ENABLE_LUT4_C.D
Info:                Defined in:
Info:                  /home/pbinkowski/work/litespi/CFU-Playground/soc/build/hps.hps_accel/gateware/hps_proto2_platform.v:2605.10-2643.2
Info:                  /home/pbinkowski/work/litespi/CFU-Playground/third_party/python/pythondata_cpu_vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv_SlimPerfCfu.v:707.23-707.52
Info:  0.3  1.3  Source VexRiscv.memory_to_writeBack_MEMORY_ENABLE_LUT4_C.F
Info:  3.0  4.2    Net VexRiscv.memory_to_writeBack_MEMORY_ENABLE_LUT4_C_Z budget 4.095000 ns (39,11) -> (40,8)
Info:                Sink VexRiscv._zz_216_WIDEFN9_B0_Z_LUT4_C.B
Info:  0.3  4.5  Source VexRiscv._zz_216_WIDEFN9_B0_Z_LUT4_C.F
Info:  7.5 12.0    Net VexRiscv._zz_216_WIDEFN9_B0_Z_LUT4_C_Z budget 2.730000 ns (40,8) -> (46,13)
Info:                Sink VexRiscv.memory_DivPlugin_div_done_LUT4_C_Z_LUT4_B.C
Info:  0.3 12.2  Source VexRiscv.memory_DivPlugin_div_done_LUT4_C_Z_LUT4_B.F
Info:  0.4 12.6    Net VexRiscv.memory_DivPlugin_div_done_LUT4_C_Z_LUT4_B_Z budget 1.364000 ns (46,13) -> (46,13)
Info:                Sink VexRiscv.execute_CsrPlugin_wfiWake_LUT4_C_Z_WIDEFN9_C0_Z_LUT4_D_Z_LUT4_C.D
Info:  0.3 12.9  Source VexRiscv.execute_CsrPlugin_wfiWake_LUT4_C_Z_WIDEFN9_C0_Z_LUT4_D_Z_LUT4_C.F
Info:  2.2 15.0    Net VexRiscv.execute_CsrPlugin_wfiWake_LUT4_C_Z_WIDEFN9_C0_Z_LUT4_D_Z_LUT4_C_Z budget 1.364000 ns (46,13) -> (44,9)
Info:                Sink VexRiscv.decode_to_execute_BYPASSABLE_EXECUTE_STAGE_LUT4_B_Z_WIDEFN9_D0$widefn_comb[0]$.SEL
Info:  0.2 15.2  Source VexRiscv.decode_to_execute_BYPASSABLE_EXECUTE_STAGE_LUT4_B_Z_WIDEFN9_D0$widefn_comb[0]$.OFX
Info:  0.3 15.5    Net VexRiscv.decode_to_execute_BYPASSABLE_EXECUTE_STAGE_LUT4_B_Z_WIDEFN9_D0_Z budget 1.364000 ns (44,9) -> (44,9)
Info:                Sink VexRiscv.decode_INSTRUCTION_ANTICIPATED_LUT4_Z.D
Info:  0.3 15.8  Source VexRiscv.decode_INSTRUCTION_ANTICIPATED_LUT4_Z.F
Info:  5.8 21.6    Net VexRiscv.decode_INSTRUCTION_ANTICIPATED[23] budget 1.364000 ns (44,9) -> (60,14)
Info:                Sink VexRiscv.RegFilePlugin_regFile.7.0.0$lutram_comb[1]$.A
Info:                Defined in:
Info:                  /home/pbinkowski/work/litespi/CFU-Playground/soc/build/hps.hps_accel/gateware/hps_proto2_platform.v:2605.10-2643.2
Info:                  /home/pbinkowski/work/litespi/CFU-Playground/third_party/python/pythondata_cpu_vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv_SlimPerfCfu.v:957.23-957.63
Info:  0.3 21.8  Source VexRiscv.RegFilePlugin_regFile.7.0.0$lutram_comb[1]$.F
Info:  0.7 22.5    Net VexRiscv.RegFilePlugin_regFile.7.0.0_DO[1] budget 1.364000 ns (60,14) -> (61,14)
Info:                Sink VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z_2$widefn_comb[1]$.D
Info:                Defined in:
Info:                  /usr/local/bin/../share/yosys/techmap.v:571.28-571.29
Info:  0.3 22.8  Source VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z_2$widefn_comb[1]$.F
Info:  0.0 22.8    Net VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z_2$widefn_f1$ budget 1.364000 ns (61,14) -> (61,14)
Info:                Sink VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z_2$widefn_comb[0]$.F1
Info:  0.1 22.9  Source VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z_2$widefn_comb[0]$.OFX
Info:  0.2 23.1    Net VexRiscv._zz_246_FD1P3IX_Q_D[1] budget 1.364000 ns (61,14) -> (61,14)
Info:                Sink VexRiscv._zz_246_FD1P3IX_Q_2.M
Info:                Defined in:
Info:                  /usr/local/bin/../share/yosys/techmap.v:575.21-575.22
Info:  0.0 23.1  Setup VexRiscv._zz_246_FD1P3IX_Q_2.M
Info: 2.5 ns logic, 20.7 ns routing
```
And the two referenced lines are here https://github.com/tcal-x/pythondata-cpu-vexriscv/blob/7490bb269e39e0be9c7192d3b82891eccfa36db2/pythondata_cpu_vexriscv/verilog/VexRiscv_PerfCfu.v#L707 and here https://github.com/tcal-x/pythondata-cpu-vexriscv/blob/7490bb269e39e0be9c7192d3b82891eccfa36db2/pythondata_cpu_vexriscv/verilog/VexRiscv_PerfCfu.v#L957

Which seems to be a path from instruction decoding to writeback stage. I will need to take a closer look.

I've also checked with a simpler CFU from `example_cfu` and in that case the critical path goes through logic generated by `cfu.py`

```
Info: Critical path report for clock 'por_clk$glb_clk' (posedge -> posedge):
Info: curr total
Info:  0.4  0.4  Source VexRiscv.decode_to_execute_RS1_FD1P3IX_Q_27.Q
Info:  1.9  2.3    Net soc_vexriscv_cfu_bus_cmd_payload_inputs_0[4] budget 2.495000 ns (43,14) -> (53,15)
Info:                Sink Cfu.fn3.fsm_state_FD1P3IX_Q_D_LUT4_Z_C_LUT4_D_C_LUT4_Z_C_CCU2_S0_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT$ccu2_comb[0]$.B
Info:                Defined in:
Info:                  /home/pbinkowski/work/litespi/CFU-Playground/soc/build/hps.example_cfu/gateware/hps_proto2_platform.v:57.13-57.54
Info:  0.4  2.7  Source Cfu.fn3.fsm_state_FD1P3IX_Q_D_LUT4_Z_C_LUT4_D_C_LUT4_Z_C_CCU2_S0_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_COUT_CIN_CCU2_ [... cut ...]
Info:  0.0  0.0  Source VexRiscv.execute_MUL_LL_MULT18X18_Z/PREADD9_CORE_x0_z1.BLSO8
Info:  0.0  0.0    Net VexRiscv.execute_MUL_LL_MULT18X18_Z/PREADD9_CORE_x0_z1$conn$BLSO8 budget 82.176003 ns (55,19) -> (55,19)
Info:                Sink VexRiscv.execute_MUL_LL_MULT18X18_Z/PREADD9_CORE_x0_z0.BLS18
Info:  0.7  0.7  Setup VexRiscv.execute_MUL_LL_MULT18X18_Z/PREADD9_CORE_x0_z0.BLS18
Info: 0.7 ns logic, 0.0 ns routing

Info: Max frequency for clock 'por_clk$glb_clk': 76.15 MHz (PASS at 41.25 MHz)
```

alanvgreen, 2021-09-22:
@piotr-binkowski Thanks for gathering all this together. It looks like we'll need to look closely at more than one path (via the nextpnr JSON output?) to get a full picture of what is going on.

piotr-binkowski, 2021-09-23:
I've generated JSON timing reports and with help of a simple python script (Currently it is located on this branch https://github.com/antmicro/CFU-Playground/tree/hps-fmax)
It can be used to extract list of n longest paths (by default 100)
Results for `hps_accel` look like this:
```
VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_2 : 23.148000717163086
VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_1 : 22.867000579833984
VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q : 22.81599998474121
VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_3 : 22.658000946044922
VexRiscv.dataCache_1.io_cpu_execute_args_wr_LUT4_A_Z_LUT4_B_Z_LUT4_D_Z_LUT4_Z_1 -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_29 : 21.572999954223633
VexRiscv.dataCache_1.io_cpu_execute_args_wr_LUT4_A_Z_LUT4_B_Z_LUT4_D -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_26 : 21.45800018310547
VexRiscv._zz_246_FD1P3IX_Q_9_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_10 : 21.367000579833984
VexRiscv._zz_246_FD1P3IX_Q_9_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_8 : 21.354000091552734
VexRiscv._zz_246_FD1P3IX_Q_9_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_11 : 21.19300079345703
VexRiscv._zz_246_FD1P3IX_Q_9_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_9 : 21.136999130249023
VexRiscv.dataCache_1.io_cpu_execute_args_wr_LUT4_A_Z_LUT4_B_Z_LUT4_D_Z_LUT4_Z_24 -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_30 : 20.993999481201172
VexRiscv.dataCache_1.io_cpu_execute_args_wr_LUT4_A_Z_LUT4_B_Z_LUT4_D_Z_LUT4_Z_16 -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_25 : 20.801000595092773
VexRiscv.dataCache_1.io_cpu_execute_args_wr_LUT4_A_Z_LUT4_B_Z_LUT4_D_Z_LUT4_Z_18 -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_28 : 20.65999984741211
VexRiscv.execute_arbitration_isValid_FD1P3IX_Q_SP_LUT4_Z -> VexRiscv.execute_arbitration_isValid_FD1P3IX_Q : 20.6299991607666
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_30 : 20.604999542236328
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_code_FD1P3IX_Q_1 : 20.604999542236328
VexRiscv.dataCache_1.io_cpu_execute_args_wr_LUT4_A_Z_LUT4_B_Z_LUT4_D_Z_LUT4_Z_17 -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_27 : 20.60300064086914
VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_code_FD1P3IX_Q_CD_LUT4_Z -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_code_FD1P3IX_Q : 20.59600067138672
VexRiscv._zz_246_FD1P3IX_Q_19_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_16 : 20.577999114990234
VexRiscv._zz_246_FD1P3IX_Q_19_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_19 : 20.479000091552734
VexRiscv.IBusCachedPlugin_fetchPc_pcRegPropagate_LUT4_Z -> VexRiscv.IBusCachedPlugin_cache.io_cpu_fetch_data_regNextWhen_FD1P3IX_Q_21 : 20.437999725341797
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_26 : 20.434999465942383
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_28 : 20.434999465942383
VexRiscv.IBusCachedPlugin_fetchPc_pcRegPropagate_LUT4_Z -> VexRiscv.IBusCachedPlugin_cache.io_cpu_fetch_data_regNextWhen_FD1P3IX_Q_13 : 20.395000457763672
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_code_FD1P3IX_Q_2 : 20.395000457763672
VexRiscv.IBusCachedPlugin_fetchPc_pcRegPropagate_LUT4_Z -> VexRiscv.IBusCachedPlugin_cache.io_cpu_fetch_data_regNextWhen_FD1P3IX_Q_12 : 20.392000198364258
VexRiscv.IBusCachedPlugin_fetchPc_pcRegPropagate_LUT4_Z -> VexRiscv.IBusCachedPlugin_cache.io_cpu_fetch_data_regNextWhen_FD1P3IX_Q_14 : 20.392000198364258
VexRiscv.IBusCachedPlugin_fetchPc_pcRegPropagate_LUT4_Z -> VexRiscv.IBusCachedPlugin_cache.io_cpu_fetch_data_regNextWhen_FD1P3IX_Q_17 : 20.392000198364258
VexRiscv.IBusCachedPlugin_fetchPc_pcRegPropagate_LUT4_Z -> VexRiscv.IBusCachedPlugin_cache.io_cpu_fetch_data_regNextWhen_FD1P3IX_Q_19 : 20.392000198364258
VexRiscv.IBusCachedPlugin_fetchPc_pcRegPropagate_LUT4_Z -> VexRiscv.IBusCachedPlugin_cache.io_cpu_fetch_data_regNextWhen_FD1P3IX_Q_5 : 20.389999389648438
VexRiscv._zz_246_FD1P3IX_Q_19_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_18 : 20.332000732421875
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_25 : 20.29199981689453
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_1 : 20.290000915527344
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_29 : 20.283000946044922
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_31 : 20.283000946044922
VexRiscv._zz_304_LUT4_D_Z_WIDEFN9_D0_Z_LUT4_C_Z_LUT4_C -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_27 : 20.281999588012695
[... cut ...]
`_zz_246` is defined [here](https://github.com/tcal-x/pythondata-cpu-vexriscv/blob/7490bb269e39e0be9c7192d3b82891eccfa36db2/pythondata_cpu_vexriscv/verilog/VexRiscv_PerfCfu.v#L1722) and it looks like it is the register file read port
`_zz_304` is defined [here](https://github.com/tcal-x/pythondata-cpu-vexriscv/blob/7490bb269e39e0be9c7192d3b82891eccfa36db2/pythondata_cpu_vexriscv/verilog/VexRiscv_PerfCfu.v#L2631) and it is an `IS_CSR` signal
and `_zz_245` looks like second read port of the register file.

This all looks to be related to instruction decoding.

piotr-binkowski, 2021-09-23:
And for comparison `example_cfu` report looks like this
```
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q : 12.696999549865723
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_14_FD1P3IX_Q : 12.633999824523926
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_17_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_16 : 12.597000122070312
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_17_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_14 : 12.583000183105469
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_1 : 12.574999809265137
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_17_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_15 : 12.572999954223633
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_13_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_12 : 12.569000244140625
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_13_FD1P3IX_Q : 12.553999900817871
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_5_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_2 : 12.520000457763672
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_5_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_4 : 12.486000061035156
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_20_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_20 : 12.472000122070312
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_13_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_11 : 12.472000122070312
VexRiscv._zz_245_FD1P3IX_Q_7_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_7 : 12.454000473022461
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_17_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_17 : 12.432999610900879
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_13_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_10 : 12.388999938964844
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_9_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_8 : 12.38700008392334
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_20_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_19 : 12.345000267028809
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_5_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_3 : 12.34000015258789
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_5_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_5 : 12.317000389099121
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_20_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_18 : 12.300999641418457
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_9_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_6 : 12.298999786376953
VexRiscv._zz_245_FD1P3IX_Q_7_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_4 : 12.270999908447266
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_9_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_7 : 12.260000228881836
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_9_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_9 : 12.258000373840332
VexRiscv._zz_245_FD1P3IX_Q_7_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_6 : 12.246000289916992
VexRiscv._zz_245_FD1P3IX_Q_9_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_9 : 12.208999633789062
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_13_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_13 : 12.166000366210938
VexRiscv._zz_245_FD1P3IX_Q_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_3 : 12.131999969482422
VexRiscv._zz_245_FD1P3IX_Q_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q : 12.09000015258789
VexRiscv._zz_245_FD1P3IX_Q_7_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_5 : 12.086999893188477
VexRiscv._zz_245_FD1P3IX_Q_9_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_11 : 12.050000190734863
VexRiscv._zz_246_FD1P3IX_Q_7_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_4 : 12.010000228881836
VexRiscv._zz_246_FD1P3IX_Q_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_1 : 12.006999969482422
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_21 : 12.003999710083008
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_23 : 12.003999710083008
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_20 : 12.003000259399414
VexRiscv._zz_245_FD1P3IX_Q_9_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_8 : 11.982000350952148
VexRiscv._zz_246_FD1P3IX_Q_7_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv._zz_246_FD1P3IX_Q_5 : 11.973999977111816
[... cut ...]
vns_interface4_bank_bus_dat_r_FD1P3IX_Q_D_WIDEFN9_Z$widefn_comb[0]$ -> vns_interface4_bank_bus_dat_r_FD1P3IX_Q : 11.46500015258789
VexRiscv._zz_245_FD1P3IX_Q_31_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_30 : 11.463000297546387
VexRiscv._zz_245_FD1P3IX_Q_31_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv._zz_245_FD1P3IX_Q_31 : 11.456000328063965
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_16 : 11.444999694824219
```
And most paths here start in data/instruction caches

piotr-binkowski, 2021-09-23:
If you want to do some tests on your side, you can use this script is located here https://github.com/antmicro/CFU-Playground/blob/hps-fmax/scripts/nexpnr-timing.py and to get JSON outputs with you will need this https://github.com/YosysHQ/nextpnr/pull/810 (rebase this on top of current master to get CrosslinkNX DDRTristate support working in nextpnr)

Branch with `nextpnr-timing.py` also contains modified Litex that automatically passes option to nextpnr to generate JSON report (it will be located in `gateware` directory where you can also find your bitstream)

piotr-binkowski, 2021-09-24:
I did another test with `SlimCfu` variant of the VexRiscV and it looks like it can reach 64MHz on some runs (around 65MHz in this one but only 60MHz in one directly before)
```
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_1 : 15.312000274658203
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_3 : 15.312000274658203
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_8 : 15.312000274658203
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3JX_Q : 15.312000274658203
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_15 : 15.303000450134277
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_17 : 15.303000450134277
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_22 : 15.303000450134277
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_23 : 15.303000450134277
VexRiscv._zz_221_FD1P3IX_Q_31_D_LUT4_Z_1 -> VexRiscv._zz_221_FD1P3IX_Q_29 : 15.295999526977539
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_18 : 15.161999702453613
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_19 : 15.161999702453613
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_fetchPc_pcReg_FD1P3IX_Q_20 : 15.161999702453613
VexRiscv._zz_221_FD1P3IX_Q_23_D_LUT4_Z -> VexRiscv._zz_221_FD1P3IX_Q_20 : 14.727999687194824
VexRiscv._zz_222_FD1P3IX_Q_23_D_LUT4_Z_1 -> VexRiscv._zz_222_FD1P3IX_Q_21 : 14.66100025177002
VexRiscv._zz_222_FD1P3IX_Q_31_D_LUT4_Z_3 -> VexRiscv._zz_222_FD1P3IX_Q_31 : 14.654000282287598
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_2 : 14.579000473022461
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_18_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_18 : 14.565999984741211
VexRiscv._zz_222_FD1P3IX_Q_27_D_LUT4_Z -> VexRiscv._zz_222_FD1P3IX_Q_24 : 14.54800033569336
VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionValidsRegs_decode_FD1P3IX_Q_CD_LUT4_Z -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionValidsRegs_decode_FD1P3IX_Q : 14.4399995803833
VexRiscv.dataCache_1.io_cpu_execute_args_wr_LUT4_B_Z_LUT4_D_Z_LUT4_D_Z_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.CsrPlugin_exceptionPortCtrl_exceptionContext_badAddr_FD1P3IX_Q_21 : 14.416999816894531
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q : 14.411999702453613
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_1 : 14.41100025177002
VexRiscv._zz_221_FD1P3IX_Q_27_D_LUT4_Z_2 -> VexRiscv._zz_221_FD1P3IX_Q_26 : 14.376999855041504
VexRiscv._zz_221_FD1P3IX_Q_27_D_LUT4_Z_1 -> VexRiscv._zz_221_FD1P3IX_Q_25 : 14.376999855041504
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_18_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_16 : 14.362000465393066
VexRiscv._zz_221_FD1P3IX_Q_27_D_LUT4_Z_3 -> VexRiscv._zz_221_FD1P3IX_Q_27 : 14.352999687194824
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_14_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_13 : 14.343000411987305
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_6_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_5 : 14.342000007629395
VexRiscv._zz_222_FD1P3IX_Q_7_D_LUT4_Z_2 -> VexRiscv._zz_222_FD1P3IX_Q_6 : 14.331999778747559
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_14_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_11 : 14.331999778747559
VexRiscv._zz_221_FD1P3IX_Q_31_D_LUT4_Z -> VexRiscv._zz_221_FD1P3IX_Q_28 : 14.317999839782715
VexRiscv._zz_221_FD1P3IX_Q_23_D_LUT4_Z_3 -> VexRiscv._zz_221_FD1P3IX_Q_23 : 14.295999526977539
VexRiscv._zz_222_FD1P3IX_Q_7_D_LUT4_Z_3 -> VexRiscv._zz_222_FD1P3IX_Q_7 : 14.265999794006348
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_6_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_4 : 14.253000259399414
VexRiscv._zz_222_FD1P3IX_Q_31_D_LUT4_Z_1 -> VexRiscv._zz_222_FD1P3IX_Q_29 : 14.251999855041504
VexRiscv._zz_222_FD1P3IX_Q_27_D_LUT4_Z_1 -> VexRiscv._zz_222_FD1P3IX_Q_25 : 14.25100040435791
VexRiscv._zz_222_FD1P3IX_Q_31_D_LUT4_Z -> VexRiscv._zz_222_FD1P3IX_Q_28 : 14.246999740600586
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_14_D_WIDEFN9_Z_2$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_12 : 14.215999603271484
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_18_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_15 : 14.192000389099121
VexRiscv._zz_222_FD1P3IX_Q_23_D_LUT4_Z_3 -> VexRiscv._zz_222_FD1P3IX_Q_23 : 14.184000015258789
VexRiscv.IBusCachedPlugin_cache._zz_13_FD1P3IX_Q_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_13_FD1P3IX_Q : 14.166000366210938
VexRiscv._zz_222_FD1P3IX_Q_31_D_LUT4_Z_2 -> VexRiscv._zz_222_FD1P3IX_Q_30 : 14.156999588012695
Cfu.macc.$199_CCU2_S0_4$ccu2_comb[0]$ -> Cfu.macc.result__payload_FD1P3IX_Q : 14.152999877929688
[... cut ...]
VexRiscv._zz_221_FD1P3IX_Q_7_D_LUT4_Z_1 -> VexRiscv._zz_221_FD1P3IX_Q_5 : 13.779999732971191
VexRiscv._zz_222_FD1P3IX_Q_9_D_LUT4_Z_3 -> VexRiscv._zz_222_FD1P3IX_Q_11 : 13.77400016784668
VexRiscv.IBusCachedPlugin_fetchPc_booted_LUT4_C -> VexRiscv.IBusCachedPlugin_cache.io_cpu_fetch_mmuRsp_isIoAccess_FD1P3IX_Q : 13.77400016784668
```
And from what I can see the only difference between SlimPerfCfu and SlimCfu is including additional performance counter CSRs which look unused in the `hps_accel` code (there are a few tests for those in the menus but thats all)

alanvgreen, 2021-09-24:
@piotr-binkowski The CSRs are typically used to tune code and then are removed before checking in.

Agree that if they're reducing fMax we probably don't want them included by default.

@tcal-x if there is a mechanism to turn CSRs on and off, that mechanism should toggle fMax up and down, too. For the time being, I'd be happy to use MCYCLE on hps_accel.

piotr-binkowski, 2021-09-29:
I did some more tests and I've noticed that in the CFU `rsp_valid` uses `cmd_valid` input from the Vex directly which creates a fairly long path which in the end is used by `memory_arbitration_isStuck` signals and this is then used by instruction/data caches that were showing issues.
I've modified the CFU FSM responsible for communication with the CPU to use only signals registered inside CFU, this is located [here](https://github.com/antmicro/CFU-Playground/commit/64d952bb3c0ac1f46e246405e13b9bf35256bfc1)
But at the moment that code fails on real hardware - golden tests show different results ie. 
```
   490M (   490056313) cycles total
FAIL -123 (actual) != -124 (expected) ***
```
But at the same time code passes all tests (executed by `make pytest` from `proj/hps_accel` dir).

Using that version of the code and a good seed (`7811` in this case) I was able to build a design that almost reached 70.4MHz (64MHz + 10%) - it reached 69.7MHz

And in fact only three paths were too slow to meet the timing requirements, remaining paths were under 14ns
```
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_14_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_11 : 14.35099983215332
VexRiscv.execute_CsrPlugin_csr_3008_LUT4_C -> VexRiscv._zz_159_FD1P3IX_Q_30 : 14.20300006866455
VexRiscv.execute_CsrPlugin_csr_3008_LUT4_C -> VexRiscv._zz_159_FD1P3IX_Q_8 : 14.20300006866455
VexRiscv.IBusCachedPlugin_cache._zz_13_FD1P3IX_Q_D_WIDEFN9_Z$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_13_FD1P3IX_Q : 13.814000129699707
VexRiscv.IBusCachedPlugin_cache._zz_13_FD1P3IX_Q_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_14_FD1P3IX_Q : 13.803000450134277
VexRiscv.IBusCachedPlugin_cache._zz_13_FD1P3IX_Q_D_WIDEFN9_Z_3$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_19 : 13.706999778747559
VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_14_D_WIDEFN9_Z_1$widefn_comb[0]$ -> VexRiscv.IBusCachedPlugin_cache._zz_10_FD1P3IX_Q_13 : 13.559000015258789
...
```

tcal-x, 2021-09-29:
> I did some more tests and I've noticed that in the CFU `rsp_valid` uses `cmd_valid` input from the Vex directly which creates a fairly long path...

There may be some options in the generation of the VexRiscv that affect how the CFU Plugin interfaces with the CPU pipeline.   This will be come easier to experiment with after PR #305 is merged.

piotr-binkowski, 2021-10-01:
With some more tinkering I've managed to get safely above 70MHz (This run used seed `3107`)
```
Info: Max frequency for clock 'por_clk$glb_clk': 76.13 MHz (PASS at 70.72 MHz)
```
Currently performance seems to vary greatly based on used seed - with the same sources I saw runs that reached only around 58MHz.

I've achieved that by using my previous change to the CFU (so that `rsp_valid` is no longer dependent on `cmd_valid` input), and then building a custom Vex SlimCfu config with 4KiB of ICache (by default it comes with 2KiB of ICache).

Increasing cache size removed 1 bit from cache tags (`zz_10` and `zz_13` signals carry ICache tags) which helped with those remaining long paths.

Finally, this is how current resource usage looks like:
```
14:49:28 Info: Device utilisation:
14:49:28 Info:              OXIDE_FF:  4111/13824    29% 
14:49:28 Info:            OXIDE_COMB:  8575/13824    62% 
14:49:28 Info:                  RAMW:    36/ 1728     2%  
14:49:28 Info:           SEIO33_CORE:     7/   23    30% 
14:49:28 Info:           SEIO18_CORE:     2/   44     4%  
14:49:28 Info:         DIFFIO18_CORE:     0/   22     0%  
14:49:28 Info:              OSC_CORE:     1/    1   100%
14:49:28 Info:             OXIDE_EBR:    16/   24    66% 
14:49:28 Info:          PREADD9_CORE:    24/   48    50% 
14:49:28 Info:            MULT9_CORE:    24/   48    50% 
14:49:28 Info:           MULT18_CORE:     4/   24    16% 
14:49:28 Info:            REG18_CORE:    24/   48    50% 
14:49:28 Info:        MULT18X36_CORE:     0/   12     0%  
14:49:28 Info:           MULT36_CORE:     0/    6     0%  
14:49:28 Info:            ACC54_CORE:     0/   12     0%  
14:49:28 Info:                   DCC:     1/   62     1%  
14:49:28 Info:               VCC_DRV:     1/   74     1%  
14:49:28 Info:             LRAM_CORE:     5/    5   100%
14:49:28 Info:               IOLOGIC:     0/   44     0%  
14:49:28 Info:              SIOLOGIC:     5/   22    22%
```
And with 2KiB cache it looks like this
```
15:37:35 Info: Device utilisation:
15:37:35 Info:              OXIDE_FF:  4133/13824    29% 
15:37:35 Info:            OXIDE_COMB:  8845/13824    63% 
15:37:35 Info:                  RAMW:    60/ 1728     3%  
15:37:35 Info:           SEIO33_CORE:     7/   23    30% 
15:37:36 Info:           SEIO18_CORE:     2/   44     4%  
15:37:36 Info:         DIFFIO18_CORE:     0/   22     0%  
15:37:36 Info:              OSC_CORE:     1/    1   100%
15:37:36 Info:             OXIDE_EBR:    14/   24    58% 
15:37:36 Info:          PREADD9_CORE:    24/   48    50% 
15:37:36 Info:            MULT9_CORE:    24/   48    50% 
15:37:36 Info:           MULT18_CORE:     4/   24    16% 
15:37:36 Info:            REG18_CORE:    24/   48    50% 
15:37:36 Info:        MULT18X36_CORE:     0/   12     0%  
15:37:36 Info:           MULT36_CORE:     0/    6     0%  
15:37:36 Info:            ACC54_CORE:     0/   12     0%  
15:37:36 Info:                   DCC:     1/   62     1%  
15:37:36 Info:               VCC_DRV:     1/   74     1%  
15:37:36 Info:             LRAM_CORE:     5/    5   100%
15:37:36 Info:               IOLOGIC:     0/   44     0%  
15:37:36 Info:              SIOLOGIC:     5/   22    22%
```
So we need only two more `OXIDE_EBR` primitives but in theory it should be possible to achieve similar results with 2KiB cache by limiting size of the code address space (for example by omitting its MSB in cache which in turn should result in one bit of the tag being optimized out)

piotr-binkowski, 2021-10-01:
And on top of that I've managed to fix the issue I had earlier with the modified Cfu so this is passing `hps_accel` golden tests on real hardware.

piotr-binkowski, 2021-10-01:
For reference, code I'm currently using is located here https://github.com/antmicro/CFU-Playground/tree/hps-fmax

alanvgreen, 2021-10-04:
@piotr-binkowski Thanks for this! Finding the critical path depends on cache size is really interesting?

When we will we be able to upstream critical path reports to CFU-Playground?

Also what tools are you using to parse and analyze the JSON?

piotr-binkowski, 2021-10-05:
@alanvgreen I've created a PR with my changes in #314, this also includes the scripts for parsing the JSONs

Critical path reporting was upstreamed in nextpnr in YosysHQ/nextpnr#810 and my PR mentioned above uses JSON output from mainline nextpnr

mithro, 2021-10-05:
@piotr-binkowski - FWIW these should separate pull requests, having multiple things in one pull request makes it hard to revert only one change if trouble occurs.

piotr-binkowski, 2021-10-06:
And a small update on fMax - using code from #317 and Vex_SlimCfu rebuilt with 4KiB of ICache I was able to reach a bit over 84MHz so 75MHz builds are possible (those require 82.5MHz to include +/-10% oscillator accuracy)
```
Info: Max frequency for clock 'por_clk$glb_clk': 84.15 MHz (PASS at 70.72 MHz)
```