google/CFU-Playground #293: Verilator-Renode simulation issue with proj/kws_micro_accel

tcal-x, 2021-09-22:
@robertszczepanski  --

In proj/kws_micro_accel, there are CFU source files cfu.sv, mac.sv, rdh.sv, and rcdbpot.sv.    cfu.sv includes the other three.

When running `make renode`, I see this error:
```
%Error: /home/tim/tcal-x/CFU-Playground/proj/kws_micro_accel/cfu.sv:15: Cannot find include file: mac.sv
`include "mac.sv"
         ^~~~~~~~
%Error: /home/tim/tcal-x/CFU-Playground/proj/kws_micro_accel/cfu.sv:15: This may be because there's no search path specified with -I<dir>.
`include "mac.sv"
         ^~~~~~~~
```

I think the build is happening in some directory (./build/renode/?).    According to the error comments, it should be a pretty easy fix.

robertszczepanski, 2021-09-22:
@tcal-x 
Did you have to change `-DVTOP="$(PROJ_DIR)/cfu.v"` to `-DVTOP="$(PROJ_DIR)/cfu.sv"` or were you able to run this without any changes? https://github.com/google/CFU-Playground/blob/main/proj/proj.mk#L182

I can see a way to solve this but I had to apply a change like the one above so I just want to confirm that it's not my issue locally. Also is there any specific reason to keep sources as SystemVerilog instead of Verilog?

Do we want to make `proj.mk` more generic to accept both `.v` and `.sv` or do we want to stick with one extension for all projects? If we want it more generic, which one is higher in hierarchy (which one should be looked for first)?

tcal-x, 2021-09-22:
I didn't make any changes locally.

There is logic in proj.mk that gives preference to cfu.sv if it exists:
```
CFU_VERILOG     := $(if $(wildcard $(PROJ_DIR)/cfu.sv), $(PROJ_DIR)/cfu.sv, $(PROJ_DIR)/cfu.v)
```

PR #301 (2021-10-06): Fix building verilated CFU for multiple sources
Fixes #293

Verilator will now treat project directory as an include directory and verilog source path is taken from `$(CFU_VERILOG)` variable. There was also a typo in this variable that created a path with additional space at the beginning (` /path/to/file` instead of `/path/to/file`),

FYI @tcal-x
closed 2021-10-06 by tcal-x