google/CFU-Playground #306: Oxide failure on hps_accel

alanvgreen, 2021-09-29:
My latest build failed with the following error

```
08:43:43 Info: constraining clock net 'por_clk' to 41.25 MHz
08:43:43 Info: Constraining LVCMOS18H IO 'user_led0_OB_O' to pin G3 (PCLKT3_2/ADC_CP13; bel X66/Y29/PIOA)
08:43:43 Info: Constraining LVCMOS18 IO 'spiflash4x_dq_BB_B_3' to pin B5 (MOSI/MD0; bel X61/Y0/PIOB)
08:43:43 Info: Constraining LVCMOS18 IO 'spiflash4x_dq_BB_B_2' to pin C4 (MISO/MD1; bel X63/Y0/PIOA)
08:43:43 Info: Constraining LVCMOS18 IO 'spiflash4x_dq_BB_B_1' to pin B3 (MD2; bel X63/Y0/PIOB)
08:43:43 Info: Constraining LVCMOS18 IO 'spiflash4x_dq_BB_B' to pin B2 (MD3; bel X65/Y0/PIOA)
08:43:43 Info: Constraining LVCMOS18 IO 'spiflash4x_cs_n_OB_O' to pin A3 (MCSN/PCLKT0_1; bel X61/Y0/PIOA)
08:43:43 Info: Constraining LVCMOS18 IO 'spiflash4x_clk_OB_O' to pin B4 (MCLK/PCLKT0_0; bel X59/Y0/PIOA)
08:43:43 Info: Constraining LVCMOS18H IO 'serial_tx_OB_O' to pin G1 (LRC_GPLL0C_IN/VREF3_2; bel X70/Y29/PIOB)
08:43:43 Info: Constraining LVCMOS18 IO 'serial_rx_IB_I' to pin E2 (EIO; bel X75/Y9/PIOB)
08:43:43 Info: Packing IOLOGIC...
08:43:43 Info: Packing DSPs...
08:43:43 Info: Packing BRAM...
08:43:43 Info:     Created 20 OXIDE_EBR cells from:
08:43:43 Info:             20x PDPSC16K_MODE
08:43:43 Info: Packing LRAM...
08:43:43 Info:     Created 5 LRAM_CORE cells from:
08:43:43 Info:              5x SP512K_MODE
08:43:43 Info: Packing carries...
08:43:43 terminate called after throwing an instance of 'nextpnr_nexus::assertion_failure'
08:43:43 what():  Assertion failure: fco->users.at(0).port == id_CIN (/usr/local/google/home/avg/src/CFU-Playground/third_party/nextpnr/nexus/pack.cc:1364)
```

alanvgreen, 2021-09-29:
```
$ nextpnr-nexus --version
nextpnr-nexus -- Next Generation Place and Route (Version bd137a8b)
```

The command line would have been something like

```
nextpnr-nexus \
        --json "${JSON}" \
        --pdc "${PDC}" \
        --fasm "${DIR}/output.fasm" \
        --device LIFCL-17-8UWG72C \
        --seed 2672
```

I have saved a copy of the input JSON (~16MB)

alanvgreen, 2021-09-30:
I tried using the same versions of yosys, nextpnr and prjoxide as are in the chromeos ebuilds, but seem to have the same errors.

```
  git -C yosys pull --ff-only origin 1cac671c70bc3da9808ceb3add15686da4a5d82e
  git -C yosys submodule update
  git -C prjoxide pull --ff-only origin 9782a46a9b4edb24cc73ea65f1ad7697ed74ab41
  git -C prjoxide submodule update
  git -C nextpnr pull --ff-only origin 318331f8b30c2e2a31cc41d51f104b671e180a8a
  git -C nextpnr submodule update
```

danc86, 2021-09-30:
I still don't know what an FCO is, but I noticed that the assertion refers to `fco` so I wondered if this might be a slightly different manifestation of the `multiple fanout on FCO` problem mentioned here?
https://github.com/google/CFU-Playground/issues/246#issuecomment-927883755

tcal-x, 2021-09-30:
> I still don't know what an FCO is, but I noticed that the assertion refers to `fco` so I wondered if this might be a slightly different manifestation of the `multiple fanout on FCO` problem mentioned here? [#246 (comment)](https://github.com/google/CFU-Playground/issues/246#issuecomment-927883755)

It did look similar, but I just tried rerunning `yosys`+`nextpnr-nexus`, removing the `-abc9` option from the `yosys` invocation, and I still saw the assertion failure.

tcal-x, 2021-09-30:
(deleted the previous comment I made)

> I still don't know what an FCO is, but I noticed that the assertion refers to `fco` so I wondered if this might be a slightly different manifestation of the `multiple fanout on FCO` problem mentioned here? [#246 (comment)](https://github.com/google/CFU-Playground/issues/246#issuecomment-927883755)

That is a good theory, and I did confirm that if I remove the `-abc9` option from the `yosys` invocation, then I don't see the assertion failure.

danc86, 2021-09-30:
FYI in PR #307 I included a workaround to avoid abc9 for now, so that we can keep working.

danc86, 2021-09-30:
Unfortunately switching from ABC9 back to ABC seems to make the runtime for router1 quite a bit longer. Not sure if this is because ABC9 was doing better optimisations before the routing step, or it could just be in my imagination.

tcal-x, 2021-11-14:
The ABC9 issue has been fixed with https://github.com/berkeley-abc/abc/pull/139 and https://github.com/YosysHQ/yosys/commit/48a628522b9c063ee0312b84737b51a8ae1948d4.

closed 2021-11-14 by tcal-x