google/CFU-Playground #312: about update to latest, which picolibc in project tree

akioolin, 2021-10-04:
Hi, 

The project I test is mnv2_first. the build command is as following.

source /opt/Xilinx/Vivado/2020.1/settings64.sh
make V=1 \
PLATFORM=common_soc \
TARGET=qmtech_wukong \
USE_VIVADO=1 \
UART_SPEED=3000000 \
TTY=/dev/ttyUSB0 \
USE_VIDEO_FRAMEBUFFER=1 \
IGNORE_TIMING=0 \
bitstream prog load

After update to picolibc enable, the error which show is console.h, uart.h.
these errors could be add "-I$(SRC_DIR)/playground_util" and 
"-I$(LITEX_HW_DIR)/base" in SHARED_FLAGS of common/Makefile. 

the error follow the makefile update is as below.

opt/CFU-Playground/proj/mnv2_first/build/src/stdio.c: In function 'readchar':
/opt/CFU-Playground/proj/mnv2_first/build/src/stdio.c:30:17: error: implicit declaration of function 'getchar' [-Werror=implicit-function-declaration]
   30 |   return (char) getchar();
      |                 ^~~~~~~
/opt/CFU-Playground/proj/mnv2_first/build/src/stdio.c:18:1: note: 'getchar' is defined in header '<stdio.h>'; did you forget to '#include <stdio.h>'?
   17 | #include <stdio.h>
  +++ |+#include <stdio.h>
   18 | 
cc1: all warnings being treated as errors

Does any one meet the following error after update to picolibc?

BR, Akio

tcal-x, 2021-10-04:
@akioolin , thank you for writing up the issue.

I have not been able to reproduce a problem using your command, but let's figure out what is causing the error that you see.

Let's start with some simple things to check:
* Try running  `./scripts/setup` again
* In the command above, change `bitstream prog load` to `clean bitstream prog load` to make sure we're not mixing old and new files in the build directories.

Another thing to check is the RISC-V toolchain.  Which are you using?   I tested with
```
riscv64-unknown-elf-gcc (SiFive GCC 8.3.0-2020.04.1) 8.3.0
```
Thanks,
Tim

akioolin, 2021-10-05:
Hi, Tim:

Many Thanks. after doing what Tim said, the errors is gone. 
the gcc which I used is SiFive GCC 10.1.0-2020.08.2.
this version of GCC is work fine for me.

I think the root cause is the version of third_party package doesn't update. 

BR, Akio

closed 2021-10-06 by akioolin