google/CFU-Playground #344: Failure during yosys execution for hps_accel

alanvgreen, 2021-11-19:
Attempting to generate a bitstream for hps results in an Assertion failure.

```
$ cd proj/hps_accel
$ make bitstream
```

produces this error:

```
5.63. Executing JSON backend.

Warnings: 4 unique messages, 4 total
End of script. Logfile hash: ec3b35b2e5, CPU: user 37.76s system 0.42s, MEM: 478.79 MB peak
Yosys 0.10+60 (git sha1 5f00bf2d, clang 11.1.0-4 -fPIC -Os)
Time spent: 24% 1x abc9_exe (12 sec), 13% 18x techmap (6 sec), ...
Info: constraining clock net 'por_clk' to 70.72 MHz
Info: Constraining LVCMOS18H IO 'user_led0_OB_O' to pin G3 (PCLKT3_2/ADC_CP13; bel X66/Y29/PIOA)
Info: Constraining LVCMOS18 IO 'spiflash4x_dq_BB_B_3' to pin B5 (MOSI/MD0; bel X61/Y0/PIOB)
Info: Constraining LVCMOS18 IO 'spiflash4x_dq_BB_B_2' to pin C4 (MISO/MD1; bel X63/Y0/PIOA)
Info: Constraining LVCMOS18 IO 'spiflash4x_dq_BB_B_1' to pin B3 (MD2; bel X63/Y0/PIOB)
Info: Constraining LVCMOS18 IO 'spiflash4x_dq_BB_B' to pin B2 (MD3; bel X65/Y0/PIOA)
Info: Constraining LVCMOS18 IO 'spiflash4x_cs_n_OB_O' to pin A3 (MCSN/PCLKT0_1; bel X61/Y0/PIOA)
Info: Constraining LVCMOS18 IO 'spiflash4x_clk_OB_O' to pin B4 (MCLK/PCLKT0_0; bel X59/Y0/PIOA)
Info: Constraining LVCMOS18H IO 'serial_tx_OB_O' to pin G1 (LRC_GPLL0C_IN/VREF3_2; bel X70/Y29/PIOB)
Info: Constraining LVCMOS18 IO 'serial_rx_IB_I' to pin E2 (EIO; bel X75/Y9/PIOB)
Info: Packing IOLOGIC...
Info:     Created 9 IOLOGIC cells from:
Info:              4x IDDRX1
Info:              5x ODDRX1
Info: Constraining IOLOGIC ODDRX1 to bel X59/Y0/SIOLOGICA
Info: Constraining IOLOGIC ODDRX1_1 to bel X61/Y0/SIOLOGICB
Info: Constraining IOLOGIC ODDRX1_2 to bel X63/Y0/SIOLOGICA
Info: Constraining IOLOGIC ODDRX1_3 to bel X63/Y0/SIOLOGICB
Info: Constraining IOLOGIC ODDRX1_4 to bel X65/Y0/SIOLOGICA
Info: Packing DSPs...
Info: Packing BRAM...
Info:     Created 16 OXIDE_EBR cells from:
Info:             16x PDPSC16K_MODE
Info: Packing LRAM...
Info:     Created 5 LRAM_CORE cells from:
Info:              5x SP512K_MODE
Info: Packing carries...
terminate called after throwing an instance of 'nextpnr_nexus::assertion_failure'
  what():  Assertion failure: fco->users.at(0).port == id_CIN (/usr/local/google/home/avg/src/CFU-Playground/third_party/nextpnr/nexus/pack.cc:1364)
build_hps_proto2_platform.sh: line 4: 3118923 Aborted                 nextpnr-nexus --json hps_proto2_platform.json --pdc hps_proto2_platform.pdc --fasm hps_proto2_platform.fasm --report=hps_proto2_platform-report.json --detailed-timing-report --device LIFCL-17-8UWG72C --seed 1 --router router1
Traceback (most recent call last):
  File "/usr/local/google/home/avg/src/CFU-Playground/soc/./hps_soc.py", line 298, in <module>
    main()
  File "/usr/local/google/home/avg/src/CFU-Playground/soc/./hps_soc.py", line 290, in main
    vns = builder.build(**builder_kwargs, run=args.build)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 315, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/soc/integration/soc.py", line 1131, in build
    return self.platform.build(self, *args, **kwargs)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/build/lattice/platform.py", line 43, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/build/lattice/oxide.py", line 169, in build
    _run_script(script)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/build/lattice/oxide.py", line 104, in _run_script
    raise OSError("Error occured during Yosys/Nextpnr's script execution.")
OSError: Error occured during Yosys/Nextpnr's script execution.
make[1]: *** [/usr/local/google/home/avg/src/CFU-Playground/soc/hps.mk:90: build/hps.hps_accel/gateware/hps_proto2_platform.bit] Error 1
make[1]: Leaving directory '/usr/local/google/home/avg/src/CFU-Playground/soc'
make: *** [../proj.mk:285: bitstream] Error 2
```

danc86, 2021-11-19:
PR #337 turned on ABC9 again after the carry-in/carry-out problem was fixed. But your yosys version 5f00bf2d is slightly too old to have the fix. If you run `scripts/install_oxide` does it give you a fresher version of yosys?

alanvgreen, 2021-11-19:
I just reran install_oxide. I'll take a closer look.

danc86, 2021-11-19:
I'm not totally sure but I think the yosys commits you need are one of these:

```
commit 48a628522b9c063ee0312b84737b51a8ae1948d4
Author: Miodrag Milanovic <mmicko@gmail.com>
Date:   Fri Nov 12 12:40:24 2021 +0100

    update abc

commit b4f68e3ccae0cea606f223b67f3580e311e36578
Author: Miodrag Milanovic <mmicko@gmail.com>
Date:   Fri Nov 12 09:00:32 2021 +0100

    Update abc

```

Commit 5f00bf2d is dated 28 October.

alanvgreen, 2021-11-19:
Thanks! I'm running `scripts/install_oxide update`

tcal-x, 2021-11-19:
I'm worried it won't be sufficient.   If it doesn't work, then go into `third_party/yosys` and `/bin/rm -r abc` (yes, just remove it), and then `cd` back up to the repo root and rerun `scripts/install_oxide update`.

I apologize for not giving you more of a heads up.

alanvgreen, 2021-11-19:
Successfully generated the bitstream after running `scripts/install_oxide update`.

@danc86 Thanks for the pointer to the version!
@tcal-x No worries. I ought to know the drill by now :)

closed 2021-11-19 by alanvgreen