google/CFU-Playground #384: gen2 CFU flakiness

alanvgreen, 2021-12-17:
Then gen2 cfu seems to work ~90% of the time. The other 10% of the time it fails in a fairly, but not competely consistent manner.

To reproduce
1. Build with #383 patched
2. Run on proto2, selecting menu option "3" (project menu) then "5" (test layer 05)
3. Keep pressing "5" until experiencing a failure. Typically, between 10 and 50 runs are needed to see the first failure.

Usually the failure report "8576 differences", though on some proto2 boards, other numbers of failures are sometimes seen. 

Observations:

In the most common form of failure, examining the first 32 words shows that many of the failures result from output appearing exactly 4 words after it should (i.e expected[n] appears at expected[n+4]). However not all of the wrong words exactly follow this pattern.

Building at target different clocks speeds, with different seeds and with/without abc9 did not appear to affect the behaviour.

alanvgreen, 2021-12-17:
Possible next steps:

1. Build with Radiant to isolate whether this is an issue with the toolchain.
2. Ensure output FIFO is empty 
3. Examine code to ensure reset logic properly resets all important state.

alanvgreen, 2021-12-17:
Radiant is not able to synthesize this design - it infers too many multipliers.

Howeve, I was able to reproduce the flakiness using yosys for synthesis and then place and route with Radiant.

alanvgreen, 2021-12-17:
Tried SlimCfu vs currently configured SlimOptCfu - no difference.

tcal-x, 2021-12-17:
I suppose getting Verilator simulation of this system working, and reproducing the issue there, would be significant.

alanvgreen, 2021-12-18:
Yes, a Verilator sim would be useful right now. I suppose LiteScope would be helpful too. But I know these things aren't free.

I am currently working on tunneling out a few values via the CFU interface which will give some insight. If I need more I might be able to cobble together something store values in an EBRAM and retrieve later via a cfu instruction.

alanvgreen, 2021-12-18:
I added a new CFU register (REG_ACC_0) that records the first accumulator value calculated in the top left accumulator of the systolic array. This is the accumulator that is used to calculate byte 0 of the output. 

Comparing the value of REG_ACC_0 on hardware compared to the simulator, I see that it is correct when the test passes (0x18cf), and has a wrong value (0x02) when the test fails.

This is evidence that the problem is not in or after the post processing pipeline.

(test was built with yosys + nextpnr)

alanvgreen, 2021-12-18:
Running 1000 itereations of the test in a tight loop causes it to succeed continuously.

tcal-x, 2021-12-18:
I got it working in Verilator.  I'm using full+debug+cfu for various reasons.   After many runs, I finally saw a failure with 05.
```
Running test layer 05
Testing Conv2D Layer 05
word |  output  | expected |
0000 | a087808d | 8080809b | *
0001 | 80808080 | 80808080 | 
0002 | 80808092 | 80808080 | *
0003 | 80808080 | 80808080 | 
0004 | 8080809b | 8080808b | *!
0005 | 80808080 | 80808080 | !
0006 | 80808080 | 80808080 | !
0007 | 80808080 | 80808080 | !
0008 | 8080808b | 80868080 | *!
0009 | 80808080 | 80808080 | !
000a | 80808080 | 80808080 | !
000b | 80808080 | 80808080 | !
000c | 80868080 | 80a28080 | *!
000d | 80808080 | 87808080 | *!
000e | 80808080 | 80808080 | !
000f | 80808080 | 80808080 | !
0010 | 80808080 | 80b78080 | *
0011 | 80808081 | 80808080 | *
0012 | 80808080 | 80808080 | !
0013 | 80828080 | 80808080 | *
0014 | 80b78080 | 80a88080 | *!
0015 | 80808080 | 80809c80 | *!
0016 | 80808080 | 80808080 | !
0017 | 80808080 | 80808b80 | *!
0018 | 80a88080 | 808b8080 | *!
0019 | 80809c80 | 80868380 | *!
001a | 80808080 | 80808080 | !
001b | 80808b80 | 8080a280 | *!
001c | 808b8080 | 80808080 | *!
001d | 80868380 | 80808080 | *!
001e | 80808080 | 80808080 | !
001f | 8080a280 | 80808080 | *!
FAIL - 8576 differences
---
```

tcal-x, 2021-12-19:
I got a second bad run in Verilator (out of about 15 tries).   It was the same as above except for the first output (0000):
```
Running test layer 05
Testing Conv2D Layer 05
word |  output  | expected |
0000 | a0878067 | 8080809b | *
0001 | 80808080 | 80808080 | 
0002 | 80808092 | 80808080 | *
0003 | 80808080 | 80808080 | 
etc.
```

alanvgreen, 2021-12-19:
That's great news!

Any chance you can send either the things you needed to make verilator work or a VCD showing the failure?

alanvgreen, 2021-12-19:
Oh, I see #385 waiting for review. Will try that!

tcal-x, 2021-12-21:
@alanvgreen do you consider this resolved?

alanvgreen, 2021-12-29:
This is fixed now. Using the verilator sim we determined that input logic was continuing after the calculation had completed, which, about 1 in 16 times would affect the following calculation.

Fix was in https://github.com/google/CFU-Playground/pull/383/commits/26580e22f364cb799964c99bc218892bced45718

closed 2021-12-29 by alanvgreen