google/CFU-Playground #401: Create a Personal Access Token for cfu-playground-bot

PiotrZierhoffer, 2022-01-05:
This needs to be done to enable mergability of bump PRs.

Steps to follow:
1. Log in to GitHub with the cfu-playground-bot account
2. Go to https://github.com/settings/tokens
3. Create a new Personal Access Token
3.1. Add a clear note on what is the purpose of this PAT
3.2. Set the "Expiration" to "No expiration"
3.3. Set the "workflow" scope. It might be a bit too much, but it should not be a problem
3.4. Click "Generate token" and **write the token down. It will not be visible again**
4. Go to https://github.com/google/CFU-Playground/settings/secrets/actions
4.1. Click "New repository secret"
4.2. Set the name to `CFU_BOT_TOKEN`
4.3. Set the value to the PAT token value you wrote down in 3.4.
5. Let me know when this is ready to be used.

tcal-x, 2022-01-06:
@PiotrZierhoffer ok I followed the steps.

PiotrZierhoffer, 2022-01-12:
I created a PR here: https://github.com/google/CFU-Playground/pull/413

I am also running CI to verify if the PRs are created as expected: https://github.com/google/CFU-Playground/actions/runs/1686818994

PiotrZierhoffer, 2022-01-12:
@tcal-x I believe we might also need to enable auto-merge for this repo. You'll find the "Allow auto-merge " setting here: https://github.com/google/CFU-Playground/settings, in the "Merge button" section.

PiotrZierhoffer, 2022-02-02:
This can be closed as we're already using this token

closed 2022-02-02 by PiotrZierhoffer