google/CFU-Playground #408: proj_template is not working

ggangliu, 2022-01-10:
In below files, nmigen should be replaced by amaranth, right?
CFU-Playground/proj/proj_template/cfu_gen.py
CFU-Playground/proj/proj_template/cfu.py

Whether nmigen_cfu should be replaced by amaranth or still use this module?

tcal-x, 2022-01-10:
Hello @ggangliu ,

Is there an actual error so that `proj_template` cannot be built?   That would be surprising since we haven't seen failures in our GitHub Actions tests.   

Yes, we will need to update the nMigen submodule and our code that references it at some point.

ggangliu, 2022-01-11:
Hi @tcal-x,

I forgot attach error log. But I made below changes in cfu.py and cfu_gen.py. The issue was resolved.

proj/proj_template/cfu.py
```
# from nmigen import *
from  amaranth import *
```

proj/proj_template/cfu_gen.py
```
# from nmigen import *
from  amaranth import *
# from nmigen.back import rtlil, verilog
from amaranth.back import rtlil, verilog
```

ggangliu, 2022-01-11:
> Hi @tcal-x,
>
> I forgot attach error log. But I made below changes in cfu.py and cfu_gen.py. The issue was resolved.
>
> proj/proj_template/cfu.py
>
> ```
> # from nmigen import *
> from  amaranth import *
> ``` [...]

If needed, you can assign me to submit this fix. I can try to study the flow of contribution.

spicker23, 2022-02-07:
Hi everyone! I have the same errors during "making" proj_template. I did all you mentioned here above and it's still doesn't work. So is there anything else that should be switched due to switching to Amaranth?

tcal-x, 2022-02-08:
Hi @spicker23  -- you didn't have any problem until you started making changes to switch over to Amaranth, right?

spicker23, 2022-02-08:
> Hi @spicker23  -- you didn't have any problem until you started making changes to switch over to Amaranth, right?
> 

Probably there was a problem with paths, because after I added it in shell everything worked as it should. It costed two days to find out.