google/CFU-Playground #411: Icestorm Package in Conda Environment

ShvetankPrakash, 2022-01-12:
After running `make install-sf` and then activating the environment (i.e. `make enter`) I tried to run the golden tests for the 1bitsquared icebreaker by first running `make prog TARGET=1bitsquared_icebreaker`.

I got the following error when doing this:

FileNotFoundError: [Errno 2] No such file or directory: 'iceprog': 'iceprog'

Is there a missing package? Thanks in advance.

tcal-x, 2022-01-12:
At first I thought it was just that `icestorm` was missing from the environment*.yml files.   But then I saw that `iceprog` was missing from the `icestorm` packages.   I filed https://github.com/hdl/conda-eda/issues/156 for this.  

Here in this repo, I still need to add the `icestorm` packages.   Or another package -- whichever is the appropriate source of `iceprog`.

PiotrZierhoffer, 2022-01-12:
Actually, there is a separate iceprog package. I added it to environment files in a PR #415. @tcal-x  please take a look

tcal-x, 2022-01-12:
Thanks @PiotrZierhoffer !

closed 2022-01-12 by tcal-x