google/CFU-Playground #412: CFU Playground on Virtual machine

ItsmeVinayReddy, 2022-01-12:
Can the CFU Playground run on the virtual machine?

tcal-x, 2022-01-12:
Hello @ItsmeVinayReddy , 

Are you considering a VM on a local machine (e.g. a linux VM on a Mac)?
In that case, usually the main difficulty is passing the USB connection to actually
connect to the board from the VM.   I am not sure that it doesn't work, but I'm also not
sure that it will.

If for now you would just be simulating with Renode or Verilator, then it wouldn't be an issue.

I did have some success running CFU Playground on the Linux container in ChromeOS,
but also there I encountered some instability in the USB connection.   It would be fine for a while,
but eventually something would crash, and then the connection wouldn't work until I rebooted.
But I haven't tested that recently -- I should try it again tomorrow.

ggangliu, 2022-03-01:
Hi, I am also using VM, I successfully connected my board from VM. Just one change may is needed. I only changed USB type as USB3.0. When I was using USB2.0 before, vivado cannot recognize my board from VM. Not sure if it is suitable for you, hope it is helpful.

tcal-x, 2022-03-01:
Thanks @ggangliu .   

I forgot to mention some new information from my side -- I have an intern only using the container Linux VM on a Chromebook, and it's been working well for him, much better than it was working for me a  year ago.