google/CFU-Playground #417: "make prog USE_SYMBIFLOW=1" ran into error

ggangliu, 2022-01-12:
Hi,

- I met below error info when I tried to make prog in template proj. May I get help, thanks. 
- I also plan to run it on XC7Z010-1CLG400C, so is there any reference guide for me ?  Thanks.

Thanks a million.

By the way, I am also working on VM, I noted that someone also want to work in VM.

```
ggang@ubuntu:~/CFU-Playground/proj/proj_template$ make prog USE_SYMBIFLOW=1
make[4]: Entering directory '/home/ggang/CFU-Playground/proj/proj_template'
/home/ggang/CFU-Playground/scripts/pyrun /home/ggang/CFU-Playground/proj/proj_template/cfu_gen.py 
make -C /home/ggang/CFU-Playground/soc -f /home/ggang/CFU-Playground/soc/common_soc.mk prog
make[5]: Entering directory '/home/ggang/CFU-Playground/soc'
Building bitstream for digilent_arty. CFU option: --cpu-cfu /home/ggang/CFU-Playground/proj/proj_template/cfu.v
MAKEFLAGS=-j8 /home/ggang/CFU-Playground/scripts/pyrun ./common_soc.py --output-dir build/digilent_arty.proj_template --csr-json build/digilent_arty.proj_template/csr.json --cpu-cfu  /home/ggang/CFU-Playground/proj/proj_template/cfu.v --uart-baudrate 1843200 --target digilent_arty  --toolchain symbiflow --build
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2022-01-12 06:46:24)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7a35ticsg324-1L.
INFO:SoC:System clock: 100.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU overriding rom mapping from 0x00000000 to 0x00000000.
INFO:SoC:CPU overriding sram mapping from 0x01000000 to 0x10000000.
INFO:SoC:CPU overriding main_ram mapping from 0x40000000 to 0x40000000.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:S7PLL:Creating S7PLL, speedgrade -1.
INFO:S7PLL:Registering Single Ended ClkIn of 100.00MHz.
INFO:S7PLL:Creating ClkOut0 sys of 100.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut1 sys4x of 400.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut2 sys4x_dqs of 400.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut3 idelay of 200.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut4 eth of 25.00MHz (+-10000.00ppm).
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x10000000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:S7PLL:Config:
[... 79 lines cut ...]
make[6]: Leaving directory '/home/ggang/CFU-Playground/soc/build/digilent_arty.proj_template/software/libc'
Traceback (most recent call last):
  File "./common_soc.py", line 57, in <module>
    main()
  File "./common_soc.py", line 53, in main
    workflow.run()
  File "/home/ggang/CFU-Playground/soc/board_specific_workflows/general.py", line 121, in run
    soc_builder = self.build_soc(soc)
  File "/home/ggang/CFU-Playground/soc/board_specific_workflows/digilent_arty.py", line 59, in build_soc
    return super().build_soc(soc, **kwargs)
  File "/home/ggang/CFU-Playground/soc/board_specific_workflows/general.py", line 98, in build_soc
    soc_builder.build(run=self.args.build, **kwargs)
  File "/home/ggang/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 320, in build
    self._generate_rom_software(compile_bios=use_bios)
  File "/home/ggang/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 264, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/home/ggang/CFU-Playground/env/conda/envs/cfu-common/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/ggang/CFU-Playground/soc/build/digilent_arty.proj_template/software/libc', '-f', '/home/ggang/CFU-Playground/third_party/python/litex/litex/soc/software/libc/Makefile']' returned non-zero exit status 2.
make[5]: *** [/home/ggang/CFU-Playground/soc/common_soc.mk:115: build/digilent_arty.proj_template/gateware/digilent_arty.bit] Error 1
make[5]: Leaving directory '/home/ggang/CFU-Playground/soc'
make[4]: *** [../proj.mk:284: prog] Error 2
make[4]: Leaving directory '/home/ggang/CFU-Playground/proj/proj_template'
ggang@ubuntu:~/CFU-Playground/proj/proj_template$ 
```

tcal-x, 2022-01-12:
@ggangliu thank you for the report.   I remember seeing this error elsewhere when LiteX first switched to using meson.  

My first suggestion is to ensure that all submodules (and *their* submodules) have been cloned correctly.   If you are having trouble with "https" you may need to switch some more to ssh "git@" style urls.

There is more information here:
https://github.com/enjoy-digital/litex/issues/1045#issuecomment-932369254

Repeating here,
> I think I saw a similar issue in CFU Playground; the solution there was: make sure that the new picolibc submodule gets a recursive update, so that it loads *its* submodules.
> 
> I think this should happen if you run ./litex_setup.py init update install, but please check that the pythondata_software_picolibc/data submodule is loaded and not an empty directory.

tcal-x, 2022-01-12:
@ggangliu about XC7Z010-1CLG400C -- I have not yet used that part myself.    Is that on the first board that you have, or that will be a second experiment?    In the command above, you would be building for the default Arty A7-35T board, with some version of XC7A35T.

ggangliu, 2022-01-13:
> @ggangliu about XC7Z010-1CLG400C -- I have not yet used that part myself. Is that on the first board that you have, or that will be a second experiment? In the command above, you would be building for the default Arty A7-35T board, with some version of XC7A35T.

It is my first board try to use. I think it is possible to change default Arty A7-35T to suitable for XC7Z010-1CLG400C. So I mean if there is any doc to guide how to support a new board as a reference. Or is there a similar project to refer, which is very close to XC7Z010-1CLG400C? If NOT, I would like try to do it. I am not gonna spend money to buy a new board.

tcal-x, 2022-01-13:
@ggangliu I understand.   To use a board in CFU Playground, first it needs to be part of [litex-boards](https://github.com/litex-hub/litex-boards/tree/master/litex_boards/platforms), so that would be the first step .   If your board isn't there, it should be straightforward to add it (you need to add a file under each of platforms/ and targets/).   The main work is adding the information about how the FPGA package pins match to resources on the board (LEDs etc.).

There are already some boards that support similar parts: https://github.com/litex-hub/litex-boards/search?q=xc7z010

ggangliu, 2022-01-13:
> @ggangliu I understand. To use a board in CFU Playground, first it needs to be part of [litex-boards](https://github.com/litex-hub/litex-boards/tree/master/litex_boards/platforms), so that would be [...]

Thank @tcal-x very much, I will feedback when I have any result.