google/CFU-Playground #421: got an issue in digilent_arty.fasm when building bitstream

ggangliu, 2022-01-16:
Hi
I ran into below issue when executing "make bitsteram USE_SYMBIFLOW=1".  But I guess it is very close to the final success, thanks for your help in advance.

```
ggang@ubuntu:~/CFU-Playground/proj/proj_template$ make bitstream USE_SYMBIFLOW=1
make[4]: Entering directory '/home/ggang/CFU-Playground/proj/proj_template'
/home/ggang/CFU-Playground/scripts/pyrun /home/ggang/CFU-Playground/proj/proj_template/cfu_gen.py 
make -C /home/ggang/CFU-Playground/soc -f /home/ggang/CFU-Playground/soc/common_soc.mk bitstream
make[5]: Entering directory '/home/ggang/CFU-Playground/soc'
Building bitstream for digilent_arty. CFU option: --cpu-cfu /home/ggang/CFU-Playground/proj/proj_template/cfu.v
MAKEFLAGS=-j8 /home/ggang/CFU-Playground/scripts/pyrun ./common_soc.py --output-dir build/digilent_arty.proj_template --csr-json build/digilent_arty.proj_template/csr.json --cpu-cfu  /home/ggang/CFU-Playground/proj/proj_template/cfu.v --uart-baudrate 1843200 --target digilent_arty  --toolchain symbiflow --build
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2022-01-15 19:02:41)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7a35ticsg324-1L.
INFO:SoC:System clock: 100.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU overriding rom mapping from 0x0 to 0x0.
INFO:SoC:CPU overriding sram mapping from 0x1000000 to 0x10000000.
INFO:SoC:CPU overriding main_ram mapping from 0x40000000 to 0x40000000.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:S7PLL:Creating S7PLL, speedgrade -1.
INFO:S7PLL:Registering Single Ended ClkIn of 100.00MHz.
INFO:S7PLL:Creating ClkOut0 sys of 100.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut1 sys4x of 400.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut2 sys4x_dqs of 400.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut3 idelay of 200.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut4 eth of 25.00MHz (+-10000.00ppm).
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x10000000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:S7PLL:Config:
divclk_divide : 1
clkout0_freq  : 100.00MHz
clkout0_divide: 16
clkout0_phase : 0.00°
clkout1_freq  : 400.00MHz
clkout1_divide: 4
[... 379 lines cut ...]
  File "./common_soc.py", line 53, in main
    workflow.run()
  File "/home/ggang/CFU-Playground/soc/board_specific_workflows/general.py", line 121, in run
    soc_builder = self.build_soc(soc)
  File "/home/ggang/CFU-Playground/soc/board_specific_workflows/digilent_arty.py", line 59, in build_soc
    return super().build_soc(soc, **kwargs)
  File "/home/ggang/CFU-Playground/soc/board_specific_workflows/general.py", line 98, in build_soc
    soc_builder.build(run=self.args.build, **kwargs)
  File "/home/ggang/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 330, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/ggang/CFU-Playground/third_party/python/litex/litex/soc/integration/soc.py", line 1139, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/ggang/CFU-Playground/third_party/python/litex/litex/build/xilinx/platform.py", line 55, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/ggang/CFU-Playground/third_party/python/litex/litex/build/xilinx/symbiflow.py", line 234, in build
    _run_make()
  File "/home/ggang/CFU-Playground/third_party/python/litex/litex/build/xilinx/symbiflow.py", line 84, in _run_make
    raise OSError("Error occured during Symbiflow's script execution.")
OSError: Error occured during Symbiflow's script execution.
make[5]: *** [/home/ggang/CFU-Playground/soc/common_soc.mk:115: build/digilent_arty.proj_template/gateware/digilent_arty.bit] Error 1
make[5]: Leaving directory '/home/ggang/CFU-Playground/soc'
make[4]: *** [../proj.mk:287: bitstream] Error 2
make[4]: Leaving directory '/home/ggang/CFU-Playground/proj/proj_template'
ggang@ubuntu:~/CFU-Playground/proj/proj_template$ 
```

tcal-x, 2022-01-16:
Hi @ggangliu , please follow these steps to get more information:
* Go to the build gateware directory: `cd /home/ggang/CFU-Playground/soc/build/digilent_arty.proj_template/gateware`
* In the `Makefile`, edit the line `symbiflow_write_fasm -e digilent_arty.eblif -d xc7a50t_test > /dev/null` to **remove** the ` > /dev/null` so that you can see the output.
* Now (still in `gateware/`), just type `make`

ggangliu, 2022-01-17:
Hi @tcal-x, As you said, I got below logs, thanks.

I got below log if running “make” without "make clean". 
```
ggang@ubuntu:~/CFU-Playground/soc/build/digilent_arty.proj_template/gateware$ make 
make[2]: Entering directory '/home/ggang/CFU-Playground/soc/build/digilent_arty.proj_template/gateware'
symbiflow_write_bitstream -d artix7 -f digilent_arty.fasm -p xc7a35tcsg324-1 -b digilent_arty.bit
Writing bitstream ...
make[2]: Leaving directory '/home/ggang/CFU-Playground/soc/build/digilent_arty.proj_template/gateware'
ggang@ubuntu:~/CFU-Playground/soc/build/digilent_arty.proj_template/gateware$ 
```
But If I run "make clean" first, then to run "make", I got the log as attached file. 
[log.txt](https://github.com/google/CFU-Playground/files/7882773/log.txt)

ggangliu, 2022-01-17:
Thanks @tcal-x I think I have knew the reason. I extended my virtual machine ram space to 3GB from 2GB, the issue was solved. Thanks a lot. All are because my personal fault.

closed 2022-01-17 by ggangliu

tcal-x, 2022-01-18:
@ggangliu thank you for your feedback.   In fact I think I have an open issue -- that we need to print out a useful and informative error in this case (running out of memory).    It is not your fault at all.  The documentation does not even say "you should have X GB of memory available".   So again, thank you for your feedback.

Also, there's a related topic -- we also don't currently print out an error if the place and route does not meet the timing constraints.   In that case, the bitstream might not work on the board.