google/CFU-Playground #425: proj_template doesn't boot after "Liftoff" while LiteX demo does

sammhho, 2022-01-21:
Hi all,

As titled, the `proj_template` app didn't work for me out-of-the-box, no prompt after "Liftoff" message.
The LiteX demo app, e.g. the donut app, did work though.
Also, the `proj_template` did work inside a Renode simulation
(didn't seem to have set a target so probably was the default digilent_arty).

Some backgrounds: I'm using the target `efinix_titanium_ti60_f225_dev_kit` in LiteX, 
created a board specific workflow just to enable `with_hyperram=True, cpu_variant='full+cfu+debug'` 
(had to bump `litehyperbus` tag from 2021.08 to 2021.12 for configurable latency to work, anyway),
and also added a "serial2" similar to the "hps" target for debugging, by `add_uartbone("serial2", 115200)`.

Now, with `gdb (in vscode)` --> `openocd (SpinalHDL fork)` --> `litex_server (/dev/ttyUSB4)` --> board, 
I can step thru and debug the bios,/demo(donut) app/software.bin from the `proj_template`, 
and I can view CPU registers and set breakpoints.

However, I'm not sure what to look for and why `proj_template` is not working.
When I set a breakpoint around main of `proj_template`, and do "Step Into" on `init_runtime()`, I hit an error that pops a red box saying:
`Exception has occurred.
Signal 0`
where in the debug console it says
`Program stopped.
0x4000045c in main () at src/main.c:77
77    init_runtime();`
Which is weird, coz gdb seems to be still working just fine, and I can proceed.

Now when I run step by step and reach 
https://github.com/enjoy-digital/litex/blob/d36e1b60bdd8e6975bb5e0ad16487a23e5aa372f/litex/soc/software/libc/stdio.c#L23
Any stepping further will result in gdb losing control, can't `Pause (F6)` to see where the CPU is running, so probably not stuck in some loop. 

At first I suspected it's something similar to https://github.com/enjoy-digital/litex/issues/876 which is due mis-alignment in linker.ld, so I looked at the `common/ld/linker.ld` but it looks fine.

Any suggestions?

tcal-x, 2022-01-21:
Thanks for the report, @sammhho.

Cool, Efinix!

Let me throw out some ideas.   Since there might be something UART related, perhaps try rebuilding setting UART speed in proj/proj.mk to the normal 115200 (maybe you are already overriding that).

I sometimes saw a hang when the BIOS attempted a dcache flush but there was no dcache.   But your build should have a dcache, and also, I think you are getting past that point.

Can you connect to the LiteX BIOS prompt using the CFU Playground build (not loading the CFU Playground `software.bin` binary)?   Use `lxterm` or even `picocom` with the correct UART speed and without the `--kernel ....` option that loads the software.

Or even before that, have you connected to the LiteX BIOS prompt with a straight LiteX build?

sammhho, 2022-01-22:
Thx for the quick reply @tcal-x.

The "serial" is running at default speed 1843200, while the "serial2" for etherbone debug is running 115200.
I presume "serial2" is running correctly as I can use `litex_client` to print stuffs and gdb can see CPU register values and step thru debugged program, etc. I can try lowering the "serial" speed.

Yes I can boot into the BIOS prompt via `litex_term` without `--kernel`, and if I previously booted the demo app with `--kernel`, in the BIOS menu I can also boot into the app with `boot 0x40000000`.
(BTW I can `gdb` step thru both BIOS and DEMO app as well)
Similarly for the CFU Playground `software.bin`, but just that it doesn't function properly, and gdb stepping thru the very first `printf` resulted in `gdb` losing connection.

sammhho, 2022-01-22:
Ok, one interesting fact: 
`make renode` runs successfully, while 
`make TARGET=efinix_titanium_ti60_f225_dev_kit` puts me in some sort of infinite looping...

tcal-x, 2022-01-22:
Hmm, that would be very useful if Renode is stuck at the same place as with hardware.

We are working on publishing a tutorial to show how useful Renode is, but for now I can show you a few of the commands that I know.

In the *monitor* window (the one that starts with the fancy RENODE logo), at the prompt you can type:
```
sysbus.cpu PC
```
to see where you are in the program.  You can match it with `./build/software.elf.dis`, or look it up directly within Renode with:
```
sysbus FindSymbolAt 0x40011334
```
(substituting the hex value for the PC that you found earlier).

You can also trace which functions you're in by doing this:
```
sysbus.cpu LogFunctionNames true
```
which sends its output the window where you ran the `make .... renode` command.

There are other things you can try at https://renode.readthedocs.io/en/latest/basic/machines.html (e.g. `peripherals`).

P.S. I did have an alignment issue early on, so it's good that you checked that.

sammhho, 2022-01-25:
@tcal-x Thx for the suggestions. The Renode emulation with gdb remote was quite useful.
After some `make clean` everything somehow worked. I'm closing this issue.
BTW, any plans for DMA based example design as mentioned in FCCM slides?

closed 2022-01-25 by sammhho

tcal-x, 2022-02-10:
> @tcal-x Thx for the suggestions. The Renode emulation with gdb remote was quite useful. After some `make clean` everything somehow worked. I'm closing this issue. BTW, any plans for DMA based example design as mentioned in FCCM slides?

Hi @sammhho , sorry I forgot to follow up on this!    In fact recently we added a special memory port (actually 4x!) to the CFU so that it could have direct access to a scratchpad memory (https://github.com/google/CFU-Playground/blob/main/soc/hps_soc.py#L198-L230, used by the hps_accel project).    This memory is also accessible from the Wishbone bus, so it can be written to by the CPU.    This isn't exactly DMA, but kind of shows us that it's possible.  If we want to allow the CFU to directly access the main memory, then instead of using special direct connections, we'd just connect the CFU to the Wishbone bus directly.   However, I haven't looked into the details of how we'd add arbitration considering the case where both the CFU and CPU want to use the bus.

tcal-x, 2022-02-10:
> Hello! Proj_template doesn't boot after Liftoff. And moreover string "Hello world" interrupts after an "H". How can I find out where to find a problem?

@spicker23 , what board and toolchain are you using?

spicker23, 2022-02-10:
> > Hello! Proj_template doesn't boot after Liftoff. And moreover string "Hello world" interrupts after an "H". How can I find out where to find a problem?
> 
> @spicker23 , what board and toolchain are you using?
> 

I'm using digilent nexys4 and toolchain described in manual

tcal-x, 2022-02-10:
> > > Hello! Proj_template doesn't boot after Liftoff. And moreover string "Hello world" interrupts after an "H". How can I find out where to find a problem?
> > 
> > 
> > @spicker23 , what board and toolchain are you using?
> 
> I'm using digilent nexys4 ddr and toolchain described in manual

Meaning SymbiFlow, right?   (We do also describe how to install Vivado in the docs.)

Fun fact: adding xc7a100t support to SymbiFlow was my first project here.
I don't have a Nexys 4 but I do have an Arty 100T, so I will try building and running on that.

In the meantime, can you try running in Renode simulation?   This does work for me:
```
make  BUILD_JOBS=4  TARGET=digilent_nexys4ddr   renode
```

tcal-x, 2022-02-10:
@spicker23 , I was able to build and run the software on the Arty 100T.   It even worked correctly at the default 100MHz; in some other cases I needed to reduce the frequency to something like 75MHz.   Although I see that for the Nexyx 4 DDR, the default frequency is already 75MHz.

Can you post the exact command lines that you used to build the gateware and software?
Thanks!