google/CFU-Playground #452: Adding custom ML tflite models with CFU

bala122, 2022-02-04:
I'm not sure about the procedure to add your own custom tflite model with CFU. Does this framework support custom ML models as well?
I'm getting this issue when I attempt at adding a new model in the makefile:

/home/shivaubuntu/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: src/models/models.o: in function `.LANCHOR0':
models.c:(.sdata.MENU+0x10): undefined reference to `no_menu'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:163: software.elf] Error 1
make[1]: Leaving directory '/home/shivaubuntu/CFU-playground/CFU-Playground/proj/proj_custom/build'
make: *** [../proj.mk:215: /home/shivaubuntu/CFU-playground/CFU-Playground/proj/proj_custom/build/software.elf] Error 2

bala122, 2022-02-04:
I've even followed all the steps by an earlier issue:
https://github.com/google/CFU-Playground/issues/221

I'm still getting the same error. I'm using the August build of the SiFive compiler:
https://github.com/sifive/freedom-tools/releases/tag/v2020.08.0
I've even tried using the riscv64 gcc compiler available in riscv-gnu toolchain, but even that doesnt seem to work.
Any suggestions on compilers to use would be helpful

alanvgreen, 2022-02-04:
Hi @bala122,

Did you follow this part of the instructions?

>  After creating your model folder, make it accessible from the menus by adding it to 
> [/common/src/models.c](https://github.com/google/CFU-Playground/blob/main/common/src/models/models.c?rgh-link-date=2021-08-12T14%3A05%3A52Z).

If that's not the problem, could you perhaps upload a Draft PR containing your code so far, and someone might be able to take a look at it.

Thanks,

Alan

bala122, 2022-02-06:
Hi, yes I've tried that, but the issue is still present.  Ive added a Draft PR:
https://github.com/google/CFU-Playground/pull/456

alanvgreen, 2022-02-06:
> make[1]: riscv64-unknown-elf-g++: Command not found

Make is saying that it can't find the RISCV C++ compiler on your path.

bala122, 2022-02-07:
I've installed the August version of the SiFive gcc compiler from freedom tools:
https://github.com/sifive/freedom-tools/releases/tag/v2020.08.0
I've even tried with the normal riscv-gnu-toolchain compiler it didnt work. I think the compiler is recognized whenever I type "which riscv64-unknown-elf-g++"

tcal-x, 2022-02-08:
> Specifically under "sudo make renode", I'm getting this:

Hi @bala122 , can you send me the link to where you see this specifically?   I'm puzzled why the instructions say to use "sudo".
Thanks!

bala122, 2022-02-08:
Hi, actually the issue is resolved I think. I deleted the soc build and re-did it with the normal "make renode" command, it worked. I'll let you know if there are any further issues.
Thanks,
Bala.

tcal-x, 2022-02-08:
Hi @bala122 , thanks for the update; if you have any suggestions for improving the documentation, we would like to hear!   I know it is quite complicated and we might not explain every step well.

Do you think you will upstream your model at some point?   It's fine either way of course.

bala122, 2022-03-24:
Hi, yes Ill try to give suggestions later on regarding the documentation. As of now I havent fully finished the design, so Ill let you know about that when its done.
Thanks.