google/CFU-Playground #454: CNN Accelerator on FPGA based on LiteX

gayathri191855, 2022-02-05:
Procedure to develop CNN Accelerator on FPGA based on LiteX

tcal-x, 2022-02-06:
Hello @gayathri191855 , is there an issue, or is this acting as a bookmark for you?

gayathri191855, 2022-02-06:
what algorithm is used to develop a CNN accelerator using LiteX and what is approach to develop

tcal-x, 2022-02-07:
@gayathri191855 , the CFU Playground uses the TensorFlow Lite for Microcontrollers framework, so it is assumed that we will be using a TFLite model to represent the neural network.   Then, you can add custom instructions to the CPU by designing your own Custom Function Unit (CFU), and modifing the TFLiteMicro kernels to use these new instructions to achieve accelerated performance.     This page shows an end-to-end example: https://cfu-playground.readthedocs.io/en/latest/step-by-step.html 

This is just one of really unlimited ways that you could add a CNN accelerator to a LiteX SoC.   You could instead make a more independent accelerator that sits on the SoC bus.   Maybe it could perform DMA accesses to memory all by itself.   It could be controlled by the CPU using memory-mapped control registers.   But that  is outside of the scope of CFU Playground.

closed 2022-03-11 by gayathri191855

disdi, 2022-03-11:
> This is just one of really unlimited ways that you could add a CNN accelerator to a LiteX SoC. You could instead make a more independent accelerator that sits on the SoC bus. Maybe it could perform [...]

@tcal-x so ideally you suggest CFU playground project output could be configured (to create a Custom Function Unit (CFU)) and be included as part of Litex like lite-CFU something similar to litespi , litevideo as seen here https://github.com/orgs/litex-hub/repositories ?