google/CFU-Playground #455: Unable to load the RISCV program onto the board when using Symbiflow

APIUM, 2022-02-06:
Board: ARTY A7
OS: Ubuntu 20.04 VM
TERM: Tested both `urxvt` and `gnome-terminal`
Running latest master of CFU-Playground, following [Setup Guide](https://cfu-playground.readthedocs.io/en/latest/setup-guide.html)

----

Command `make load USE_SYMBIFLOW=1` does not finish, stuck at `/home/{user}/projects/CFU-Playground/soc/bin/litex_term --speed 1843200  --kernel /home/{user}/projects/CFU-Playground/proj/proj_template/build/software.bin /dev/ttyUSB1`.

Pressing ENTER or typing `ENTER` as the Setup Guide states does not work, have left it stuck at that line for several hours with no change. The previous step `make prog` completes successfully with the LED sequence displayed on the ARTY. 

Rerunning script/setup has no change, all expected packages are installed. 

Works with Vivado with command `make load` - Golden tests pass successfully. I don't think this is an issue with the USB passthrough as a result, but not sure what else it could be. Has this been observed before?

tcal-x, 2022-02-07:
@APIUM , thank you for the report!

I think this is due to the place and route tools not meeting timing, but not reporting on it.  So the generated bitstream just won't work at 100MHz.   I'm working to address that (https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1928).

In the mean time, we can try reducing the target clock rate for the Symbiflow build.    Let me verify that I have the correct incantation to do that, then I'll post the instructions here, then you can check if it works for you.

tcal-x, 2022-02-07:
@APIUM , this worked for me on my Arty board.   That's not a 100% guarantee it will work on your board.   This drops the system clock frequency to 75MHz:
```
% make prog USE_SYMBIFLOW=1 EXTRA_LITEX_ARGS="--sys-clk-freq 75000000"
% make load USE_SYMBIFLOW=1 EXTRA_LITEX_ARGS="--sys-clk-freq 75000000"
```

Note that you can't drop the frequency arbitrarily low; when you get close to 60MHz or lower, the DDR3 memory will not function correctly.

APIUM, 2022-02-08:
Thanks @tcal-x that worked perfectly for me. I've updated the documentation to suggest that users following the Setup Guide with the recommended hardware try this: https://github.com/google/CFU-Playground/pull/459.

APIUM, 2022-02-09:
Closing this with the merge of https://github.com/google/CFU-Playground/pull/459, thanks again for the help @tcal-x !

closed 2022-02-09 by APIUM