google/CFU-Playground #469: Uprev yosys in CI to >= 0.10

alanvgreen, 2022-02-11:
The github CI seems to install yosys from Ubuntu with a pre-0.10 version number:

```
Get:37 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 yosys amd64 0.9-1build2 [2131 kB]
```

This is blocking the nMigen to Amaranth migration (#466).

tcal-x, 2022-02-11:
Thanks for debugging @alanvgreen -- should be an easy fix.

tcal-x, 2022-02-11:
Actually it's not quite what I was expecting.    I didn't realize, or forgot, that our setup script would grab the yosys package via apt.   It should never do that, whether in CI, or whether a user is setting up.

tcal-x, 2022-02-11:
I've appended commits to #466 to ./scripts/setup to do a local binary install of Yosys in if there is no system version, or if the system version is too old.   This takes a few seconds.

I also considered and rejected:
* Clone and build of Yosys: this took 12 minutes
* Set up a Conda environment with just Yosys and then copy the files out:  this took 1.5 minutes

closed 2022-02-11 by tcal-x