google/CFU-Playground #517: Support 100T as variant of digilent_arty in makefiles

gits00, 2022-04-02:
I am following the method outlined in setup guidelines at https://cfu-playground.readthedocs.io/en/latest/setup-guide.html. My board is Digilent Arty A7 100T. 

It gets hung at - 

.....litex_term --speed 1843200  --kernel ..../CFU-Playground/proj/proj_template/build/software.bin /dev/ttyUSB1

This is mostly because default is Arty A7 35T in the workflow . How to change it to use  Arty A7 100T (code changes or makefile parameters) ? I am trying this with the Vivado 2021.2 toolchain and not Symbiflow.

tcal-x, 2022-04-02:
Hi @gits00 , can you check the hints on https://github.com/google/CFU-Playground/wiki/Supported-Boards#arty-a7-35t100t  and see if that works for you?

gits00, 2022-04-03:
@tcal-x. Thank  you. The  EXTRA_LITEX_ARGS='--variant=a7-100' passed during make worked and I could boot to prompt!

closed 2022-04-03 by gits00