google/CFU-Playground #536: ERROR: Can't load module `./dsp-ff'

alanvgreen, 2022-04-19:
I get the following error when building hps_accel:

```
6.2. Continuing TECHMAP pass.
Using template VHI for cells of type VHI.
Using template VLO for cells of type VLO.
No more expansions possible.
<suppressed ~34 debug messages>
ERROR: Can't load module `./dsp-ff': /usr/local/google/home/avg/src/CFU-Playground/third_party/usr/local/bin/../share/yosys/plugins/dsp-ff.so: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "/usr/local/google/home/avg/src/CFU-Playground/soc/./hps_soc.py", line 353, in <module>
    main()
  File "/usr/local/google/home/avg/src/CFU-Playground/soc/./hps_soc.py", line 345, in main
    vns = builder.build(**builder_kwargs, run=args.build)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 350, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/soc/integration/soc.py", line 1147, in build
    return self.platform.build(self, *args, **kwargs)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/build/lattice/platform.py", line 43, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/build/lattice/oxide.py", line 169, in build
    _run_script(script)
  File "/usr/local/google/home/avg/src/CFU-Playground/third_party/python/litex/litex/build/lattice/oxide.py", line 104, in _run_script
    raise OSError("Error occured during Yosys/Nextpnr's script execution.")
OSError: Error occured during Yosys/Nextpnr's script execution.
make[1]: *** [/usr/local/google/home/avg/src/CFU-Playground/soc/hps.mk:93: build/hps.hps_accel/gateware/hps_proto2_platform.bit] Error 1
make[1]: Leaving directory '/usr/local/google/home/avg/src/CFU-Playground/soc'
make: *** [../proj.mk:300: bitstream] Error 2
```

- I used `cd proj/hps_accel; make gateware`
- I am on commit b77f26db0a78417f0f32e92ac2d9ef809db656c0.
- I have run `scripts/setup`
- I have run `scripts/install_oxide update`
- There is no file named `dsp-ff.so` anywhere in the CFU-Playground tree

Am I missing a yosys update or build flag?

tcal-x, 2022-04-19:
I probably forgot to add the new plugin to the `install_oxide` flow.

tmichalak, 2022-04-20:
After applying the changes from #540 I was able to build the plugins correctly, so the `dsp-ff.so` file showed up in `./third_party/usr/local/share/yosys/plugins/dsp-ff.so` and the error suring synthesis was gone.

tcal-x, 2022-04-27:
@alanvgreen can you update the status here when you get a chance?