google/CFU-Playground #539: scripts/install_oxide failure: *** "Didn't find 'yosys-config' under '/usr'"

alanvgreen, 2022-04-20:
I made a fresh checkout of CFU-Playground, then ran `scripts/setup` and `scripts/install_oxide`. It failed with an error message: `../Makefile_plugin.common:47: *** "Didn't find 'yosys-config' under '/usr'".  Stop.`

Running install_oxide a second time gives the same error:
```
~/src/CFU-Playground (main %|u=)
14:07:04 $ scripts/install_oxide
Sudo Installing pre-requisites
[sudo] password for avg: 
Hit:1 https://rapture-prod.corp.google.com goobuntu-workstation@avg920.syd.corp.google.com InRelease
Hit:2 https://rapture-prod.corp.google.com glitter@avg920.syd.corp.google.com InRelease
Get:3 https://rapture-prod.corp.google.com goobuntu-puppet-addons InRelease [8,137 B]
Hit:4 https://rapture-prod.corp.google.com glinux-rodete-base-20220323.03.06 InRelease
Hit:5 https://rapture-prod.corp.google.com glinux-rodete-base-priority-20220323.03.06 InRelease
Hit:6 https://rapture-prod.corp.google.com glinux-rodete-canaries-20220323.03.06@avg920.syd.corp.google.com InRelease
Hit:7 https://rapture-prod.corp.google.com glinux-rodete-extra-20220323.03.06 InRelease
Hit:8 https://rapture-prod.corp.google.com glinux-rodete-extra-priority-20220323.03.06 InRelease
Hit:9 https://rapture-prod.corp.google.com typescript-stable InRelease
Hit:10 https://rapture-prod.corp.google.com usblock-stable InRelease
Get:11 https://rapture-prod.corp.google.com goobuntu-puppet-addons/main amd64 Packages [20.7 kB]
Get:12 https://rapture-prod.corp.google.com goobuntu-puppet-addons/main i386 Packages [19.6 kB]
Fetched 48.4 kB in 3s (13.9 kB/s)                           
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
bison is already the newest version (2:3.8.2+dfsg-1).
clang is already the newest version (1:13.0-54).
cmake is already the newest version (3.22.1-1+build1).
flex is already the newest version (2.6.4-8).
libboost-all-dev is already the newest version (1.74.0.3).
libeigen3-dev is already the newest version (3.4.0-2).
tcl-dev is already the newest version (8.6.11+1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
/usr/local/google/home/avg/.cargo/bin/cargo

CLONING REPOSITORIES

BUILDING YOSYS

  Build successful.

mkdir -p /usr/local/google/home/avg/src/CFU-Playground/third_party/usr/local/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc /usr/local/google/home/avg/src/CFU-Playground/third_party/usr/local/bin
strip -S /usr/local/google/home/avg/src/CFU-Playground/third_party/usr/local/bin/yosys
strip /usr/local/google/home/avg/src/CFU-Playground/third_party/usr/local/bin/yosys-abc
strip /usr/local/google/home/avg/src/CFU-Playground/third_party/usr/local/bin/yosys-filterlib
mkdir -p /usr/local/google/home/avg/src/CFU-Playground/third_party/usr/local/share/yosys
cp -r share/. /usr/local/google/home/avg/src/CFU-Playground/third_party/usr/local/share/yosys/.

BUILDING YOSYS PLUGINS
../Makefile_plugin.common:47: *** "Didn't find 'yosys-config' under '/usr'".  Stop.
```

alanvgreen, 2022-04-20:
The yosys plugins make file `../Makefile_plugin.common` contains these lines:

```
# Either find yosys in system and use its path or use the given path
YOSYS_PATH ?= $(realpath $(dir $(shell which yosys))/..)

# Find yosys-config, throw an error if not found
YOSYS_CONFIG ?= $(YOSYS_PATH)/bin/yosys-config
```

This sets `YOSYS_PATH` to /usr/yosys, since it found a copy of yosys previously installed by `apt install yosys`. Removing that copy of yosys then causes the makefile to look for `yosys-config` in `/`, which is wildly unhelpful.

I think the right fix here is for `install_oxide` to add `third_party/usr/local/bin` to  `PATH` before starting the yosys-plugins build. This will then find the yosys-config at that location.

alanvgreen, 2022-04-20:
After some more investigation, a combination of changes to both the install_oxide script and the Yosys Makefile were required to fix this problem. 

- https://github.com/YosysHQ/yosys/pull/3285
- #540

alanvgreen, 2022-04-21:
On reflection, and with [some guidance from @Xiretza](https://github.com/YosysHQ/yosys/pull/3285#issuecomment-1103502369) , it looks like we should update `install_oxide` to install yosys into the correct directories without using `DESTDIR`. [Standard `DESTDIR` usage](https://www.gnu.org/prep/standards/html_node/DESTDIR.html) is that it is only for `install` and `uninstall` rules, and only for staging installs, not for setting the final file destintations..

tcal-x, 2022-04-22:
I believe this issue is resolved, reopen if not.

closed 2022-04-22 by tcal-x