google/CFU-Playground #546: hps_accel: make renode-headless fails with verilator error

alanvgreen, 2022-04-22:
In proj/hps_accel, `make renode-headless` fails with the following verilator error.

```
-- Executing Verilator...
Verilator command: "/usr/bin/verilator_bin --compiler gcc --prefix Vcfu --Mdir /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/build/renode/CMakeFiles/libVtop.dir/Vcfu.dir --make cmake --cc -y /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel -Wno-WIDTH -Wno-CASEINCOMPLETE -Wno-CASEOVERLAP /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/cfu.v"
Output:
%Error: /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/cfu.v:6533:3: Cannot find file containing module: 'MULTADDSUB9X9WIDE'
 6533 |   MULTADDSUB9X9WIDE #(
      |   ^~~~~~~~~~~~~~~~~
        ... Looked in:
             /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/MULTADDSUB9X9WIDE
             /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/MULTADDSUB9X9WIDE.v
             /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/MULTADDSUB9X9WIDE.sv
             MULTADDSUB9X9WIDE
             MULTADDSUB9X9WIDE.v
             MULTADDSUB9X9WIDE.sv
             /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/build/renode/CMakeFiles/libVtop.dir/Vcfu.dir/MULTADDSUB9X9WIDE
             /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/build/renode/CMakeFiles/libVtop.dir/Vcfu.dir/MULTADDSUB9X9WIDE.v
             /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/build/renode/CMakeFiles/libVtop.dir/Vcfu.dir/MULTADDSUB9X9WIDE.sv
%Error: /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/cfu.v:6325:3: Cannot find file containing module: 'MULTADDSUB9X9WIDE'
 6325 |   MULTADDSUB9X9WIDE #(
      |   ^~~~~~~~~~~~~~~~~
%Error: /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/cfu.v:6114:3: Cannot find file containing module: 'MULTADDSUB9X9WIDE'
 6114 |   MULTADDSUB9X9WIDE #(
      |   ^~~~~~~~~~~~~~~~~
%Error: /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/cfu.v:5905:3: Cannot find file containing module: 'MULTADDSUB9X9WIDE'
 5905 |   MULTADDSUB9X9WIDE #(
      |   ^~~~~~~~~~~~~~~~~
%Error: /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/cfu.v:5693:3: Cannot find file containing module: 'MULTADDSUB9X9WIDE'
 5693 |   MULTADDSUB9X9WIDE #(
      |   ^~~~~~~~~~~~~~~~~
%Error: /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/cfu.v:5484:3: Cannot find file containing module: 'MULTADDSUB9X9WIDE'
 5484 |   MULTADDSUB9X9WIDE #(
      |   ^~~~~~~~~~~~~~~~~
%Error: /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/cfu.v:5272:3: Cannot find file containing module: 'MULTADDSUB9X9WIDE'
 5272 |   MULTADDSUB9X9WIDE #(
      |   ^~~~~~~~~~~~~~~~~
%Error: /usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/cfu.v:5063:3: Cannot find file containing module: 'MULTADDSUB9X9WIDE'
 5063 |   MULTADDSUB9X9WIDE #(
      |   ^~~~~~~~~~~~~~~~~
%Error: Exiting due to 8 error(s)

CMake Error at /usr/share/verilator/verilator-config.cmake:212 (message):
  Verilator command failed (return code=1)
Call Stack (most recent call first):
  CMakeLists.txt:30 (verilate)

-- Configuring incomplete, errors occurred!
See also "/usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/build/renode/CMakeFiles/CMakeOutput.log".
See also "/usr/local/google/home/avg/src/CFU-Playground/proj/hps_accel/build/renode/CMakeFiles/CMakeError.log".
make: *** [../proj.mk:205: renode-scripts] Error 1
```

tcal-x, 2022-04-22:
I believe this is not supported currently.   The immediate issue here could be resolved by removing the `--specialize-nx` so that the Nexus primitives are not directly instantiated.  

But the larger issue is modeling the extra memory ports to the Activation Arena -- this is modeled in the 100% Verilator simulation, but not in the Renode simulation.

alanvgreen, 2022-04-28:
A

alanvgreen, 2022-04-28:
Nope. Not sure what happened.