google/CFU-Playground #561: proj_template: make renode-headless fails

alanvgreen, 2022-04-28:
I updated to ToT, ran `scripts/setup` and reran `scripts/install_oxide`

`make renode-headless` fails with this error:

```
~/src/CFU-Playground/proj/proj_template/build/renode ~/src/CFU-Playground/proj/proj_template
CMake Warning at CMakeLists.txt:20 (find_package):
  By not providing "Findverilator.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "verilator", but CMake did not find one.

  Could not find a package configuration file provided by "verilator" with
  any of the following names:

    verilatorConfig.cmake
    verilator-config.cmake

  Add the installation prefix of "verilator" to CMAKE_PREFIX_PATH or set
  "verilator_DIR" to a directory containing one of the above files.  If
  "verilator" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at CMakeLists.txt:30 (verilate):
  Unknown CMake command "verilate".

-- Configuring incomplete, errors occurred!
See also "/home/avg/src/CFU-Playground/proj/proj_template/build/renode/CMakeFiles/CMakeOutput.log".
```

tcal-x, 2022-04-28:
I gave this a quick try (on gLinux), and it worked for me, with one difference -- I didn't run `install_oxide` and I have nothing under `third_party/usr/local`.   I'll run `install_oxide` and see if it makes a difference.

tcal-x, 2022-04-28:
Hmm, even after doing `install_oxide`, it still works for me.   I do need to hit ENTER to get the menu, but then it runs.  To exit, hit ESC, then type "quit"\<enter\>.

Oh, maybe try this at your  end:   remove your current renode `rm -rf third_party/renode`, then rerun `./scripts/setup` to get the current version.

alanvgreen, 2022-04-28:
I did have a different renode version, but even after updating I'm getting the same error.

I am working on ChromeOS, so perhaps my verilator is too old? It appears to be 4.010-1.

```
$ sudo apt install verilator
Reading package lists... Done
Building dependency tree       
Reading state information... Done
verilator is already the newest version (4.010-1).
The following packages were automatically installed and are no longer required:
  berkeley-abc gir1.2-atk-1.0 gir1.2-freedesktop gir1.2-gdkpixbuf-2.0 gir1.2-gtk-3.0 gir1.2-pango-1.0 libpangoxft-1.0-0 python3-cairo python3-gi-cairo xdot
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

I'll keep looking.

tcal-x, 2022-04-28:
I have
```
verilator is already the newest version (4.038-1).
```
so that could be it.   I can also try reproducing on my Pixelbook.

tcal-x, 2022-04-28:
On ChromeOS, it works for me.
But, I have Verilator 4.038-1.
I'm on ChromeOS Version 102.0.5005.6 (Official Build) dev (64-bit).