google/CFU-Playground #568: Colab: Verilator compile in Conda env can't find the compiler

tcal-x, 2022-05-04:
To reproduce: go to this colab: https://colab.research.google.com/drive/1_GlX-pO4rune8GMIK4q_IuhxpIhnPTni?resourcekey=0-jMbF8wzg0csZu0fxpOy78A&usp=sharing (currently only readable within Google)

Then select "Runtime" --> "Run all"

Output:
```
make -j -C /content/CFU-Playground/soc/build/sim.proj_template_v/gateware/obj_dir -f Vsim.mk Vsim
make[3]: Entering directory '/content/CFU-Playground/soc/build/sim.proj_template_v/gateware'
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
Vsim.mk:13: warning: NUL character seen; rest of line ignored
ccache x86_64-conda-linux-gnu-c++ -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /content/CFU-Playground/env/conda/envs/cfu-common/include -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -isystem /content/CFU-Playground/env/conda/envs/cfu-common/include -I.  -MMD -I/content/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include -I/content/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /content/CFU-Playground/env/conda/envs/cfu-common/include -ggdb -Wall -O0   -DTRACE_FST -I/content/CFU-Playground/third_party/python/litex/litex/build/sim/core  -std=gnu++14 -Os -c -o veril.o /content/CFU-Playground/third_party/python/litex/litex/build/sim/core/veril.cpp
ccache: error: Could not find compiler "x86_64-conda-linux-gnu-c++" in PATH
Vsim.mk:71: recipe for target 'veril.o' failed
make[3]: *** [veril.o] Error 1
make[3]: Leaving directory '/content/CFU-Playground/soc/build/sim.proj_template_v/gateware/obj_dir'
/content/CFU-Playground/third_party/python/litex/litex/build/sim/core/Makefile:38: recipe for target 'sim' failed
make[2]: *** [sim] Error 2
make[2]: *** Waiting for unfinished jobs....
make[4]: Leaving directory '/content/CFU-Playground/soc/build/sim.proj_template_v/gateware/modules/gmii_ethernet'
cp gmii_ethernet/gmii_ethernet.so gmii_ethernet.so
make[3]: Leaving directory '/content/CFU-Playground/soc/build/sim.proj_template_v/gateware/modules'
make[2]: Leaving directory '/content/CFU-Playground/soc/build/sim.proj_template_v/gateware'
/content/CFU-Playground/soc/sim.mk:56: recipe for target 'run' failed
make[1]: *** [run] Error 1
make[1]: Leaving directory '/content/CFU-Playground/soc'
../proj.mk:354: recipe for target 'load' failed
make: *** [load] Error 2
```

To highlight the error:
```
ccache: error: Could not find compiler "x86_64-conda-linux-gnu-c++" in PATH
```

Looking under `env/conda/envs/cfu-common/bin/`, I see:
```
/content/CFU-Playground/env/conda/bin/x86_64-conda-linux-gnu-ld
/content/CFU-Playground/env/conda/bin/x86_64-conda_cos6-linux-gnu-ld
```
and nothing else with an `x86_64` prefix.

tcal-x, 2022-05-04:
Looking at the earlier warning
```
Vsim.mk:13: warning: NUL character seen; rest of line ignored
```

...that line in `/content/CFU-Playground/soc/build/sim.proj_template_v/gateware/obj_dir/Vsim.mk`  _is_ suspicious:
```
VERILATOR_ROOT = /content/CFU-Playground/env/conda/envs/cfu-common/share/verilator                                                                                                                                                                                                              
```

tcal-x, 2022-05-04:
There is a similar issue when running locally directly (not in colab) in the Conda env.   The error is still
```
ccache: error: Could not find compiler "x86_64-conda-linux-gnu-c++" in PATH
```

However, looking in `$CFU_ROOT/env/conda/envs/cfu-common/bin`, there are now many executables, but with a slightly different prefix including "`cos6`":
```
x86_64-conda_cos6-linux-gnu-addr2line     x86_64-conda_cos6-linux-gnu-g++         x86_64-conda_cos6-linux-gnu-ld.bfd
x86_64-conda_cos6-linux-gnu-ar            x86_64-conda_cos6-linux-gnu-gcc         x86_64-conda_cos6-linux-gnu-ld.gold
x86_64-conda_cos6-linux-gnu-as            x86_64-conda_cos6-linux-gnu-gcc-ar      x86_64-conda_cos6-linux-gnu-nm
x86_64-conda_cos6-linux-gnu-c++           x86_64-conda_cos6-linux-gnu-gcc-nm      x86_64-conda_cos6-linux-gnu-objcopy
x86_64-conda_cos6-linux-gnu-cc            x86_64-conda_cos6-linux-gnu-gcc-ranlib  x86_64-conda_cos6-linux-gnu-objdump
x86_64-conda_cos6-linux-gnu-c++filt       x86_64-conda_cos6-linux-gnu-gcov        x86_64-conda_cos6-linux-gnu-ranlib
x86_64-conda_cos6-linux-gnu-cpp           x86_64-conda_cos6-linux-gnu-gcov-dump   x86_64-conda_cos6-linux-gnu-readelf
x86_64-conda_cos6-linux-gnu-ct-ng.config  x86_64-conda_cos6-linux-gnu-gcov-tool   x86_64-conda_cos6-linux-gnu-size
x86_64-conda_cos6-linux-gnu-dwp           x86_64-conda_cos6-linux-gnu-gprof       x86_64-conda_cos6-linux-gnu-strings
x86_64-conda_cos6-linux-gnu-elfedit       x86_64-conda_cos6-linux-gnu-ld          x86_64-conda_cos6-linux-gnu-strip
```
To reproduce locally, set up CFU Playground according to instructions, then:
```
make env
make enter
cd proj/proj_template_v
make PLATFORM=sim load
```

tcal-x, 2022-05-04:
This looks similar: https://stackoverflow.com/a/46457813 -- although that is more than four years old -- you'd think they'd have fixed the issue by now.

mithro, 2022-05-05:
@PiotrZierhoffer - Any idea what is going on here?

tcal-x, 2022-05-05:
@PiotrZierhoffer I gave you edit permission on the colab.

I added symbolic links in the colab to provide the executables where the build expects them to be.  I also figured out that the x86_64 compiler is gcc 7.3, which is quite old and may be related to the **new error** after adding the links:

```
x86_64-conda-linux-gnu-c++: error: unrecognized command line option '-fcf-protection=none'; did you mean '-flto-partition=none'?
Vsim.mk:71: recipe for target 'veril.o' failed
make[3]: *** [veril.o] Error 1
```

To see the old behavior, comment out the `ln -s ...` lines in the last code cell.

P.S. now in the colab, I see the same behavior as "local" described above (many x86_64 executables, all with `cos6` in their name).   I don't know what changed!

tcal-x, 2022-05-05:
I may have found a mistake on my part: I was using `source env/conda/bin/activate && ....` when I should have been doing `source env/conda/bin/activate cfu-common && ....`.    I'm rerunning now; we'll see if it makes a difference.

tcal-x, 2022-06-16:
This is a candidate workaround for this issue (for when running locally, not in colab):

0) Install Verilator in your system via `sudo apt get verilator`.
1) don't be in the conda shell (if you have typed `make enter`, then type `exit`).
2) delete the current Conda env: from `$CFU_ROOT`, `rm -rf env/`
3) edit `./conf/environment.yml` to remove or comment out (#) the "verilator" line
4) rebuild the env: `make env`
5) enter the new env: `make enter`

Now if you cd to a project and "make PLATFORM=sim load", it should work; at least it works for me.

aman26kbm, 2022-06-17:
I'm seeing this same issue. When I use Tim's workaround mentioned in the last comment, I get this error instead:

```
OSError: Unable to find any of the cross compilation toolchains:
- riscv64-unknown-elf
- riscv64-unknown-linux-gnu
- riscv64-elf
- riscv64-linux
- riscv64-linux-gnu
- riscv-sifive-elf
- riscv64-none-elf
- riscv32-unknown-elf
- riscv32-unknown-linux-gnu
- riscv32-elf
- riscv-none-embed
- riscv-none-elf
```

tcal-x, 2022-06-17:
@aman26kbm , if you did re-enter the environment after rebuilding (after commenting out Verilator), then you should get a usable cross compiler.   Can you try typing "riscv" and then auto-complete (usually tab) to see what is in your path with that prefix?

aman26kbm, 2022-06-22:
Thanks for your message, @tcal-x .

For some reason, `make enter` doesn't seem to work for me. I have to activate the environment manually. When I do that, I don't get the missing riscv toolchain error, but I get another verilator related error.

I am moving to another machine for multiple reasons. If I still see issues, I'll report back.

aman26kbm, 2022-06-22:
The verilator error was related to the version of verilator that got installed when I did "sudo apt-get install verilator" (it was 3.something). I ended up cloning verilator, pulling tag v4.038 (to match the version you had mentioned earlier in your message) and then building and installing it. After that `make PLATFORM=sim load` worked.

tcal-x, 2022-07-07:
I added https://github.com/tcal-x/verilator-make-env-test as a minimal testcase showing the issue.

ajelinski, 2022-07-08:
Hi @tcal-x! Thanks for the minimal testcase, it was super useful for debugging this problem.

I've researched the problem in a clean `debian:bullseye` container and it seems by default Conda installs an old compiler package (v7.3.0) which doesn't contain the `x86_64-conda-...`-prefixed executables, just the `x86_64-conda_cos6-...` ones. The g++ 8+ Conda compiler packages contain both (the `_cosX` ones are links).

There's no problem to install the latest compiler package right after using `make env && make enter` which is the simplest workaround:
```
root@01fd5943cb29:~/verilator-make-env-test# conda install 'gxx_linux-64>11' 

(...)

The following packages will be UPDATED:

  binutils_impl_lin~                      2.33.1-he6710b0_7 --> 2.38-h2a08ee3_1
  binutils_linux-64                      2.33.1-h9595d00_15 --> 2.38.0-hc2dff05_0
  gcc_impl_linux-64                        7.3.0-habb00fd_1 --> 11.2.0-h1234567_1
  gcc_linux-64                            7.3.0-h553295d_15 --> 11.2.0-h5c386dc_0
  gxx_impl_linux-64                        7.3.0-hdf63c60_1 --> 11.2.0-h1234567_1
  gxx_linux-64                            7.3.0-h553295d_15 --> 11.2.0-hc2dff05_0
  ld_impl_linux-64                        2.33.1-h53a641e_7 --> 2.38-h1181459_1

Preparing transaction: done
Verifying transaction: done
Executing transaction: done

root@01fd5943cb29:~/verilator-make-env-test# cd test/

root@01fd5943cb29:~/verilator-make-env-test/test# make
make[4]: Entering directory '/root/verilator-make-env-test/test'
verilator -cc add.v
make -C obj_dir -f Vadd.mk
make[5]: Entering directory '/root/verilator-make-env-test/test/obj_dir'
Vadd.mk:13: warning: NUL character seen; rest of line ignored
perl /root/verilator-make-env-test/env/conda/envs/verilator-make-env/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vadd.cpp Vadd___024root__DepSet_h53499b2f__0.cpp Vadd___024root__Slow.cpp Vadd___024root__DepSet_h53499b2f__0__Slow.cpp Vadd__Syms.cpp > Vadd__ALL.cpp
x86_64-conda-linux-gnu-c++ -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /root/verilator-make-env-test/env/conda/envs/verilator-make-env/include -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -isystem /root/verilator-make-env-test/env/conda/envs/verilator-make-env/include -I.  -MMD -I/root/verilator-make-env-test/env/conda/envs/verilator-make-env/share/verilator/include -I/root/verilator-make-env-test/env/conda/envs/verilator-make-env/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o Vadd__ALL.o Vadd__ALL.cpp
echo "" > Vadd__ALL.verilator_deplist.tmp
Archive x86_64-conda-linux-gnu-ar -rcs Vadd__ALL.a Vadd__ALL.o
rm Vadd__ALL.verilator_deplist.tmp
make[5]: Leaving directory '/root/verilator-make-env-test/test/obj_dir'
make[4]: Leaving directory '/root/verilator-make-env-test/test'
```
(please notice the Verilator isn't downgraded in the process so it doesn't conflict with the latest compiler packages)

I'm going to make the new Verilator packages require the compiler to be v8+ on Linux because obviously it can't cooperate with the older ones and, if there won't be any additional problems, the fixed Verilator package might be available even today.

You might want to add `gxx_linux-64 >11` package to the `environment.yml` files to fix the problem. It should help until the fixed Verilator v2.224 package gets uploaded.

Btw:
1. The latest compiler package contains `_cos7` links since the compiler is based on CentOS 7: https://anaconda.org/anaconda/gxx_impl_linux-64/11.2.0/download/linux-64/gxx_impl_linux-64-11.2.0-h1234567_1.tar.bz2
   so making the Verilator use `_cosX`-prefixed compiler wouldn't be a good option.
2. From the download counts, it seems the v7.3.0 package is the Conda's favourite for some mysterious reasons: https://anaconda.org/anaconda/gxx_impl_linux-64/files
   For "normal" packages, always the one with the "highest" version string is used unless it conflicts with other packages so it might be somehow forced to be the default one.

tcal-x, 2022-07-28:
Current status -- still broken:
```
x86_64-conda-linux-gnu-c++ -Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,-rpath,/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/lib -Wl,-rpath-link,/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/lib -L/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/lib -lpthread -Wl,--no-as-needed -ljson-c -lz -lm -lstdc++ -Wl,--no-as-needed -ldl -levent    veril.o sim_init.o verilated.o verilated_dpi.o verilated_vcd_c.o Vsim__ALL.a   modules.o pads.o sim.o libdylib.o parse.o -Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,-rpath,/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/lib -Wl,-rpath-link,/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/lib -L/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/lib -lpthread -Wl,--no-as-needed -ljson-c -lz -lm -lstdc++ -Wl,--no-as-needed -ldl -levent    -o Vsim
/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/bin/../lib/gcc/x86_64-conda-linux-gnu/11.2.0/../../../../x86_64-conda-linux-gnu/bin/ld: /media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/lib/libz.so: undefined reference to `memcpy@GLIBC_2.14'
/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/bin/../lib/gcc/x86_64-conda-linux-gnu/11.2.0/../../../../x86_64-conda-linux-gnu/bin/ld: /media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/lib/libstdc++.so: undefined reference to `aligned_alloc@GLIBC_2.16'
/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/bin/../lib/gcc/x86_64-conda-linux-gnu/11.2.0/../../../../x86_64-conda-linux-gnu/bin/ld: /media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/lib/libstdc++.so: undefined reference to `clock_gettime@GLIBC_2.17'
collect2: error: ld returned 1 exit status
make[4]: *** [Vsim.mk:76: Vsim] Error 1
rm Vsim__ALL.verilator_deplist.tmp
make[4]: Leaving directory '/media/tim/GIT/tcal-x/CFU-Playground/soc/build/sim.proj_template_v/gateware/obj_dir'
make[3]: *** [/media/tim/GIT/tcal-x/CFU-Playground/third_party/python/litex/litex/build/sim/core/Makefile:39: sim] Error 2
make[3]: Leaving directory '/media/tim/GIT/tcal-x/CFU-Playground/soc/build/sim.proj_template_v/gateware'
make[2]: *** [/media/tim/GIT/tcal-x/CFU-Playground/soc/sim.mk:56: run] Error 1
make[2]: Leaving directory '/media/tim/GIT/tcal-x/CFU-Playground/soc'
make[1]: *** [../proj.mk:367: load] Error 2
make[1]: Leaving directory '/media/tim/GIT/tcal-x/CFU-Playground/proj/proj_template_v'
```
That is without any modification / pinning in the `environment.yml` file.

tcal-x, 2022-07-28:
Note: it does work correctly on my work laptop (gLinux).   So yes, confirming that it is dependent on the host system environment.

** I did need to add `libevent` and `json-c` to `conf/environment.yml` **   I'll open a PR to make that change.

tcal-x, 2022-07-28:
@ajelinski ping -- hopefully you have access to an Ubuntu 20.04 system to take a look.

I'm fine with pinning the compiler/library versions if you can find a combination that works in CFU Playground (I wasn't able to).

tcal-x, 2022-07-28:
@ajelinski @PiotrZierhoffer important update -- I am playing with a conda-verilator job in CI (running on GHA's default Ubuntu 20.04), and I DON'T see the issue there.   So there may be something messed up with my laptop's Ubuntu installation.

It would of course be preferable that there was no dependence on the host system at all.   But, for now, we can treat this as lower priority.

ajelinski, 2022-07-29:
It's really weird glibc 2.14/2.16/2.17 errors occurred on Ubuntu 20.04 since it seems to have been released with glibc 2.31 from the very beginning... I have really no idea what could cause that.

I tested CFU with a fresh Ubuntu 20.04 docker container and found a single irrelevant minor issue running commands from the workflow: https://github.com/tcal-x/CFU-Playground/commit/73e2256af43004ce2f1563fa91fd40fae26bf36c -- I had to remove `sudo` for the automatic installation of the missing packages to work here: https://github.com/google/CFU-Playground/blob/main/scripts/setup#L71

Apart from that, I could run the commands without any surprises and both `make PLATFORM=sim RUN_MENU_ITEMS="5" run` and `make renode` worked (though without GUI it's better to use `make renode-headless` which also worked).

As for the dependence on the host system it's just the Conda's secret requirement that it requires glibc >= 2.17 and even if we modify Verilator then there will be problems with many other packages. The error you had seems to be about `ld`.

Btw. I think this issue can be closed because the compiler troubles caused by Verilator have been solved.

tcal-x, 2022-07-29:
## Findings

Not all Ubuntu 20.04.4 LTS  systems are the same. My laptop has kernel `5.4.0-122-generic`, while Github Actions runners and GCP (both 20.04.4) have kernels `5.15.0-1014-azure` and `5.15.0-1013-gcp` respectively.

Github and GCP hosts with the 5.15 kernel **do not see this issue**; Verilator compiles and runs fine in Conda.
It's only on my laptop with the 5.4 kernel.

I diffed the Conda package downloads and found this:
```diff
122,123c122,125
< sysroot_linux-64-2.12-he073ed8_15
< sysroot_linux-64-2.12-he073ed8_15.tar.bz2
---
> sysroot_linux-64-2.17-h57e8cba_10
> sysroot_linux-64-2.17-h57e8cba_10.tar.bz2
> _sysroot_linux-64_curr_repodata_hack-3-haa98f57_10
> _sysroot_linux-64_curr_repodata_hack-3-haa98f57_10.tar.bz2
```
and also found this difference:

my laptop:  `env/conda/envs/cfu-common/x86_64-conda-linux-gnu/sysroot/lib/libc-2.12.so`
Github/GCP: `env/conda/envs/cfu-common/x86_64-conda-linux-gnu/sysroot/lib/libc-2.17.so`

So, it's **not** using a host `libc.so`; Conda is downloading different libc versions for some reason.

I tried the simple fix, adding this to `environment.yml` on my laptop to try to force 2.17:
```
- sysroot_linux-64=2.17
```
...and it works!  It also fixes #629 (similar issue running renode).

We need to make sure this change works correctly on Ubuntu 18 (used by colab).

tcal-x, 2022-07-29:
Hi @ajelinski , I thought I had submitted the above comment last night, but I hadn't.   Anyway, I think I got to the bottom of it; at least the fix #631 is working for me on my laptop, and it doesn't break things anywhere else that I've tested it.

ajelinski, 2022-07-29:
Oh, that's an interesting finding. So it seems Conda makes a mess by using its own `libc` if it's available.

I call it a mess because the `sysroot_linux` package you mentioned is mostly installed with the compiler packages so it's used during linking instead of possibly-the-very-latest host libc. It isn't required by other packages. For example, if you check the packages in the `base` environment after installing Conda you won't find `sysroot_linux` there even though there are `libstdcxx-ng`, Python etc. packages which all won't work without `libc`.

Nevertheless, this won't be an issue in any new installation because the old package has been removed and isn't available anymore: https://anaconda.org/main/sysroot_linux-64/files

ajelinski, 2022-07-29:
Btw. I've just created a PR for the `sudo` problem: https://github.com/google/CFU-Playground/pull/632

tcal-x, 2022-08-01:
The remaining issue seems to be specific to my laptop (it has not been reproduced on other Ubuntu 20.04 systems), so I'll close this as not planned.

closed 2022-08-01 by tcal-x