google/CFU-Playground #574: Program Not Working/Flashing On Tang Nano 9K

minexo79, 2022-05-07:
I'm Try To Run CFU-Playground in Tang Nano 9K, But It's Stucked.
The Command:
```
make load TTY=/dev/ttyUSB1 TARGET=sipeed_tang_nano_9k
```
The Output:
```
/home/minexo79/CodeWorkspace/CFU-Playground/scripts/pyrun /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/cfu_gen.py 
make -C /home/minexo79/CodeWorkspace/CFU-Playground/soc -f /home/minexo79/CodeWorkspace/CFU-Playground/soc/common_soc.mk litex-software
make[1]: Entering directory '/home/minexo79/CodeWorkspace/CFU-Playground/soc'
make[1]: Nothing to be done for 'litex-software'.
make[1]: Leaving directory '/home/minexo79/CodeWorkspace/CFU-Playground/soc'
Copying tflite-micro files
for d in tensorflow/lite tensorflow/lite/c tensorflow/lite/core/api tensorflow/lite/kernels tensorflow/lite/kernels/internal tensorflow/lite/kernels/internal/optimized tensorflow/lite/kernels/internal/reference tensorflow/lite/kernels/internal/reference/integer_ops tensorflow/lite/micro tensorflow/lite/micro/kernels tensorflow/lite/micro/memory_planner tensorflow/lite/micro/arena_allocator tensorflow/lite/schema; do \
        mkdir -p /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/$d; \
        /bin/cp -a `find /home/minexo79/CodeWorkspace/CFU-Playground/third_party/tflite-micro/$d -maxdepth 1 -type f -not -name '*_test*' -regex '.*\.\(h\|c\|cc\)'` /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/$d; \
done
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/third_party/tflite-micro/tensorflow/lite/micro/kernels/conv_test* /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/tensorflow/lite/micro/kernels
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/third_party/tflite-micro/tensorflow/lite/micro/kernels/depthwise_conv_test* /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/tensorflow/lite/micro/kernels
/bin/cp: missing destination file operand after '/home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/tensorflow/lite/micro/models'
Try '/bin/cp --help' for more information.
mkdir -p /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/tensorflow/lite/micro/examples/person_detection
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/third_party/tflite-micro/tensorflow/lite/micro/examples/person_detection/model_settings* /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/tensorflow/lite/micro/examples/person_detection
TfLM: copying selected third_party files
mkdir -p /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/third_party/gemmlowp
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/third_party/tflite-micro/third_party/gemmlowp/fixedpoint /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/third_party/gemmlowp
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/third_party/tflite-micro/third_party/gemmlowp/internal /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/third_party/internal
mkdir -p /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/third_party/flatbuffers/include
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/third_party/tflite-micro/third_party/flatbuffers/include/* /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/third_party/flatbuffers/include
mkdir -p /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/third_party/ruy/ruy/profiler
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/third_party/tflite-micro/third_party/ruy/ruy/profiler/instrumentation.h /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src/third_party/ruy/ruy/profiler
build-dir: copying source to build dir
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/common/*              /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/third_party/mlcommons/*       /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/third_party/SaxonSoc/riscv.h     /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src
/bin/cp -a /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/src/*                 /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/src
/bin/rm -rf                                  /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/_*
make -C /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build all -j 8
make[1]: Entering directory '/home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build'
Running interactively on FPGA Board
make -C /home/minexo79/CodeWorkspace/CFU-Playground/soc -f /home/minexo79/CodeWorkspace/CFU-Playground/soc/common_soc.mk load_hook
make[1]: Entering directory '/home/minexo79/CodeWorkspace/CFU-Playground/soc'
MAKEFLAGS=-j8 /home/minexo79/CodeWorkspace/CFU-Playground/scripts/pyrun ./common_soc.py --output-dir build/sipeed_tang_nano_9k.proj_template --csr-json build/sipeed_tang_nano_9k.proj_template/csr.json --cpu-cfu  /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/cfu.v --uart-baudrate 11200 --target sipeed_tang_nano_9k  --software-load --software-path /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/software.bin
make[1]: Leaving directory '/home/minexo79/CodeWorkspace/CFU-Playground/soc'
/home/minexo79/CodeWorkspace/CFU-Playground/soc/bin/litex_term --speed 11200  --kernel /home/minexo79/CodeWorkspace/CFU-Playground/proj/proj_template/build/software.bin /dev/ttyUSB1
```
The Program Was Stucked At Last Output.
I'm Also Try To Tun Same Command With Another Com Port(`/dev/ttyUSB0`), But Result is same:
```
make load TTY=/dev/ttyUSB0 TARGET=sipeed_tang_nano_9k
```

Oliver0804, 2022-05-09:
First, u should confirm the existence of your board.
$ls /dev/tty.usb
or 
$ls /dev/ttyUSB*
u need check your serialprot shows in the system.

tcal-x, 2022-05-09:
@minexo79 thanks for giving this a try!   Some boards have their quirks and need some special handling.

Have you built for the board in the basic LiteX environment (outside of CFU Playground) and interacted with the BIOS?

The next step would be to try running the LiteX bare metal demo, to confirm that software can be loaded by the BIOS.

minexo79, 2022-05-09:
@tcal-x Thanks your helping! Next time I will try.

minexo79, 2022-05-11:
@tcal-x I Found the problem!
The CFU Playground Command `make prog` just **only** flash bitstream into sram, but LiteX needs bios/bootloader.
In basic LiteX, Type `--flash` can flash bitstream to embbed flash and bios to external flash.

tcal-x, 2022-05-11:
@minexo79 Ok, then it's a bit like Fomu.   With Fomu, we flash the bitstream and the binary together (to the same external flash).    

There is board-specific code to handle it here:
https://github.com/google/CFU-Playground/blob/main/soc/board_specific_workflows/kosagi_fomu.py 

I'm not sure if the exact same approach (building a single image to flash) will work with the Tang Nano 9k, but perhaps it is a good starting point.

Or, we might change CFU Playground to directly use the LiteX flash utility for the 9k board.