google/CFU-Playground #599: Running `make PLATFORM=sim load` out of the box to leverage Verilator fails

ShvetankPrakash, 2022-06-08:
After cloning the repo, running the setup script, installing the env, and then running `make PLATFORM=sim load` I obtain the following error:
```
ccache x86_64-conda-linux-gnu-c++ -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/sprakash/Documents/Repos/CFU-Playground/env/conda/envs/cfu-symbiflow/include -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -isystem /home/sprakash/Documents/Repos/CFU-Playground/env/conda/envs/cfu-symbiflow/include -I.  -MMD -I/home/sprakash/Documents/Repos/CFU-Playground/env/conda/envs/cfu-symbiflow/share/verilator/include -I/home/sprakash/Documents/Repos/CFU-Playground/env/conda/envs/cfu-symbiflow/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/sprakash/Documents/Repos/CFU-Playground/env/conda/envs/cfu-symbiflow/include -ggdb -Wall -O0   -DTRACE_FST -I/home/sprakash/Documents/Repos/CFU-Playground/third_party/python/litex/litex/build/sim/core  -std=gnu++14 -Os -c -o sim_init.o ../sim_init.cpp
ccache: error: Could not find compiler "x86_64-conda-linux-gnu-c++" in PATH
make[5]: *** [Vsim.mk:71: veril.o] Error 1
make[5]: *** Waiting for unfinished jobs....
ccache: error: Could not find compiler "x86_64-conda-linux-gnu-c++" in PATH
make[5]: *** [Vsim.mk:73: sim_init.o] Error 1
make[5]: Leaving directory '/home/sprakash/Documents/Repos/CFU-Playground/soc/build/sim.proj_template/gateware/obj_dir'
make[4]: *** [/home/sprakash/Documents/Repos/CFU-Playground/third_party/python/litex/litex/build/sim/core/Makefile:39: sim] Error 2
make[4]: Leaving directory '/home/sprakash/Documents/Repos/CFU-Playground/soc/build/sim.proj_template/gateware'
make[3]: *** [/home/sprakash/Documents/Repos/CFU-Playground/soc/sim.mk:56: run] Error 1
make[3]: Leaving directory '/home/sprakash/Documents/Repos/CFU-Playground/soc'
make[2]: *** [../proj.mk:367: load] Error 2
make[2]: Leaving directory '/home/sprakash/Documents/Repos/CFU-Playground/proj/proj_template'
```
It appears the compiler it is looking for is not there (i.e. `x86_64-conda-linux-gnu-c++`)?
I was only able to find `x86_64-conda_cos6-linux-gnu-c++` in the env when I looked...is this is what is should be trying to compile with?

tcal-x, 2022-06-09:
Possible duplicate of #568.

ShvetankPrakash, 2022-06-09:
Looking at #568 it is a duplicate, thanks for noticing! Looking at that thread there seems to be no update/resolution?

tcal-x, 2022-06-09:
Correct, no known fix or workaround.

Well, the workaround is to use Verilator outside of the Conda env.   I wonder what would happen if we took Verilator out of the conda configuration, so that it had to use the system Verilator?   It might still trip up on the conda x86 compiler issue.

tcal-x, 2022-06-09:
Closing as duplicate of #568.

closed 2022-06-09 by tcal-x