google/CFU-Playground #605: yosys: error while loading shared libraries: libffi.so.7:

MichaelinChrist, 2022-06-22:
When I ran the command below, the following error occurred. How can I get rid of this? 
mji@XPS-8930-5:~/CFU-Playground/proj/proj_template$ **make prog TARGET=digilent_arty USE_SYMBIFLOW=1**
make[2]: Entering directory '/home/mji/CFU-Playground/proj/proj_template'
/home/mji/CFU-Playground/scripts/pyrun /home/mji/CFU-Playground/proj/proj_template/cfu_gen.py 
Traceback (most recent call last):
  File "/home/mji/CFU-Playground/proj/proj_template/cfu_gen.py", line 38, in <module>
    main()
  File "/home/mji/CFU-Playground/proj/proj_template/cfu_gen.py", line 31, in main
    new_verilog = verilog.convert(cfu, name='Cfu', ports=cfu.ports)
  File "/home/mji/CFU-Playground/third_party/python/amaranth/amaranth/back/verilog.py", line 51, in convert
    verilog_text, name_map = convert_fragment(fragment, name, emit_src=emit_src, strip_internal_attrs=strip_internal_attrs)
  File "/home/mji/CFU-Playground/third_party/python/amaranth/amaranth/back/verilog.py", line 42, in convert_fragment
    return _convert_rtlil_text(rtlil_text, strip_internal_attrs=strip_internal_attrs), name_map
  File "/home/mji/CFU-Playground/third_party/python/amaranth/amaranth/back/verilog.py", line 13, in _convert_rtlil_text
    yosys = find_yosys(lambda ver: ver >= (0, 10))
  File "/home/mji/CFU-Playground/third_party/python/amaranth/amaranth/_toolchain/yosys.py", line 220, in find_yosys
    version = proxy.version()
  File "/home/mji/CFU-Playground/third_party/python/amaranth/amaranth/_toolchain/yosys.py", line 157, in version
    version = cls.run(["-V"])
  File "/home/mji/CFU-Playground/third_party/python/amaranth/amaranth/_toolchain/yosys.py", line 186, in run
    return cls._process_result(popen.returncode, stdout, stderr, ignore_warnings, src_loc_at)
  File "/home/mji/CFU-Playground/third_party/python/amaranth/amaranth/_toolchain/yosys.py", line 108, in _process_result
    raise YosysError(stderr.strip())
**amaranth._toolchain.yosys.YosysError: yosys: error while loading shared libraries: libffi.so.7: cannot open shared object file: No such file or directory**
../proj.mk:241: recipe for target 'generate_cfu' failed
make[2]: *** [generate_cfu] Error 1
make[2]: Leaving directory '/home/mji/CFU-Playground/proj/proj_template'

MichaelinChrist, 2022-06-22:
It seems I need to follow the instructions at the link below to install libffi7. On my machine previously it had libffi6 only. 
https://askubuntu.com/questions/1286772/libffi-so-7-cannot-open-shared-object-file-no-such-file-or-directory

Once libffi7 was installed, this error went away. But the following error occurred:
INFO:SoC:--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "./common_soc.py", line 57, in <module>
    main()
  File "./common_soc.py", line 53, in main
    workflow.run()
  File "/home/mji/CFU-Playground/soc/board_specific_workflows/general.py", line 125, in run
    soc_builder = self.build_soc(soc)
  File "/home/mji/CFU-Playground/soc/board_specific_workflows/digilent_arty.py", line 73, in build_soc
    return super().build_soc(soc, **kwargs)
  File "/home/mji/CFU-Playground/soc/board_specific_workflows/general.py", line 102, in build_soc
    soc_builder.build(run=self.args.build, **kwargs)
  File "/home/mji/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 320, in build
    self._generate_includes(with_bios=with_bios)
  File "/home/mji/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 175, in _generate_includes
    variables_contents = self._get_variables_contents()
  File "/home/mji/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 145, in _get_variables_contents
    for k, v in export.get_cpu_mak(self.soc.cpu, self.compile_software):
  File "/home/mji/CFU-Playground/third_party/python/litex/litex/soc/integration/export.py", line 94, in get_cpu_mak
    ("TRIPLE",        select_triple(triple)),
  File "/home/mji/CFU-Playground/third_party/python/litex/litex/soc/integration/export.py", line 89, in select_triple
    raise OSError(msg)
******OSError: Unable to find any of the cross compilation toolchains:******
- riscv64-unknown-elf
- riscv64-unknown-linux-gnu
- riscv64-elf
- riscv64-linux
- riscv64-linux-gnu
- riscv-sifive-elf
- riscv64-none-elf
- riscv32-unknown-elf
- riscv32-unknown-linux-gnu
- riscv32-elf
- riscv-none-embed
- riscv-none-elf

/home/mji/CFU-Playground/soc/common_soc.mk:114: recipe for target 'build/digilent_arty.proj_template/gateware/digilent_arty.bit' failed
make[1]: *** [build/digilent_arty.proj_template/gateware/digilent_arty.bit] Error 1

tcal-x, 2022-06-23:
Hi @mjchipguru , thank you for the report!  I am on vacation right now without my laptop and so can only provide limited help.

I have seen the libffi.7 issue myself but I guess have always found a workaround since I don't see any other issues filed about it.  Hopefully here we can figure it out once and for all.

I'm guessing you are using Ubuntu 18.04 or older?

For your second issue about the riscv tool chain -- are you using the Conda-installed tools, or the tar archive from SiFive?  If you're using Conda "make env", you may have forgotten to re-enter the environment ("make enter") after fixing the libffi issue.

mjchipguru, 2022-06-24:
I am closing the issue.

closed 2022-06-24 by mjchipguru

wohlbier, 2022-06-24:
I am also seeing the same error. To work around it I installed `libffi7` on ubuntu 20.04.