google/CFU-Playground #612: Step-by-step adding a new model to CFU-Playground

hoanggiamta, 2022-07-14:
Hi all,
I follow the steps in https://github.com/google/CFU-Playground/issues/221 to add a new model to CFU-Playground. I create my_model folder and I copy "model_name.tflite" to "CFU-Playground/common/src/models/my_model/"     directory. But it seems that I need to create the 2 files: "model_name.h" and "model_name.cc" and place them in this folder, because I could see 2 files: "mnv2.h" and "mnv2.cc" in  "CFU-Playground/common/src/models/mnv2/". Could you please tell me how to generate the 2 files. Should I write by hands? Or is there any tools to generate them automatically based on my model  "model_name.tflite"? Thank you.

hoanggiamta, 2022-07-15:
Thank you @JosephBushagour. 
So I am modifying the mnv2.cc of the mnv2 model. For the model inference, they used  "tflite_set_input_unsigned(input_00001_18027);" to set input data for the model. I could see the file input_00001_18027.dat in common/src/models/mnv2/. But I don't know how to generate the .dat file from an original image like "1.jpg". Could you show me how to set input for the model?

JosephBushagour, 2022-07-15:
`.dat` files are converted directly into arrays of raw bytes at build time, see https://github.com/google/CFU-Playground/blob/main/common/Makefile#L175-L177. Whatever your model takes in as input can just be binary dumped to a corresponding `.dat` file.

tcal-x, 2022-07-15:
@hoanggiamta  see my comments here: #495 . Also https://github.com/google/CFU-Playground/blob/demo/proj/demo/images.readme has an example of creating the raw data file from an arbitrary image file --- this was for pdti8 not mnv2 so the data format may be different as well as dimensions -- the data file must match the expected dimensions exactly.

hoanggiamta, 2022-07-21:
Thank you, @JosephBushagour and @tcal-x. The issue was solved. The input image should be formatted as the same as for the the model running on host cpu.

closed 2022-07-21 by hoanggiamta