google/CFU-Playground #617: SymbiFlow not working in CFU Playground

tcal-x, 2022-07-22:
Before a recent change (#616), when I tried to install and use SymbiFlow as follows:
```
make install-sf
make enter-sf
cd proj/proj_template_v
make USE_SYMBIFLOW=1 bitstream
```
...I saw the following:
```
make[4]: Entering directory '/media/tim/GIT/google/CFU-Playground/soc/build/digilent_arty.proj_template_v/gateware'
symbiflow_synth -t digilent_arty -v /media/tim/GIT/google/CFU-Playground/proj/proj_template_v/cfu.v /media/tim/GIT/google/CFU-Playground/third_party/python/pythondata_cpu_vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv_FullCfu.v /media/tim/GIT/google/CFU-Playground/soc/build/digilent_arty.proj_template_v/gateware/digilent_arty.v -d artix7 -p xc7a35tcsg324-1 -x digilent_arty.xdc > /dev/null
/media/tim/GIT/google/CFU-Playground/env/symbiflow/share/symbiflow/techmaps/xc7_vpr/techmap/cells_map.v:0: ERROR: Can't find object for defparam `RAM_EXTENSION_A`!
make[4]: *** [Makefile:34: digilent_arty.eblif] Error 1
make[4]: Leaving directory '/media/tim/GIT/google/CFU-Playground/soc/build/digilent_arty.proj_template_v/gateware'
Traceback (most recent call last):
  File "./common_soc.py", line 57, in <module>
    main()
  File "./common_soc.py", line 53, in main
    workflow.run()
  File "/media/tim/GIT/google/CFU-Playground/soc/board_specific_workflows/general.py", line 125, in run
    soc_builder = self.build_soc(soc)
  File "/media/tim/GIT/google/CFU-Playground/soc/board_specific_workflows/digilent_arty.py", line 73, in build_soc
    return super().build_soc(soc, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/soc/board_specific_workflows/general.py", line 102, in build_soc
    soc_builder.build(run=self.args.build, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 350, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/soc/integration/soc.py", line 1147, in build
    return self.platform.build(self, *args, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/build/xilinx/platform.py", line 73, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/build/xilinx/symbiflow.py", line 241, in build
    _run_make()
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/build/xilinx/symbiflow.py", line 84, in _run_make
    raise OSError("Error occured during Symbiflow's script execution.")
OSError: Error occured during Symbiflow's script execution.
make[3]: *** [/media/tim/GIT/google/CFU-Playground/soc/common_soc.mk:115: build/digilent_arty.proj_template_v/gateware/digilent_arty.bit] Error 1
make[3]: Leaving directory '/media/tim/GIT/google/CFU-Playground/soc'
make[2]: *** [../proj.mk:313: bitstream] Error 2
make[2]: Leaving directory '/media/tim/GIT/google/CFU-Playground/proj/proj_template_v'
```
This might have been due to the Makefile explicitly downloading fairly old tars. *Although I'm puzzled, since the same error didn't occur in CI with the symbiflow.yml test*

After updating the tar urls (#616) to use the following:
```
F4PGA_TIMESTAMP='20220714-173445'
F4PGA_HASH='f7afc12'
```
and re-installing SymbiFlow, the symptom is then the missing scripts:
```
make[4]: Leaving directory '/media/tim/GIT/google/CFU-Playground/soc/build/digilent_arty.proj_template_v/software/bios'
INFO:SoC:Initializing ROM rom with contents (Size: 0x6380).
INFO:SoC:Auto-Resizing ROM rom from 0x20000 to 0x6380.
Traceback (most recent call last):
  File "./common_soc.py", line 57, in <module>
    main()
  File "./common_soc.py", line 53, in main
    workflow.run()
  File "/media/tim/GIT/google/CFU-Playground/soc/board_specific_workflows/general.py", line 125, in run
    soc_builder = self.build_soc(soc)
  File "/media/tim/GIT/google/CFU-Playground/soc/board_specific_workflows/digilent_arty.py", line 73, in build_soc
    return super().build_soc(soc, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/soc/board_specific_workflows/general.py", line 102, in build_soc
    soc_builder.build(run=self.args.build, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 350, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/soc/integration/soc.py", line 1147, in build
    return self.platform.build(self, *args, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/build/xilinx/platform.py", line 73, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/build/xilinx/symbiflow.py", line 241, in build
    _run_make()
  File "/media/tim/GIT/google/CFU-Playground/third_party/python/litex/litex/build/xilinx/symbiflow.py", line 81, in _run_make
    raise OSError(msg)
OSError: Unable to find Symbiflow toolchain, please:
- Add Symbiflow toolchain to your $PATH.
make[3]: *** [/media/tim/GIT/google/CFU-Playground/soc/common_soc.mk:115: build/digilent_arty.proj_template_v/gateware/digilent_arty.bit] Error 1
make[3]: Leaving directory '/media/tim/GIT/google/CFU-Playground/soc'
make[2]: *** [../proj.mk:313: bitstream] Error 2
make[2]: Leaving directory '/media/tim/GIT/google/CFU-Playground/proj/proj_template_v'
```
@umarcor, @tmichalak, where do we get the `symbiflow_*` scripts now?

tcal-x, 2022-07-22:
Note #1: I haven't updated to current LiteX.   If that is necessary, I can do that soon.

Note #2: this is where I add the path for the symbiflow scripts: https://github.com/google/CFU-Playground/blob/main/Makefile#L73.    That directory `env/symbiflow/bin/` is no longer present.

tcal-x, 2022-07-22:
Ok, I'm playing around adding some pip packages to `conf/requirements-symbiflow.txt` (in particular `main.zip`) and now I have the `symbiflow_*` scripts, but now some other files are not where they're expected (maybe this is a missing LiteX change).   The build is *looking* for `synth.tcl` at this location:
```
env/symbiflow/xc7/install/share/symbiflow/scripts/xc7/synth.tcl
```
but it's actually here:
```
env/symbiflow/share/symbiflow/scripts/xc7/synth.tcl
```
Or perhaps I should change where I install the tars to `env/symbiflow/xc7/install/` instead of `env/symbiflow/`.

(`F4PGA_INSTALL_DIR=/media/tim/GIT/google/CFU-Playground/env/symbiflow/`)

tcal-x, 2022-07-22:
@umarcor, @tmichalak, I think I have it figured out.   I'll send a PR tomorrow for you to review.

closed 2022-07-22 by tcal-x

mamuneeb, 2022-09-20:
Hii
I am still strugging with this.
While running `make prog TARGET=digilent_arty USE_SYMBIFLOW=1`
i am getting this 
 meson /home/mamuneeb/CFU-Playground/third_party/python/pythondata-software-picolibc/pythondata_software_picolibc/data \
	-Dmultilib=false \
	-Dpicocrt=false \
	-Datomic-ungetc=false \
	-Dthread-local-storage=false \
	-Dio-long-long=true \
	-Dformat-default=integer \
	-Dincludedir=picolibc/riscv64-unknown-elf/include \
	-Dlibdir=picolibc/riscv64-unknown-elf/lib \
	--cross-file cross.txt
WARNING: Unknown CPU family riscv, please report this at https://github.com/mesonbuild/meson/issues/new
The Meson build system
Version: 0.63.99
Source dir: /home/mamuneeb/CFU-Playground/third_party/python/pythondata-software-picolibc/pythondata_software_picolibc/data
Build dir: /home/mamuneeb/CFU-Playground/soc/build/digilent_arty.proj_template/software/libc
Build type: cross build
Project name: picolibc
Project version: 1.7.7

../../../../../third_party/python/pythondata-software-picolibc/pythondata_software_picolibc/data/meson.build:35:0: ERROR: Unable to detect GNU compiler type:
Compiler stdout:

-----
Compiler stderr:
riscv64-unknown-elf-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.

-----

A full log can be found at /home/mamuneeb/CFU-Playground/soc/build/digilent_arty.proj_template/software/libc/meson-logs/meson-log.txt
make[6]: *** [/home/mamuneeb/CFU-Playground/third_party/python/litex/litex/soc/software/libc/Makefile:43: __libc.a] Error 1
make[6]: Leaving directory '/home/mamuneeb/CFU-Playground/soc/build/digilent_arty.proj_template/software/libc'
Traceback (most recent call last):
  File "./common_soc.py", line 57, in <module>
    main()
  File "./common_soc.py", line 53, in main
    workflow.run()
  File "/home/mamuneeb/CFU-Playground/soc/board_specific_workflows/general.py", line 125, in run
    soc_builder = self.build_soc(soc)
  File "/home/mamuneeb/CFU-Playground/soc/board_specific_workflows/digilent_arty.py", line 73, in build_soc
    return super().build_soc(soc, **kwargs)
  File "/home/mamuneeb/CFU-Playground/soc/board_specific_workflows/general.py", line 102, in build_soc
    soc_builder.build(run=self.args.build, **kwargs)
  File "/home/mamuneeb/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 344, in build
    self._generate_rom_software(compile_bios=use_bios)
  File "/home/mamuneeb/CFU-Playground/third_party/python/litex/litex/soc/integration/builder.py", line 281, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/home/mamuneeb/CFU-Playground/env/conda/envs/cfu-symbiflow/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/mamuneeb/CFU-Playground/soc/build/digilent_arty.proj_template/software/libc', '-f', '/home/mamuneeb/CFU-Playground/third_party/python/litex/litex/soc/software/libc/Makefile']' returned non-zero exit status 2.
make[5]: *** [/home/mamuneeb/CFU-Playground/soc/common_soc.mk:115: build/digilent_arty.proj_template/gateware/digilent_arty.bit] Error 1
make[5]: Leaving directory '/home/mamuneeb/CFU-Playground/soc'
make[4]: *** [../proj.mk:312: prog] Error 2
make[4]: Leaving directory '/home/mamuneeb/CFU-Playground/proj/proj_template' 

Please help me with this.