google/CFU-Playground #629: "make renode" is broken when using conda env

tcal-x, 2022-07-28:
To reproduce:
```
make env
make enter
cd proj/proj_template_v
make renode
```
What I see:
```
make[5]: Entering directory '/media/tim/GIT/tcal-x/CFU-Playground/proj/proj_template_v/build/renode'
[  8%] Building CXX object CMakeFiles/libVtop.dir/sim_main.cpp.o
[ 16%] Building CXX object CMakeFiles/libVtop.dir/media/tim/GIT/tcal-x/CFU-Playground/third_party/renode/verilator-integration-library/src/buses/cfu.cpp.o
[ 25%] Building CXX object CMakeFiles/libVtop.dir/media/tim/GIT/tcal-x/CFU-Playground/third_party/renode/verilator-integration-library/src/renode_cfu.cpp.o
[ 33%] Building CXX object CMakeFiles/libVtop.dir/CMakeFiles/libVtop.dir/Vcfu.dir/Vcfu.cpp.o
[ 41%] Building CXX object CMakeFiles/libVtop.dir/CMakeFiles/libVtop.dir/Vcfu.dir/Vcfu___024root__DepSet_h1d3cd730__0.cpp.o
[ 50%] Building CXX object CMakeFiles/libVtop.dir/CMakeFiles/libVtop.dir/Vcfu.dir/Vcfu___024root__Slow.cpp.o
[ 58%] Building CXX object CMakeFiles/libVtop.dir/CMakeFiles/libVtop.dir/Vcfu.dir/Vcfu___024root__DepSet_h1d3cd730__0__Slow.cpp.o
[ 66%] Building CXX object CMakeFiles/libVtop.dir/CMakeFiles/libVtop.dir/Vcfu.dir/Vcfu__Syms.cpp.o
[ 75%] Building CXX object CMakeFiles/libVtop.dir/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated.cpp.o
In file included from /media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/x86_64-conda-linux-gnu/include/c++/11.2.0/chrono:42,
                 from /media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/x86_64-conda-linux-gnu/include/c++/11.2.0/condition_variable:38,
                 from /media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_trace.h:31,
                 from /media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated.cpp:75:
/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/x86_64-conda-linux-gnu/include/c++/11.2.0/ctime:80:11: error: 'timespec_get' has not been declared in '::'
   80 |   using ::timespec_get;
      |           ^~~~~~~~~~~~
make[5]: *** [CMakeFiles/libVtop.dir/build.make:213: CMakeFiles/libVtop.dir/media/tim/GIT/tcal-x/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated.cpp.o] Error 1
make[5]: Leaving directory '/media/tim/GIT/tcal-x/CFU-Playground/proj/proj_template_v/build/renode'
make[4]: *** [CMakeFiles/Makefile2:83: CMakeFiles/libVtop.dir/all] Error 2
make[4]: Leaving directory '/media/tim/GIT/tcal-x/CFU-Playground/proj/proj_template_v/build/renode'
make[3]: *** [CMakeFiles/Makefile2:90: CMakeFiles/libVtop.dir/rule] Error 2
make[3]: Leaving directory '/media/tim/GIT/tcal-x/CFU-Playground/proj/proj_template_v/build/renode'
make[2]: *** [Makefile:124: libVtop] Error 2
make[2]: Leaving directory '/media/tim/GIT/tcal-x/CFU-Playground/proj/proj_template_v/build/renode'
make[1]: *** [../proj.mk:211: renode-scripts] Error 2
make[1]: Leaving directory '/media/tim/GIT/tcal-x/CFU-Playground/proj/proj_template_v'
```
Potentially interrelated to #568, since this is also in the Verilator x86 compilation step.

tcal-x, 2022-07-28:
@ajelinski while investigating #568, also check that any solution fixes this, assuming they're related.

ajelinski, 2022-07-29:
Unfortunately I wasn't able to reproduce the problem.

I ran these exact commands in a fresh Ubuntu 20.04 docker container (started with `docker -ti run ubuntu:20.04`):
```
apt update
apt install -t git
git clone https://github.com/google/CFU-Playground
cd CFU-Playground/
sed -i 's#sudo ##g' scripts/setup
bash scripts/setup -ci
make env
make enter
cd proj/proj_template_v
```
and then tried both `make renode` and `make renode-headless` and they both worked.

ajelinski, 2022-07-29:
Btw. I've just created a PR for the `sudo` problem: https://github.com/google/CFU-Playground/pull/632

tcal-x, 2022-08-01:
Closing as not reproducible.   It has only been seen on my laptop.

closed 2022-08-01 by tcal-x